espressif/esp-idf #1831: Advertise Interval is Wrong!

Alireza89, 2018-04-06:
According to Bluetooth Core v4.0, Volume 6, Part B, Section 4.4.2.2 (Advertising Interval):

 <q>"If the advertising event type is either a scannable undirected event type or a non-connectable undirected event type, the advInterval shall not be less than 100 ms." </q>

However in the lines 53-54 of the ibeacon_demo.c file, the following code is present:
`    .adv_int_min        = 0x20,
    .adv_int_max        = 0x40,`

The problem is the adv_int_min is set to (0x20)*0.625ms = 20 ms and the adv_int_max is set to (0x40)*0.625ms = 40 ms but both of the values should be bigger the 0xA0 which yields a (0xA0)*0.625ms = 100 ms time interval.

jrickard, 2018-04-06:
I added a function to the Arduino version at ESP32-SNIPPETS allowing you to setMinInterval and setMaxInterval just to address this. It was adopted.

Jack

chegewara, 2018-04-06:
Im not saying i disagree with that, but here is interesting source:
`To be discovered by the Apple product, the accessory should first use the recommended advertising interval
of 20 ms for at least 30 seconds. If it is not discovered within the initial 30 seconds, Apple recommends using
one of the following longer intervals to increase chances of discovery by the Apple product`
Chapter 10.5
https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf

Weijian-Espressif, 2018-04-27:
@Alireza89 our bluetooth version is 4.2, 
the spec define:
4.4.2.2 Advertising Interval
For all undirected advertising events or connectable directed advertising events used in a low duty cycle mode, the time between the start of two consecutive advertising events (T_advEvent) is computed as follows for each advertising event:
T_advEvent = advInterval + advDelay
The advInterval shall be an integer multiple of 0.625 ms in the range of 20 ms to 10.24 s. If the advertising event type is **either a scannable undirected event type or a non-connectable undirected event type**, the advInterval shall not be less than 100 ms. **If the advertising event type is a connectable undirected event type or connectable directed event type used in a low duty cycle mode, the advInterval can be 20 ms or greater.**

closed 2018-05-09 by FayeY