espressif/esp-idf #1834: Bluetooth Controller: Crash on invalid LE Connection Parameters

mringwal, 2018-04-06:
When responding to an LE Remote Connection Parameter Update Request with an LE Remote Connection Paramater Request Reply, with conn min = conn max = 0, the Bluetooth Controller crashes.

Of course, the connection interval parameter are invalid, but it would be better if the controller would respond with a HCI Command Complete - INVALID_HCI_COMMAND_PARAMETERS instead of crashing. Given that the parameters are not validated, I guess related commands might cause similar behaviour.

Log snippet
--
...
Incoming Connection
[00:00:01.617] EVT <= 3E 13 01 00 00 00 01 00 D2 1B 6D 34 FE 18 18 00 04 00 48 00 01
[00:00:01.618] LOG -- hci.c.2280: LE Connection_complete (status=0) type 0, 18:FE:34:6D:1B:D2
..
LE Remote Connection Parameter Update Request
[00:00:03.276] EVT <= 3E 0B 06 00 00 0C 00 0C 00 04 00 64 00

LE Remote Connection Paramater Request Reply, with conn min = conn max = 0
[00:00:03.277] CMD => 20 20 0E 00 00 00 00 00 00 04 00 64 00 00 00 FF FF

Guru Meditation Error: Core  0 panic'ed (IntegerDivideByZero)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x4004ba16  PS      : 0x00060730  A0      : 0x80049e7c  A1      : 0x3ffd0a80
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x000011c0  A5      : 0x00000030
A6      : 0x00000078  A7      : 0x00000000  A8      : 0x00000003  A9      : 0x00000000
A10     : 0x3ffc0a88  A11     : 0x00060323  A12     : 0x00060320  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff

Backtrace: 0x4004ba16:0x3ffd0a80 0x40049e79:0x3ffd0aa0 0x40041b7b:0x3ffd0af0 0x40042386:0x3ffd0b20 0x4001a637:0x3ffd0b40 0x40019d11:0x3ffd0b70 0x40055b4d:0x3ffd0b90 0x400ea55b:0x3ffd0bb0 0x400ea5ab:0x3ffd0bd0
0x400ea55b: r_rw_schedule at ??:?

0x400ea5ab: btdm_controller_task at ??:?

================= CORE DUMP START =================
9BUAAAYAAAB0AQAA
bAz9P8AJ/T9kDP0/
8An9PwAM/T+jfy4oYDP8P2Az/D9sDP0/WDP8PwIAAAA0/Pw/NPz8P2wM/T8AAAAA
FwAAAGj8/D9idENvbnRyb2xsZXIAEvcAAAAAAGQM/T8AAAAAIAcGABcAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADSq/D+cqvw/BKv8PwAAAAAAAAAA
AQAAAAAAAABMKEA/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAKZiM
DQAAABa6BEAwBwYAfJ4EgIAK/T8AAAAAAAAAAMARAAAwAAAAeAAAAAAAAAADAAAA
AAAAAIgK/D8jAwYAIAMGAAAAAAABAAAAAAAAAB0AAAAGAAAAAAAAAGzEAEB3xABA
/////yADBgAAAAAA0B0IQKhF/D8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAKhF/D8AAAAAAAAAAAAAAAB+GwSAoAr9PwAAAABU2fs/
AAAAAAAAAAAAAAAAAAAAAIkjBIDwCv0/DOf6P+y3+z8AAAAAAwAAAAMAAD98JgE/
IB4IQHC0CEAAAAYAAwAAAAAAAAABAAAAAAAAAA4AAACst/s/DOf6P8yT+z8AAAAA
OqYBgCAL/T8AAAAA7Lf7PyAgAAAAAAAAAwAAACMGBgAGAAAAAJ4EQIDY+z+st/s/
FJ0BgEAL/T/gt/s/YIP7PwYAAAAgIAAAZAAAAAEAAABQWwWAcAv9P5SkAUCkgfs/
UFsFgHAL/T+EdQFApIH7PwUIAABo/fo/8Ir5PwAAAABepQ6AkAv9PwEAAAAAAAAA
HwAAAAAAAAAAAAAAsAX8P66lDoCwC/0/AQAAANAL/T9U/Pw/AAAAAFz8/D+wBfw/
AAAAANAL/T9YJvw/fCb8P/////9sDP0/AQAAAAAAAAAAAAAAAAz9PwAAAAAAAAAA
AQAAAAAAAAAAAAAAAAAAAAAAAAB4Jvw/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAMDP0/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAA==
5Nj8P5DU/D/c2Pw/
kNT8P3DY/D/UlUEAqDH8P6gx/D/k2Pw/oDH8PxgAAABPOvIAHKLAz+TY/D8AAAAA
AQAAAODG/D9tYWluAIhR1U2lvQxKZdkAAAAAANzY/D8AAAAAIwYGAAEAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADSq/D+cqvw/BKv8PwAAAAAAAAAA
AQAAAAAAAABMKEA/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA38Lw
IB4IQKk4CEAwCQYAtXEIgFDV/D8AAAAABwAAADAO/T8AAAAAAAAAAAEAAACpOAiA
MNX8P4AAAAAAAAAAAAAAAAAAAAACAAAAAAAA/xkAAACnjAWAENX8P+DCAED2wgBA
AAAAABkgCEAAAAAATLYIQBgS/D8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
[... 67 lines cut ...]
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAoL38PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAP////8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAMC9/D8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAADMvfw/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAA
aMX8P+DD/D9gxfw/
4MP8PwDF/D9H6ftdPL78PyQx/D9oxfw/HDH8PwEAAAA4wfw/OMH8P2jF/D8wwfw/
GAAAAGTB/D9pcGMwANMXxawRcytCKkAAAAAAAGDF/D8AAAAAIQAGABgAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADSq/D+cqvw/BKv8PwAAAAAAAAAA
AQAAAAAAAABMKEA/AAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAzB1/
IB4IQHC0CEAwAAYAJxYIgKDE/D8Mwfw/AAAAAFjB/D8AAAAAAQAAAAAAAABwtAiA
gMT8PwAAAACAM/w/gDP8P/A6/j/kCPw/AgAAAAAAAAClpaWlpaWlpQAAAAAAAAAA
AAAAABkgCEDwOv4/TLYIQKj++z8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA4MT8PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAP////8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAADF/D8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABBDgiAoDv+Pwgx/D/MM/w/
AAAAAAAAAAAMxfw/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
================= CORE DUMP END =================
CPU halted.

Weijian-Espressif, 2018-04-26:
sorry, I will fix it as soon as possable.

mringwal, 2018-04-26:
No problem - BTstack isn't sending invalid conn intervals usually. Please check if there are other ways to run into this crash (e.g. in a le create connection command or similar)

FayeY, 2018-05-21:
It has been merged into the latest master, please have a try. Thanks.

closed 2018-05-21 by FayeY