espressif/esp-idf #1835: lwip - memory leak

Reiner1210, 2018-04-07:
Hello,

I assume that there is a memory leak in lwip.

I have made a "little" example.

Full working example can be downloaded from here
http://tmp.natureimpression.com/lwip_test.tgz

set via make menuconfig xou SSID and password. The code open the port 80 and echos all inputs. If 5 sec no input is detected the client thread is closed. Every 10 sec the free heap size is traced out.

For testing use telnet 80. Then type some nice words and press Enter - you should the the echoed string. Now wait - you will see that the client thread goes home.

zhangyanjiaoesp, 2018-04-11:
@Reiner1210  when we create a socket, we will create a lock to protect the socket, this will take up some memory. when the socket is closed, the memory that occupied by the lock will not be set to free.
, only when the code is closed totally, the lock will be set to free.
   So you will find the memory isn't equal to the start of the code. But when you restart the code, you will find there is no memory leak.

Reiner1210, 2018-04-11:
What do you mean with closed totally ?
What I understand is that if a server is running and listen on a port every 
time a client connects some memory get lost - not good!

Reiner

Mit AquaMail Android
https://www.mobisystems.com/aqua-mail

zhangyanjiaoesp, 2018-04-11:
By saying `close totally`, I mean reboot or restart the code. 
Current the lwip can create 10 sockets(sock id 4096~4015), for each socket, there is a lock. If you have created 10 sockets, and you close some of the socket, such as `sock 4096`, the socket closed, but the lock isn't set to free. The next time you want to create a socket, the `sock 4096` will be created again, then there is no need to create a lock for `sock 4096` again.  I think this design is reasonable.
Can you understand what I mean?

Reiner1210, 2018-04-11:
Ok. I get it. You have a pool of locks which are reused. That's ok. So we 
will not run out of memory

Mit AquaMail Android
https://www.mobisystems.com/aqua-mail

closed 2018-04-12 by FayeY