espressif/esp-idf #1837: Deadlock in fsync - thread safe?

markwj, 2018-04-08:
I have an app that calls:

```
fwrite ...
fflush ...
fsync ...
```

That may be called from different threads.

It is randomly locking up the FAT filesystem that file is on.

Stack traces of two threads are:

```
0x4008526a: lock_acquire_generic at /Users/mark/esp/esp-idf/components/newlib/./locks.c:157
0x4008ebed: xQueueGenericReceive at /Users/mark/esp/esp-idf/components/freertos/./queue.c:2037
0x4015e6b5: esp_vfs_write at /Users/mark/esp/esp-idf/components/vfs/./vfs.c:540
0x4008533c: _lock_acquire at /Users/mark/esp/esp-idf/components/newlib/./locks.c:165
0x401e2e53: vfs_fat_fsync at /Users/mark/esp/esp-idf/components/fatfs/src/vfs_fat.c:350
0x40096a8c: _fflush_r at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fflush.c:282
0x4015eb7d: fsync at /Users/mark/esp/esp-idf/components/vfs/./vfs.c:552 (discriminator 3)
...
```

and:

```
0x4008526a: lock_acquire_generic at /Users/mark/esp/esp-idf/components/newlib/./locks.c:157
0x4008ebed: xQueueGenericReceive at /Users/mark/esp/esp-idf/components/freertos/./queue.c:2037
0x4015e6b5: esp_vfs_write at /Users/mark/esp/esp-idf/components/vfs/./vfs.c:540
0x4008533c: _lock_acquire at /Users/mark/esp/esp-idf/components/newlib/./locks.c:165
0x40000000: ?? ??:0
0x401e2e53: vfs_fat_fsync at /Users/mark/esp/esp-idf/components/fatfs/src/vfs_fat.c:350
0x4015eb7d: fsync at /Users/mark/esp/esp-idf/components/vfs/./vfs.c:552 (discriminator 3)
...

```

If I put a mutex around the fsync call, it seems to resolve the situation.

igrr, 2018-04-11:
@markwj I've tried writing a test to reproduce this, without much success. Does this issue happen often? If you can reproduce it easily, could you please try one thing — remove `_lock_acquire`/`_lock_release` calls from `vfs_fat_fsync` in components/fatfs/vfs_fat.c.

igrr, 2019-04-10:
Closing due to lack of feedback. Please feel free to reopen if the issue still exists.

closed 2019-04-10 by igrr