espressif/esp-idf #1839: rx state

jixunx, 2018-04-09:
typedef struct {
    signed rssi:8;            /**< signal intensity of packet */
    unsigned rate:5;          /**< data rate */
    unsigned :1;              /**< reserve */
    unsigned sig_mode:2;      /**< 0:is not 11n packet; 1:is 11n packet */
    unsigned :16;             /**< reserve */
    unsigned mcs:7;           /**< if is 11n packet, shows the modulation(range from 0 to 76) */
    unsigned cwb:1;           /**< if is 11n packet, shows if is HT40 packet or not */
    unsigned :16;             /**< reserve */
    unsigned smoothing:1;     /**< reserve */
    unsigned not_sounding:1;  /**< reserve */
    unsigned :1;              /**< reserve */
    unsigned aggregation:1;   /**< Aggregation */
    unsigned stbc:2;          /**< STBC */
    unsigned fec_coding:1;    /**< Flag is set for 11n packets which are LDPC */
    unsigned sgi:1;           /**< SGI */
    unsigned noise_floor:8;   /**< noise floor */
    unsigned ampdu_cnt:8;     /**< ampdu cnt */
    unsigned channel:4;       /**< which channel this packet in */
    unsigned :12;             /**< reserve */
    unsigned timestamp:32;    /**< timestamp */
    unsigned :32;             /**< reserve */
    unsigned :32;             /**< reserve */
    unsigned sig_len:12;      /**< length of packet */
    unsigned :12;             /**< reserve */
    unsigned rx_state:8;      /**< rx state */
} wifi_pkt_rx_ctrl_t;

From this structure, we can see that 8 bits of them describe rx_state. Can you provide information about the meaning of each value of rx_state?

liuzfesp, 2018-04-09:
@jixunx , if rx_state == 0, it means no error, if rx_state == 0x62, it means esp32 receives MIMO packets which esp32 doesn't support. For all other values, the WiFi driver will discard the packets and the packet will not be filtered to application, in other words, you can only see value 0 or 0x62.

jixunx, 2018-04-09:
Thanks for your reply.It is really helpful.

jixunx, 2018-04-09:
But I hope to further analyze the reasons for RX error. I hope to grab the value of rx_state through the promiscuous mode. And the promiscuous mode should see all the error values. It's not only 0 and 0x62, but can you tell me the meaning of every  value of rx_state?

jixunx, 2018-04-09:
Thanks!
Is there a table about the meaning of every value of rx_state?

liuzfesp, 2018-04-09:
Yes, we a table, but now it's not public

closed 2018-04-17 by FayeY