espressif/esp-idf #1840: Wakeup from Deep Sleep

hvishu, 2018-04-09:
We are using deep sleep in our project and below test app is how we are using it.

We have observed that after wakeup "sleep_time_ms" is -ve, is this a known issue?  Thought issue is not reproducible everytime, when it happens the value is about -1234 Seconds...

I (34) boot: ESP-IDF v3.0-dev-1187-g09d2791-dirty 2nd stage bootloader
I (34) boot: compile time 18:39:52
I (34) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (44) boot: SPI Mode       : DIO
I (48) boot: SPI Flash Size : 4MB
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 filt_cong        unknown          41 00 00010000 00001000
I (85) boot:  3 aws_cert         unknown          40 00 00011000 00002000
I (92) boot:  4 factory          factory app      00 00 00020000 00140000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x4862c (296492) map
I (216) esp_image: segment 1: paddr=0x00068654 vaddr=0x3ffc0000 size=0x04808 ( 18440) load
I (224) esp_image: segment 2: paddr=0x0006ce64 vaddr=0x40080000 size=0x00400 (  1024) load
I (224) esp_image: segment 3: paddr=0x0006d26c vaddr=0x40080400 size=0x02da4 ( 11684) load
I (237) esp_image: segment 4: paddr=0x00070018 vaddr=0x400d0018 size=0xd6150 (876880) map
I (545) esp_image: segment 5: paddr=0x00146170 vaddr=0x400831a4 size=0x12dfc ( 77308) load
I (577) esp_image: segment 6: paddr=0x00158f74 vaddr=0x400c0000 size=0x00064 (   100) 
I (577) esp_image: segment 7: paddr=0x00158fe0 vaddr=0x50000000 size=0x00008 (     8) 
I (596) boot: Loaded app from partition at offset 0x20000
I (596) boot: Disabling RNG early entropy source...
I (597) cpu_start: Pro cpu up.
I (601) cpu_start: Starting app cpu, entry point is 0x40081320
I (0) cpu_start: App cpu up.
I (611) heap_init: Initializing. RAM available for dynamic allocation:
I (618) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (624) heap_init: At 3FFD4E98 len 0000B168 (44 KiB): DRAM
I (630) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (636) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (643) heap_init: At 40095FA0 len 0000A060 (40 KiB): IRAM
I (649) cpu_start: Pro cpu start user code
I (331) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (333) APP_DEBUG: Hello world!
I (333) APP_DEBUG: This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, 
I (343) APP_DEBUG: silicon revision 1, 
I (343) APP_DEBUG: 4MB external flash
[pwr_detect.c.zip](https://github.com/espressif/esp-idf/files/1888829/pwr_detect.c.zip)

igrr, 2018-04-13:
I ran the code you attached for a while, but couldn't reproduce the issue.

Could you please log the values of `now.tv_sec` and `sleep_enter_time.tv_sec` separately?

Also, could you please attach your sdkconfig file? I'm interested in configuration settings related to timekeeping.

hvishu, 2018-04-17:
Attached is the sdk config file being used.  
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/1919043/sdkconfig.zip)

Also the problem is not consistently reproducible, it occurs occasionally that too, when kept for duration testing (board connected to raw power, and sometimes the problem occurs when there is a loss of main power)

hvishu, 2018-04-29:
@igrr Are you able to reproduce this issue?  Any updates for me?

igrr, 2018-05-07:
@hvishu What reset reason do you get when the issue happens? The reset reason can be seen in ROM boot log, the line looks like this:

```
rst:0x1 (POWERON_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
```

Also could you please log values of `now.tv_sec` and `sleep_enter_time.tv_sec` separately? It's not clear if it is the current time that is incorrect, or `sleep_enter_time`.

hvishu, 2018-05-07:
@igrr Please find answers to your question below
After waking from deep sleep,
rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

below four mesages, outputs
        ESP_LOGE(DEBUG_MSGS,"WakeUp Reason = %d",esp_sleep_get_wakeup_cause());
        ESP_LOGE(DEBUG_MSGS,"now.tv_sec = %ld sleep_enter_time.tv_sec = %ld",now.tv_sec,sleep_enter_time.tv_sec);
        ESP_LOGE(DEBUG_MSGS,"Time during sleep returned is -ve: %d",sleep_time_sec);
        ESP_LOGE(DEBUG_MSGS,"Forcing it to Zero, System will have same time it has saved before going to deepSleep");

E (3175) APP_DEBUG: WakeUp Reason = 1
E (3175) APP_DEBUG: now.tv_sec = 1525690747 sleep_enter_time.tv_sec = 1525690748
E (3185) APP_DEBUG: Time during sleep returned is -ve: -1
E (3195) APP_DEBUG: Forcing it to Zero, System will have same time it has saved before going to deepSleep

igrr, 2018-05-07:
Thanks a lot @hvishu, that's very useful. May I ask you to log two more things, before and after sleep?

```c++
#include "esp_clk.h"
#include "soc/rtc.h"

uint32_t slowclk_cal = esp_clk_slowclk_cal_get();
uint64_t rtc_time = rtc_time_get();
ESP_LOGE(DEBUG_MSGS,"slowclk_cal=%d rtc_time=%lld", slowclk_cal, rtc_time);
```

When you see the issue, what is the actual deep sleep duration? (seconds, minutes?)

hvishu, 2018-05-08:
Actual sleep duration is from few seconds to minutes (seems not depend on how much time spent in Deep Sleep).  I will get the additional debug prints you requested when this happens.

hvishu, 2018-05-08:
I am not sure whether this will help or not, we have the prints after waking from the sleep.  We will put the slowclk_cal & rtc_time prints before going to DS and update.

E (3175) APP_DEBUG: slowclk_cal=3069850 rtc_time=160975216
E (3175) APP_DEBUG: WakeUp Reason = 1
E (3185) APP_DEBUG: now.tv_sec = 1525779595 sleep_enter_time.tv_sec = 1525779605
E (3195) APP_DEBUG: Time during sleep returned is -ve: -10

hvishu, 2018-05-08:
Another trial
Before going to Deep Sleep:
slowclk_cal=2988211 rtc_time=157802062
Wake from Deep Sleep:
E (3175) APP_DEBUG: slowclk_cal=3065011 rtc_time=159259082
E (3175) APP_DEBUG: WakeUp Reason = 1
E (3185) APP_DEBUG: now.tv_sec = 1525783153 sleep_enter_time.tv_sec = 1525783171
E (3195) APP_DEBUG: Time during sleep returned is -ve: -18

hvishu, 2018-05-30:
Dear @igrr,
Solving this issue is very critical for us, waiting for your fix/suggestions.

igrr, 2018-05-30:
Hi @hvishu, sorry for the delay. We have set up a few boards and are running tests on them. So far, we couldn't reproduce this issue. Could you please give some hint, are there any significant changes in temperature or supply voltage during device operation? Clock calibration value changes from 2988211 to 3065011, this is a fairly large change, even for the internal RTC oscillator.

Another thing, could you please choose "RTC" as the value of "Timers used for gettimeofday function" under menuconfig, component config, ESP32-specific, instead of "RTC and high-resolution timer" you have chosen now, and give it a try? Thanks.

hvishu, 2018-06-04:
Supply voltage is driven by a super capacitor during the deep sleep time.  The idea is to keep the ESP32 as much as possible.  And during this period voltage will be goind down.  See attached schematics.

Now we have kept RTC as per your suggestion and didn't face the problem in limited testing that's carried out.  Will update further.  Could you tell me the differences between these two options and how it is helping to resolve this issue?
[image]

KonstantinKKK, 2018-06-07:
Hi @hvishu. We think what this behavior due to the dispersion of the calibration coefficient of the RC chain.  Based on this, we can reproduce this behavior. 
In case of use option "RTC and high-resolution timer" the function gettimeofday() returns time based on two counters RTC and FRC. For RTC need exactly calibration coefficient but we see that it have more dispersion. The difference between the two counters accumulates and this leads to incorrect timing.

We decided to synchronize the two counters before entering deep sleep.
We will make a correction soon. You can use this patch or wait for the IDF to update or remain on the "RTC" option.
[2501_bugfix_wakeup_from_deep_sleep.zip](https://github.com/espressif/esp-idf/files/2078761/2501_bugfix_wakeup_from_deep_sleep.zip)
Thanks.

hvishu, 2018-06-07:
Hi @KonstantinKKK.
Thank you very much for the details.  So far on using RTC configuration we haven't seen the issue (-ve value).
I would like to remain in RTC, if it serves the purpose without any other problems. If you could tell us advantage/difference of "RTC and high-resolution timer" compared to "RTC", it will help us to make the decision.

closed 2018-06-08 by igrr

KonstantinKKK, 2018-06-09:
Hi @hvishu.
If you want to remain in "RTC" it is well. It must work without any problem. 
Differences between the options 1."RTC and high-resolution timer" and 3."RTC" in a resolution of counters. 1 option has more resolution and therefore has the exact counting of time.

For more detail information: https://esp-idf.readthedocs.io/en/latest/api-reference/kconfig.html#esp32-time-syscall