espressif/esp-idf #1841: rtc_module.c compile error

fxzengfrank, 2018-04-09:
while make all

CC build/driver/rtc_module.o
/esp-idf/components/driver/./rtc_module.c: In function 'adc_convert':
/esp-idf/components/driver/./rtc_module.c:1245:12: error: 'adc_value' may be used uninitialized in this function [-Werror=maybe-uninitialized]
     return adc_value;
            ^
cc1: some warnings being treated as errors
make[1]: *** [rtc_module.o] Error 1
make: *** [component-driver-build] Error 2

moraleja39, 2018-04-11:
Until it gets fixed, you can compile by setting "Assertion level" to anything but "Disabled" under "Compiler options".
IE. the -DNDEBUG seems to be the cause to the compilation error.

panfeng-espressif, 2018-04-16:
Hello fxzengfrank,
Would you mind sharing your commit id of ESP-IDF?
Is it a demo code or your own project?
Thanks

mahavirj, 2018-04-17:
@fxzengfrank Thanks for reporting this. Fix for this is in progress, should be merged soon.

closed 2018-04-19 by igrr