espressif/esp-idf #1842: installng ESP-IDF in centos 7.4 with minimum install options

stoikos, 2018-04-10:
I tried to install the latest ESP-IDF in Centos 7.4 with minimum install options. 
I followed exactly the process described in [Standard Setup of Toolchain for Linux](https://esp-idf.readthedocs.io/en/latest/get-started/linux-setup.html) and [Get ESP-IDF](https://esp-idf.readthedocs.io/en/latest/get-started/index.html#get-esp-idf)

in Centos  .profile was not created so i have put
```
export PATH="$PATH:$HOME/esp/xtensa-esp32-elf/bin"
export IDF_PATH=~/esp/esp-idf
```
in .bash_profile. This is the result of it
```
printenv PATH
/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/home/george/esp/xtensa-esp32-elf/bin:/home/george/.local/bin:/home/george/bin

printenv IDF_PATH
/home/george/esp/esp-idf
```
then i did what [Start a Project](https://esp-idf.readthedocs.io/en/latest/get-started/index.html#start-a-project) suggests. 

Then I executed 'make menuconfig'. The first error i got was that bash couldn't find cc.  To resolve this i added
```
export CC=~/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
```
in .bash_profile 
then the next error i got was 
```
make[1]: Entering directory `/home/george/esp/esp-idf/tools/kconfig'
/home/george/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc    -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o mconf.o mconf.c
In file included from mconf.c:22:0:
lkc.h:12:22: fatal error: libintl.h: No such file or directory
compilation terminated.
make[1]: *** [mconf.o] Error 1
make[1]: Leaving directory `/home/george/esp/esp-idf/tools/kconfig'
make: *** No rule to make target `/home/george/esp/esp-idf/tools/kconfig/conf', needed by `/home/george/esp/hello_world/build/include/config/auto.conf'.  Stop.
```
please note the 
```
lkc.h:12:22: fatal error: libintl.h: No such file or directory
```
this is what i get 
```
sudo yum install gettext
[sudo] password for george:
Loaded plugins: fastestmirror
Loading mirror speeds from cached hostfile
 * base: mirror.ash.fastserv.com
 * extras: mirrors.advancedhosters.com
 * updates: mirror.metrocast.net
Package gettext-0.19.8.1-2.el7.x86_64 already installed and latest version
Nothing to do
```
I installed gcc with yum but nothing changed. Any ideas?

chegewara, 2018-04-10:
I know this is for rasp pi but maybe it can help you to find prerequisites:
https://raspberrypi.stackexchange.com/questions/1/how-do-i-build-a-gcc-4-7-toolchain-for-cross-compiling

milamber-ls, 2018-04-10:
I did install the dependencies from [Setup Linux Toolchain from Scratch](https://esp-idf.readthedocs.io/en/latest/get-started/linux-setup-scratch.html) for Centos 7
this is the result
```
 sudo yum install gawk gperf grep gettext ncurses-devel python python-devel automake bison flex texinfo help2man libtool
Loaded plugins: fastestmirror
Loading mirror speeds from cached hostfile
 * base: mirror.ash.fastserv.com
 * extras: mirrors.advancedhosters.com
 * updates: mirror.metrocast.net
Package gawk-4.0.2-4.el7_3.1.x86_64 already installed and latest version
Package gperf-3.0.4-8.el7.x86_64 already installed and latest version
Package grep-2.20-3.el7.x86_64 already installed and latest version
Package gettext-0.19.8.1-2.el7.x86_64 already installed and latest version
Package ncurses-devel-5.9-14.20130511.el7_4.x86_64 already installed and latest version
Package python-2.7.5-58.el7.x86_64 already installed and latest version
Package python-devel-2.7.5-58.el7.x86_64 already installed and latest version
Package automake-1.13.4-3.el7.noarch already installed and latest version
Package bison-3.0.4-1.el7.x86_64 already installed and latest version
Package flex-2.5.37-3.el7.x86_64 already installed and latest version
Package texinfo-5.1-4.el7.x86_64 already installed and latest version
Package help2man-1.41.1-3.el7.noarch already installed and latest version
Package libtool-2.4.2-22.el7_3.x86_64 already installed and latest version
Nothing to do
```

chegewara, 2018-04-10:
Did you reboot system?

milamber-ls, 2018-04-10:
i did. the issue persists

negativekelvin, 2018-04-10:
gettext-devel ?

milamber-ls, 2018-04-10:
i installed gettext-devel
```
Installed:
  gettext-devel.x86_64 0:0.19.8.1-2.el7

Dependency Installed:
  gettext-common-devel.noarch 0:0.19.8.1-2.el7
```
but the issue persists

jcmvbkbc, 2018-04-10:
> make[1]: Entering directory `/home/george/esp/esp-idf/tools/kconfig'
> /home/george/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc    -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c [...]

You're trying to build your development environment with target CC. That's not what you want. Use host CC for that. I.e. install host gcc package and unset CC environment variable.

milamber-ls, 2018-04-10:
@jcmvbkbc . success! after i deleted the CC it compiled. I have 2 more questions.
1. my assumption is that it didn't work at the beginning due to the 'minimal install' setting. it required the installation of gcc package from the repo. is this correct?
2. this is what i get 
```
which gcc
/usr/bin/gcc
```
which compiler compiles the esp32 program? the xtensa or the standard gcc?

Again thank you!

projectgus, 2018-04-10:
Hi @milamber-ls ,

Glad you got everything working.

> which compiler compiles the esp32 program? the xtensa or the standard gcc?

xtensa-esp32-elf-gcc compiles programs for ESP32 (this is the "target" in cross-compiler language).

gcc compiles programs for the host (ie your local system running CentOS).

The ESP-IDF build system assumes `xtensa-esp32-elf-gcc` is the name of the target compiler. This can be changed in "make menuconfig" under the SDK Tools menu, but most people will never need to do this. However, if the CC variable is set, the build system uses this to override the host compiler.

mconf is a host-side tool which is compiled (using the host compiler, ie normal gcc) by the build system to run the "menuconfig" menu.

closed 2018-04-11 by milamber-ls