espressif/esp-idf #1843: esp_wifi_set_protocol() error on some boards

moraleja39, 2018-04-11:
Hi. I have a couple of ESP-WROOM-32 dev boards. I'm trying to set the WiFi protocol as b or bg, using this portion of code:

```
...
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
ESP_ERROR_CHECK( esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_11B) );
...
```

On some of the boards, when running the program, it throws the following error:

```
I (525) wifi: wifi driver task: 3ffc3168, prio:23, stack:4096
I (525) wifi: Init static rx buffer num: 10
I (535) wifi: Init dynamic rx buffer num: 32
ESP_ERROR_CHECK failed: esp_err_t 0x102 at 0x400d3319
0x400d3319: ESPNow::init() at /home/david/esp/esp32-client/main/./espnow.cpp:32 (discriminator 1)

file: "/home/david/esp/esp32-client/main/./espnow.cpp" line 32
func: void ESPNow::init()
expression: esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_11B)

Backtrace: 0x4008e9a0:0x3ffba6f0 0x4008ee35:0x3ffba710 0x400d3319:0x3ffba730 0x400d307d:0x3ffba7a0 0x400d2919:0x3ffba7e0 0x400d2a29:0x3ffba800 0x400d0e1a:0x3ffba820
0x4008e9a0: invoke_abort at /home/david/esp/esp-idf/components/esp32/./panic.c:648

0x4008ee35: _esp_error_check_failed at /home/david/esp/esp-idf/components/esp32/./panic.c:660

0x400d3319: ESPNow::init() at /home/david/esp/esp32-client/main/./espnow.cpp:32 (discriminator 1)

```

However, on some other (identical) boards the same program runs fine, and the WiFi protocol is switched correctly.

Any help on this would be highly appreciated.

liuzfesp, 2018-04-17:
@zhangyanjiaoesp please help.

igrr, 2018-04-17:
Perhaps you need to set the mode first (STA or AP_STA) before trying to set the protocol? If the board has been previously switched to STA mode, and settings saved to flash, you will not get the error. If it has not been used before, it may default to AP mode, hence you get an error when trying to set protocol for STA interface.

zhangyanjiaoesp, 2018-06-01:
@moraleja39 Does igrr's suggestion help you?

closed 2018-06-13 by FayeY