espressif/esp-idf #1844: Is there a way to toggle an I/O on wifi transmit?

S-March, 2018-04-11:
I have a product that I am developing that has very tight battery requirements, it also requires an amplifier on the TX side of a WiFi transmit. I would ideally like to keep the amplifier off during the RX side. 

Is there a way to expose an I/O to toggle on/off during transmit so I can enable my amplifier?

negativekelvin, 2018-04-11:
It should be possible using an antenna select gpio signal but I don't know if it is supported.

negativekelvin, 2018-04-26:
@liuzfesp he means external PA

S-March, 2018-04-26:
Exactly. Looking for a way to expose an IO to toggle an external PA on TX only.

liuzfesp, 2018-06-22:
@S-March, ESP32's GPIO_matrix can mux out the function to 1~4 IOs to control external PA and RF switch, the API hasn't been ready yet.

negativekelvin, 2018-06-22:
https://github.com/espressif/esp-idf/commit/04c74efedd62ef50beddca7e1050beba9d21f419

closed 2019-03-19 by Alvin1Zhang