espressif/esp-idf #1847: Routing Wifi AP to PPPoS client

crismansantos, 2018-04-11:
Hi!
I need help kkk
I'm trying route Wifi Ap to PPPoS interface, but it's not working.
I tryed many things, but nothing work.
Now I trying use netif_set_default to set ppp interface as default network, but isnt work too
Any ideas?

Sorry for my english, I'm brazilian kkkk

liuzfesp, 2018-06-04:
@crismansantos sorry, currently we don't have example to route WiFi AP to PPPoS interface. I think you can provide your demo code here firstly then more people can join the discussion.

liuzfesp, 2018-06-22:
Hi @crismansantos, I'm afraid to let you know that don't have plan to provide support for PPP in short term. Will let you know once we plan to support it.

gorkembendin, 2019-04-12:
Hi, 
Are there any updates about this topic?
We have a ZTE ME3630 GSM module and our purpose is to provide internet connection via GSM module to all WiFi Access Point users. So, could you provide me any solution about that?

suda-morris, 2019-04-12:
@gorkembendin We have some examples illustrating how to use these Celluar modules, based on what you may implement your GSM<->AP application although the bitrate might be very slow.  https://github.com/espressif/esp-idf/tree/master/examples/protocols/pppos_client

gorkembendin, 2019-04-12:
Thank you for your reply. I will check it and give feedback!

gorkembendin, 2019-04-16:
Hey again!
I have checked the example that you suggested but couldn't find anything about Wifi Access Point connection using internet which is provided by GSM module. :(

suda-morris, 2019-04-17:
@gerekon Sorry, your code is largely different from the example. It seems like you're not using the latest code from IDF. We have updated the example since [this commit](https://github.com/espressif/esp-idf/commit/2e1f98f8f5042c9132132534205f0ffe120f4aed)

gorkembendin, 2019-04-17:
Thank you! I have updated my IDF library and checked the new example but there is also nothing about passing data from GSM to Wifi Access Point.

suda-morris, 2019-04-17:
@gorkembendin Yes, the example just illustrate how to drive the modem modules but won't do any complex work (e.g. data forward from GSM to Wi-Fi AP). We may add such an example/solution in the future but currently it not high priority. PR is welcome if you have more ideas to share.

gorkembendin, 2019-04-17:
Thank you for your interest and reply :)

will-emmerson, 2020-08-11:
I've just got this working, although I don't really know what I'm doing and mostly copied https://github.com/jonask1337/esp-idf-nat-example. It's slow so is only going to work for sending small, infrequent messages.

- combine wifi ap example with pppos_client example such that pppos is starting first and wifi_ap isn't trying to re-initialize netif stuff. Traffic won't be forwarded between AP and pppos yet.

 - enable CONFIG_LWIP_IP_FORWARD and CONFIG_LWIP_IPV4_NAPT:
IP_FORWARD forwards packets from one interface to another i.e. AP and pppos.  
NAPT (network address translation) forwards traffic to and from the Internet for clients on the AP. Or something.
Note the `esp-idf-nat-example` above uses a modified lwip to get these features but that's not needed anymore if you use esp-idf master.

- enable dns/dhcp so that AP clients get an ip address (after enabling pppos and before enabling wifi):
```c
    // Enable DNS (offer) for dhcp server
    dhcps_offer_t dhcps_dns_value = OFFER_DNS;
    dhcps_set_option_info(DOMAIN_NAME_SERVER, &dhcps_dns_value, sizeof(dhcps_dns_value));

    // Set custom dns server address for dhcp server
    ip_addr_t dnsserver = {
            .u_addr.ip4.addr = esp_ip4addr_aton("8.8.8.8"),
            .type = IPADDR_TYPE_V4,
    };
    dhcps_dns_setserver(&dnsserver);
```
I don't really understand this part.

- lastly, enable NAT for AP interface:
```c
ip_napt_enable(esp_ip4addr_aton("192.168.4.1"), true);
```

JosueSanchezEZM, 2021-05-03:
I did try doing the same and finally I raise the AP interface to connect after the PPP session obtains an IP address for the cellular network. But now, after getting this IP and enabling DNS and NAPT I am not able to do a ping to my remote devices from the local network of the AP. Have you experienced something similar? I know it is something to be with the NAPT configuration because when I do a ping from a remote server I have to the IP my device has, it success.

will-emmerson, 2021-05-06:
I actually gave up on that code because a) I didn't really understand all the NAT stuff, b) I felt like I was doing something which wasn't properly supported in IDF and c) it was a bit flakey, and for those reasons I didn't feel confident using it in a production environment.
I tried to instead use the mesh sdk, where the root connected using pppos and other nodes connected to it but also didn't get that to work. If you're not bothered about ip connectivity on your nodes, only on your root, then that would be the sensible way to do it, because the mesh sdk is doing all the routing for you.

closed 2021-09-10 by espressif-bot

Guitton-S, 2022-01-05:
Hi all,

I see that this issue is closed and the corresponding commit though I do not see how this commit is resolving this issue.
Is there a way or an example on the current ESP-IDF to route the Wifi AP to PPPoS client??

Thank you!