espressif/esp-idf #1851: Very short SMP_RSP_TIMEOUT during BLE pairing

semireg, 2018-04-12:
I'm using gatt_security_server to implement pairing/encryption for PyCom WiPY based on ESP32. I can successfully run the example but am having issues with only 3-4 seconds between getting the passkey prompt in iOS/Android and the ESP32 timing out w/ SMP_RSP_TIMEOUT. In other issues/threads/documentation it says that 30 seconds should be allowed. Does anyone know why I experience shorter timeouts?

mahavirj, 2018-04-19:
@Yulong-espressif Could you please comment here?

Yulong-espressif, 2018-04-19:
@semireg You should used the esp_ble_gap_set_prefer_conn_params function to set the supervision timeout value. To give the user a better experience, we set the default value for the supervision timeout to 6s. Hope this answer can help you, thanks.

closed 2018-04-24 by FayeY