espressif/esp-idf #1853: Guru Meditation Error: Core  0 panic'ed (LoadProhibited) . Exception was unhandled.

mohinishnaik, 2018-04-13:
Hi,
I am getting following error when I run my code.

Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x400813aaGuru Meditation Error: Core  0 panic'ed (LoadStoreError)
0x400813aa: esp_crosscore_int_send at /home/pod/esp/esp-idf/components/esp32/./crosscore_int.c:100

. Exception was unhandled.
Core 0 register dump:
PC      : 0x4008cc79Guru Meditation Error: Core  0 panic'ed (LoadStoreError)
0x4008cc79: panicPutStr at /home/pod/esp/esp-idf/components/esp32/./panic.c:648

. Exception was unhandled.
Core 0 register dump:
PC      : 0x4008cc79Guru Meditation Error: Core  0 panic'ed (LoadStoreError)
0x4008cc79: panicPutStr at /home/pod/esp/esp-idf/components/esp32/./panic.c:648

. Exception was unhandled.
Core 0 register dump:
PC      : 0x4008cc79Guru Meditation Error: Core  0 panic'ed (LoadStoreError)
0x4008cc79: panicPutStr at /home/pod/esp/esp-idf/components/esp32/./panic.c:648

. Exception was unhandled.
Core 0 register dump:
PC      : 0x4008cc79Guru Meditation Error: Core  0 panic'ed (LoadStoreError)
0x4008cc79: panicPutStr at /home/pod/esp/esp-idf/components/esp32/./panic.c:648

..........................continue

It continues to output . Exception was unhandled. and then it dumps the backtrace on serial monitor 

Backtrace: 0x4008cc79:0x3ffb33c0 0x4008cf46:0x3ffb33e0 0x4008d098:0x3ffb3460 0x4008d2a0:0x3ffb3480 0x400821f2:0x3ffb34a0 0x4008cc76:0x3ffb3560 0x4008cc76:0x3ffb3580 0x4008d098:0x3ffb3600 0x4008d2a0:0x3ffb3620 0x400821f2:0x3ffb3640 0x4008cc76:0x3ffb3700 0x4008cc76:0x3ffb3720 0x4008d098:0x3ffb37a0 0x4008d2a0:0x3ffb37c0 0x400821f2:0x3ffb37e0 0x4008cc76:0x3ffb38a0 0x4008cc76:0x3ffb38c0 0x4008d098:0x3ffb3940 0x4008d2a0:0x3ffb3960 0x400821f2:0x3ffb3980 0x4008cc76:0x3ffb3a40 0x4008cc76:0x3ffb3a60 0x4008d098:0x3ffb3ae0 0x4008d2a0:0x3ffb3b00 0x400821f2:0x3ffb3b20 0x4008cc76:0x3ffb3be0 0x4008cc76:0x3ffb3c00 0x4008d098:0x3ffb3c80 0x4008d2a0:0x3ffb3ca0 0x400821f2:0x3ffb3cc0 0x4008cc76:0x3ffb3d80 0x4008cc76:0x3ffb3da0 0x4008d098:0x3ffb3e20 0x4008d2a0:0x3ffb3e40 0x400821f2:0x3ffb3e60 0x4008cc76:0x3ffb3f20 0x4008cc76:0x3ffb3f40 0x4008d098:0x3ffb3fc0 0x4008d2a0:0x3ffb3fe0 0x400821f2:0x3ffb4000 0x4008cc76:0x3ffb40c0 0x4008cc76:0x3ffb40e0 0x4008d098:0x3ffb4160 0x4008d2a0:0x3ffb4180 0x400821f2:0x3ffb41a0 0x4008cc76:0x3ffb4260 0x4008cc76:0x3ffb4280 0x4008d098:0x3ffb4300 0x4008d2a0:0x3ffb4320 0x400821f2:0x3ffb4340 0x4008cc76:0x3ffb4400
0x4008cc79: panicPutStr at /home/pod/esp/esp-idf/components/esp32/./panic.c:648

0x4008cf46: commonErrorHandler_dump at /home/pod/esp/esp-idf/components/esp32/./panic.c:648

0x4008d098: commonErrorHandler at /home/pod/esp/esp-idf/components/esp32/./panic.c:648

0x4008d2a0: xt_unhandled_exception at /home/pod/esp/esp-idf/components/esp32/./panic.c:648

0x400821f2: _xt_user_exc at /home/pod/esp/esp-idf/components/freertos/./xtensa_vectors.S:718

0x4008cc76: panicPutStr at /home/pod/esp/esp-idf/components/esp32/./panic.c:648

0x4008cc76: panicPutStr at /home/pod/esp/esp-idf/components/esp32/./panic.c:648

......................................................continues

In my application I communicates to MQTT server in cloud. I have basically five tasks running, first is aws_iot_task which is almost same as given in espressif eg. code, second jsonParser task and third jsonStringifyTask. Both this tasks communicate to aws_iot_task through queues.

From the monitor log I see that the app seems to crash once jsonParser Task receives message through queue and it starts processing it.

Please help to to debug this issues, How can I exactly know what is the entry point of breakage.

mimansamaheshwari, 2018-09-18:
I am using MCP23017 IO expander with  Esp32 for performing certain tasks. I am using the standard Adafruit  MCP23017 library on Arduino IDE with a little modifications needed for esp32. Everything was working fine but from last few days I am facing the problem of esp reset. The following message appears on Serial Monitor

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x400d0d63  PS      : 0x00060930  A0      : 0x800d156d  A1      : 0x3ffe3b60  
A2      : 0x3ffc228c  A3      : 0x00000000  A4      : 0x0000000d  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00004e20  A8      : 0x800d0d5d  A9      : 0x3ffe3b40  
A10     : 0x3ffc228c  A11     : 0x0000000c  A12     : 0x0000000d  A13     : 0x000186a0  
A14     : 0x7ff00000  A15     : 0x00000000  SAR     : 0x0000000d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x400d3b74  LEND    : 0x400d3b93  LCOUNT  : 0x00000000  

Backtrace: 0x400d0d63:0x3ffe3b60 0x400d156a:0x3ffe3b80 0x400d1719:0x3ffe3ba0 0x400d175a:0x3ffe3bc0 0x400d63d7:0x3ffe3be0 0x40082e2d:0x3ffe3c00 0x40082fed:0x3ffe3c30 0x400788a6:0x3ffe3c50 0x4007890d:0x3ffe3c80 0x40078a52:0x3ffe3cb0 0x40078b7a:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Can any one tell me proper reason for this problem or the way how to troubleshoot this?

chegewara, 2018-09-18:
try to use `gdb` or `make monitor` to see what this means and what is causing app crash:
```
Backtrace: 0x400d0d63:0x3ffe3b60 0x400d156a:0x3ffe3b80 0x400d1719:0x3ffe3ba0 0x400d175a:0x3ffe3bc0 0x400d63d7:0x3ffe3be0 0x40082e2d:0x3ffe3c00 0x40082fed:0x3ffe3c30 0x400788a6:0x3ffe3c50 0x4007890d:0x3ffe3c80 0x40078a52:0x3ffe3cb0 0x40078b7a:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20
```

umer-ilyas, 2020-03-02:
I am getting following error when I run my code.

Guru Meditation Error: Core 1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC : 0x15383520 PS : 0x00060c30 A0 : 0x800dfcac A1 : 0x3fff7590
A2 : 0x00000000 A3 : 0x3ffb965c A4 : 0x15383520 A5 : 0x6c000001
A6 : 0x003fff74 A7 : 0x60000000 A8 : 0x800e0c11 A9 : 0x3fff7460
A10 : 0x3fff7590 A11 : 0x312e3738 A12 : 0x00000115 A13 : 0x3fff746c
A14 : 0x00000000 A15 : 0x3fff7560 SAR : 0x00000008 EXCCAUSE: 0x00000014
EXCVADDR: 0x15383520 LBEG : 0x4000c349 LEND : 0x4000c36b LCOUNT : 0x00000000

ELF file SHA256:

no backtrace

KonstantinKondrashov, 2020-03-10:
Hi @umer-ilyas!
Your case is not usual because the panic handler was not able to print `ELF file SHA256` could you describe in which case do you get that error? That kind of error can mean that cash is disabled while the panic handler is trying to get the `ELF file SHA256` value.
I provide a patch [7529.zip](https://github.com/espressif/esp-idf/files/4312441/7529.zip) to fix this case. Could you give feedback this helps or not? (with this patch you able to see backtrace and print `ELF file SHA256`).
Thanks

Alvin1Zhang, 2020-06-10:
@umer-ilyas Thanks for sharing the information, would you please help share if any updates for the issue? Thanks.

o-marshmallow, 2021-11-17:
@umer-ilyas Please open a new issue if still relevant. I am closing this issue.

closed 2021-11-17 by o-marshmallow