espressif/esp-idf #1855: removing "phy: phy_version: 383.0, 79a622c, Jan 30 2018, 15:38:06, 0, 0" debug

davctv, 2018-04-13:
Hi,
I'm not able to remove the 

`phy: phy_version: 383.0, 79a622c, Jan 30 2018, 15:38:06, 0, 0`

information debug. It seems in libphy.a library. This debug is always printed even if all the logs are disabled at very startup of app_main() using function

`esp_log_level_set("*", ESP_LOG_NONE);`

Can you please remove it?

Thanks.

davctv, 2018-05-22:
Hi @FayeY,

Is there any update regarding this issue?

Thanks

TimXia, 2018-06-07:
@davctv You can substitute function `in phy_printf(const char* format, ...)` with an empty function in components/esp32/lib_printf.c to solve your problem.

davctv, 2018-06-08:
@TimXia Thank you. I will do a test and let you know.

davctv, 2018-06-11:
@TimXia your suggestion worked. I close the issue. Thanks

gunarkroeger, 2019-10-22:
This should not be closed i think, and should be a Bug. Reproduced with esp-idf-v3.3

Alvin1Zhang, 2020-12-16:
We have a fix under internal reviewing and the issue will close automatically once the fix is available on GitHub, thanks.

closed 2021-01-05 by espressif-bot