espressif/esp-idf #1859: GPIO 25 and 26 interrupt triggered when reading from adc1

trad0v, 2018-04-14:
Hello,
In this code i'm reading the voltage of GPIO 34, at the same time i made interrupts on pulled up  GPIOs 25 and 26. The output on the serial port is the following 

> 116 mV
> GPIO[25] intr, val: 1
> GPIO[26] intr, val: 1
> 135 mV
> GPIO[25] intr, val: 1
> GPIO[26] intr, val: 1
> 136 mV
> GPIO[25] intr, val: 1
> GPIO[26] intr, val: 1
> 131 mV
> GPIO[25] intr, val: 1
> GPIO[26] intr, val: 1
> 124 mV
> GPIO[25] intr, val: 1
> GPIO[26] intr, val: 1
So interrupts are triggering on both GPIOs when the ADC is reading, if i disable the adc readings, no interrupt happens.

```
#include <stdio.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "driver/gpio.h"
#include "driver/adc.h"
#include "esp_system.h"
#include "esp_adc_cal.h"

#define V_REF   1100
#define ADC1_TEST_CHANNEL (ADC1_CHANNEL_6)      //GPIO 34

#define GPIO_INPUT_IO_0     25
#define GPIO_INPUT_IO_1     26
#define GPIO_INPUT_PIN_SEL  ((1ULL<<GPIO_INPUT_IO_0) | (1ULL<<GPIO_INPUT_IO_1))
#define ESP_INTR_FLAG_DEFAULT 0
static xQueueHandle gpio_evt_queue = NULL;
static void IRAM_ATTR gpio_isr_handler(void* arg)
{
    uint32_t gpio_num = (uint32_t) arg;
    xQueueSendFromISR(gpio_evt_queue, &gpio_num, NULL);
}
static void gpio_task_example(void* arg)
{
    uint32_t io_num;
    for(;;) {
        if(xQueueReceive(gpio_evt_queue, &io_num, portMAX_DELAY)) {
            printf("GPIO[%d] intr, val: %d\n", io_num, gpio_get_level(io_num));
        }
    }
}
void app_main(void)
{
   esp_adc_cal_characteristics_t characteristics;
    adc1_config_width(ADC_WIDTH_BIT_12);
    adc1_config_channel_atten(ADC1_TEST_CHANNEL, ADC_ATTEN_DB_0);
    esp_adc_cal_get_characteristics(V_REF, ADC_ATTEN_DB_0, ADC_WIDTH_BIT_12, &characteristics);
    gpio_config_t io_conf;

    io_conf.intr_type = GPIO_PIN_INTR_POSEDGE;
    io_conf.pin_bit_mask = GPIO_INPUT_PIN_SEL;
    io_conf.mode = GPIO_MODE_INPUT;
    io_conf.pull_up_en = 1;
    gpio_config(&io_conf);
    gpio_evt_queue = xQueueCreate(10, sizeof(uint32_t));
    xTaskCreate(gpio_task_example, "gpio_task_example", 2048, NULL, 10, NULL);
    gpio_install_isr_service(ESP_INTR_FLAG_DEFAULT);
    gpio_isr_handler_add(GPIO_INPUT_IO_0, gpio_isr_handler, (void*) GPIO_INPUT_IO_0);
    gpio_isr_handler_add(GPIO_INPUT_IO_1, gpio_isr_handler, (void*) GPIO_INPUT_IO_1);

    uint32_t voltage;
    while(1){
        voltage = adc1_to_voltage(ADC1_TEST_CHANNEL, &characteristics);
        printf("%d mV\n",voltage);
        vTaskDelay(pdMS_TO_TICKS(1000));
    }

}
```

koobest, 2018-04-16:
Hi @trad0v, thanks for you feedback. I have reproduced this issue with your code. I found that if initialize ADC after the configuation of the GPIO can solve this issue. I will replay to you if I find the main reasion.
thanks !!

trad0v, 2018-04-16:
Hello @koobest , thank you for your reply. I tried to initialize the ADC after the GPIO configuration, that didn't solve the issue. Do i have to pull-down the pins ?

`   gpio_config_t io_conf;
   io_conf.intr_type = GPIO_PIN_INTR_POSEDGE;
   io_conf.pin_bit_mask = GPIO_INPUT_PIN_SEL;
   io_conf.mode = GPIO_MODE_INPUT;
   io_conf.pull_up_en = 1;
   gpio_config(&io_conf);

   gpio_evt_queue = xQueueCreate(10, sizeof(uint32_t));
   xTaskCreate(gpio_task_example, "gpio_task_example", 2048, NULL, 10, NULL);
   gpio_install_isr_service(ESP_INTR_FLAG_DEFAULT);
   gpio_isr_handler_add(GPIO_INPUT_IO_0, gpio_isr_handler, (void*) GPIO_INPUT_IO_0);
   gpio_isr_handler_add(GPIO_INPUT_IO_1, gpio_isr_handler, (void*) GPIO_INPUT_IO_1);

   esp_adc_cal_characteristics_t characteristics;
   adc1_config_width(ADC_WIDTH_BIT_12);
   adc1_config_channel_atten(ADC1_TEST_CHANNEL, ADC_ATTEN_DB_0);
   esp_adc_cal_get_characteristics(V_REF, ADC_ATTEN_DB_0, ADC_WIDTH_BIT_12, &characteristics);

    uint32_t voltage;
    while(1){
        voltage = adc1_to_voltage(ADC1_TEST_CHANNEL, &characteristics);
        printf("%d mV\n",voltage);
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
`

trad0v, 2018-04-16:
@koobest It worked now! thanks alot!!!
It's a strange issue, i think its related to the DAC.
Thanks again!
Code:
```C
 gpio_config_t io_conf;
   io_conf.intr_type = GPIO_PIN_INTR_POSEDGE;
   io_conf.pin_bit_mask = GPIO_INPUT_PIN_SEL;
   io_conf.mode = GPIO_MODE_INPUT;
   io_conf.pull_up_en =0;
   io_conf.pull_down_en =1;
   gpio_config(&io_conf);

   esp_adc_cal_characteristics_t characteristics;
   adc1_config_width(ADC_WIDTH_BIT_12);
   adc1_config_channel_atten(ADC1_TEST_CHANNEL, ADC_ATTEN_DB_0);
   esp_adc_cal_get_characteristics(V_REF, ADC_ATTEN_DB_0, ADC_WIDTH_BIT_12, &characteristics);

   gpio_evt_queue = xQueueCreate(10, sizeof(uint32_t));
   xTaskCreate(gpio_task_example, "gpio_task_example", 2048, NULL, 10, NULL);
   gpio_install_isr_service(ESP_INTR_FLAG_DEFAULT);
   gpio_isr_handler_add(GPIO_INPUT_IO_0, gpio_isr_handler, (void*) GPIO_INPUT_IO_0);
   gpio_isr_handler_add(GPIO_INPUT_IO_1, gpio_isr_handler, (void*) GPIO_INPUT_IO_1);

    uint32_t voltage;
    while(1){
        voltage = adc1_to_voltage(ADC1_TEST_CHANNEL, &characteristics);
        printf("%d mV\n",voltage);
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
```

trad0v, 2018-04-16:
@koobest All worked when i put the provided code to close DAC. Thank you!

closed 2018-04-16 by trad0v