espressif/esp-idf #1862: spi_master: Guru Meditation Error

jjfretired, 2018-04-15:
I copied an image from loboris ESP32_TFT_library and placed it in main under examples/peripherals/spi_master.  I rename the installed image.jpg and rename test4.jpg to image.jpg.  Did a make flash monitor and it went into a panic loop.  I'm using th wrover kit V3 with the ILI9341 monitor and downloaded esp-idf today.  I'm running on a ubuntu system.
 Any help in what the causing this. or how to fix it.

LOG of make flash monitor

Flashing binaries to serial port /dev/ttyUSB1 (app at offset 0x10000)...
esptool.py v2.3.1
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 19680 bytes to 11639...

Writing at 0x00001000... (100 %)
Wrote 19680 bytes (11639 compressed) at 0x00001000 in 1.1 seconds (effective 146.9 kbit/s)...
Hash of data verified.
Compressed 215600 bytes to 129576...

Writing at 0x00010000... (12 %)
Writing at 0x00014000... (25 %)
Writing at 0x00018000... (37 %)
Writing at 0x0001c000... (50 %)
Writing at 0x00020000... (62 %)
Writing at 0x00024000... (75 %)
Writing at 0x00028000... (87 %)
Writing at 0x0002c000... (100 %)
Wrote 215600 bytes (129576 compressed) at 0x00010000 in 11.7 seconds (effective 147.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 138...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (138 compressed) at 0x00008000 in 0.0 seconds (effective 756.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
MONITOR
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5648
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:13936
entry 0x40078fd8
[0;32mI (30) boot: ESP-IDF v3.1-dev-743-g6c44fc7-dirty 2nd stage bootloader[0m
[0;32mI (30) boot: compile time 15:09:05[0m
[0;32mI (33) boot: Enabling RNG early entropy source...[0m
[0;32mI (37) boot: SPI Speed      : 40MHz[0m
[0;32mI (41) boot: SPI Mode       : DIO[0m
[0;32mI (45) boot: SPI Flash Size : 4MB[0m
[0;32mI (49) boot: Partition Table:[0m
[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (60) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (67) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (75) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (82) boot:  3 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (90) boot:  4 ota_0            OTA app          00 10 00110000 00100000[0m
[0;32mI (97) boot:  5 ota_1            OTA app          00 11 00210000 00100000[0m
[0;32mI (105) boot: End of partition table[0m
[0;32mI (109) boot: Defaulting to factory image[0m
[0;32mI (113) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x13b0c ( 80652) map[0m
[0;32mI (151) esp_image: segment 1: paddr=0x00023b34 vaddr=0x3ffb0000 size=0x024bc (  9404) load[0m
[0;32mI (155) esp_image: segment 2: paddr=0x00025ff8 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (158) esp_image: segment 3: paddr=0x00026400 vaddr=0x40080400 size=0x09128 ( 37160) load[0m
[0;32mI (182) esp_image: segment 4: paddr=0x0002f530 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (182) esp_image: segment 5: paddr=0x0002f538 vaddr=0x00000000 size=0x00ad8 (  2776) [0m
[0;32mI (189) esp_image: segment 6: paddr=0x00030018 vaddr=0x400d0018 size=0x149f4 ( 84468) map[0m
[0;32mI (232) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (232) boot: Disabling RNG early entropy source...[0m
[0;32mI (233) cpu_start: Pro cpu up.[0m
[0;32mI (236) cpu_start: Starting app cpu, entry point is 0x40080e80[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (247) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (254) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (260) heap_init: At 3FFB3228 len 0002CDD8 (179 KiB): DRAM[0m
[0;32mI (266) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (272) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (279) heap_init: At 40089528 len 00016AD8 (90 KiB): IRAM[0m
[0;32mI (285) cpu_start: Pro cpu start user code[0m
[0;32mI (303) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
LCD ID: 00000000
ILI9341 detected.
LCD ILI9341 initialization.
Guru Meditation Error: Core  0 panic'ed (StoreProhibited)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x400e445f  PS      : 0x00060b30  A0      : 0x80061b1e  A1      : 0x3ffb4dc0  
A2      : 0x3ffb4e60  A3      : 0x3ffe1a00  A4      : 0x3ffb4de0  A5      : 0x3ff9ba04  
A6      : 0x400e4414  A7      : 0x00000010  A8      : 0x00fff5ff  A9      : 0x09cd09cd  
A10     : 0x00000000  A11     : 0x00000101  A12     : 0x3ffb4edc  A13     : 0x3ffe19b1  
A14     : 0x00000001  A15     : 0x3ffe19b4  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x09cd09cd  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x400e445f:0x3ffb4dc0 0x40061b1b:0x3ffb4de0 0x400d2920:0x3ffb4e60 0x400d27ee:0x3ffb4f10 0x400d2606:0x3ffb4f30 0x400d0b1e:0x3ffb4f90

Rebooting.

jjfretired, 2018-04-16:
Output not in tee file but on screen when I reflashed/
uru Meditation Error: Core  0 panic'ed (StoreProhibited)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x400e4483  PS      : 0x00060b30  A0      : 0x80061b1e  A1      : 0x3ffb4dc0
0x400e4483: outfunc at /home/jjf/spi_master/main/decode_image.c:74 (discriminator 3)

A2      : 0x3ffb4e60  A3      : 0x3ffe1a00  A4      : 0x3ffb4de0  A5      : 0x3ff9ba04
A6      : 0x400e4438  A7      : 0x00000010  A8      : 0x00fff5ff  A9      : 0x09cd09cd
0x400e4438: outfunc at /home/jjf/spi_master/main/decode_image.c:62

A10     : 0x00000000  A11     : 0x00000101  A12     : 0x3ffb4edc  A13     : 0x3ffe19b1
A14     : 0x00000001  A15     : 0x3ffe19b4  SAR     : 0x00000000  EXCCAUSE: 0x0000001d
EXCVADDR: 0x09cd09cd  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000

Backtrace: 0x400e4483:0x3ffb4dc0 0x40061b1b:0x3ffb4de0 0x400d2944:0x3ffb4e60 0x400d2812:0x3ffb4f10 0x400d262a:0x3ffb4f30 0x400d0b26:0x3ffb4f90
0x400e4483: outfunc at /home/jjf/spi_master/main/decode_image.c:74 (discriminator 3)

0x400d2944: decode_image at /home/jjf/spi_master/main/decode_image.c:133

0x400d2812: pretty_effect_init at /home/jjf/spi_master/main/pretty_effect.c:60

0x400d262a: app_main at /home/jjf/spi_master/main/spi_master_example_main.c:362

0x400d0b26: main_task at /home/jjf/esp/esp-idf/components/esp32/cpu_start.c:467

negativekelvin, 2018-04-17:
https://github.com/espressif/esp-idf/blob/6c44fc70bc374b29b079115a5b3551edc9f48be9/examples/peripherals/spi_master/main/decode_image.c#L30-L33

FayeY, 2018-04-24:
Hi @jjfretired , did the above suggestion help you resolve the issue?

jjfretired, 2018-04-25:
There is no #define for the values in the code or .h files.  The main code has 320 and 240 code throughout the module.  I change the malloc which had 320 to 336 but it also failed.  I do know from putting printfs in pretty_effects.c and decode_image.c that the failure is in outfunc of decode_image.c.  I'm working on getting gdb functional so that i can try to debug it.

jjfretired, 2018-04-25:
I stand corrected, there are #defines in decode_image.c and I did change these earlier but the failure still occurred.

jjfretired, 2018-04-29:
Closing for now and will reopen at another time with more information?

closed 2018-04-29 by jjfretired