espressif/esp-idf #1864: Light sleep stalls (blocks) at random intervals and is disabling watchdogs

Electrozest, 2018-04-16:
Calling esp_light_sleep_start() with less than one second delay set, and calling the function every second, the esp_light_sleep_start() function stalls (blocks) after 1-3 hours and blocks.

My code is enabling the hardware WDT at startup so I would have expected a WDT reset following the blocking condition, however the hardware WDT was failing to act. On inspection of the code I realise that unexpectedly the esp_light_sleep_start() is enabling and disabling the hardware WDT. Unfortunately on exit the function leaves the hardware watchdog timer disabled!!

There are two critical issues:
i) esp_light_sleep_start() blocks at random
ii) esp_light_sleep_start() is disabling the hardware WDT

Until these issues are resolved the ESP32 is not suitable for remote operation if light sleep mode is required.

This issue is also related to: https://github.com/espressif/arduino-esp32/issues/908

igrr, 2018-04-20:
Hi @Electrozest, we have fixes for light sleep blocking at random in review at the moment. Should be merged in the next few days.

Regarding hardware WDT, light sleep uses RTC watchdog. RTC watchdog is not supposed to be used by applications directly, please use Timer Group watchdogs (which are described in the TRM).

igrr, 2018-04-20:
> configure light sleep to wake on additional interrupts related to task and/or interrupt timer watchdogs

It is not possible to wake up from light sleep based on peripheral interrupts, since the peripherals are clock gated in light sleep. The only wakeup methods are the ones in RTC domain, and are listed in IDF sleep modes documentation.

FreeRTOS uses Timer Group 0 and Timer Group 1 watchdogs for task and interrupt watchdog functionality.

Electrozest, 2018-05-22:
Following the commit for this issue, I am observing a new issue.

Light sleep on its own is observed to function without stalling.

However any light sleep operations after a single deep sleep operation (and subsequent restart) are stalling forever.

igrr, 2018-05-22:
@Electrozest Failure to exit light sleep after going through deep sleep has been fixed in https://github.com/espressif/esp-idf/commit/ac623a97561929750c8cdcc8d679b21d288a2ab1. Please update to latest master and check. If you are still seeing the issue, please share the code snippet to reproduce and your sdkconfig file. For reference, our test case for light sleep and deep sleep is here: https://github.com/espressif/esp-idf/blob/92c469b5993b001a21970f0e58b25acd3f375807/components/esp32/test/test_sleep.c#L47-L52. Running it multiple times (i.e. light sleep - deep sleep - light sleep - deep sleep - ...) seems to work.

Electrozest, 2018-05-22:
I have updated to my esp-idf to master and recompiled. The low power sleep is still failing after deep sleep. Note that I last updated last week, which was after your commit, but now have latest. Note that the first low power sleep after deep sleep is ok. It is the subsequent low power sleep that locks.

So a sequence such as:
light sleep - light sleep - light sleep - light sleep - deep sleep - light sleep - light sleep - light sleep

igrr, 2018-05-22:
Just to check, when you say "low power sleep", do you mean light sleep?

Electrozest, 2018-05-22:
Yes.

igrr, 2018-05-22:
I've run the following test, which worked (for a minute or so):

```c++
#include "esp_sleep.h"

void app_main()
{
    esp_sleep_enable_timer_wakeup(100000);

    esp_light_sleep_start();
    esp_light_sleep_start();
    esp_light_sleep_start();
    esp_light_sleep_start();
    esp_light_sleep_start();

    esp_deep_sleep_start();
}
```

Is this representative of the case you are testing?

Electrozest, 2018-05-22:
Body of app_main is similar to below. 
`
  vTaskDelay(1000);
  esp_sleep_enable_timer_wakeup(1000000);
  esp_light_sleep_start();
  esp_light_sleep_start();
  esp_light_sleep_start();
  esp_light_sleep_start();
  esp_light_sleep_start();
  esp_sleep_enable_timer_wakeup(5000000);
  esp_deep_sleep_start();
`

I notice now that I have updated to latest ESP-IDF master and recompiled that my wifi client code sends TCPIP, but my PC server no longer receives any packets. Identical hardware unit with same WiFi code prior to ESP-IDF update from this morning still transmits packet. Looks like I need to revert to older version of ESP-IDF as my application requires WiFi and TCPIP.

Electrozest, 2018-05-22:
Confirmed, I have tried switching to/from esp-idf master and esp-idf from mid April and WiFiClient TCPIP using client.connect() and client.write() no longer sends packets even though there are no error messages from the tcpip client, but works correctly with esp-idf from mid April.

igrr, 2018-05-23:
> After boot, I switch from internal RTC clock to external 32.768kHz XTAL CLK

Sorry, at the moment there is no supported way in ESP-IDF to do that. You need to decide whether 32k XTAL is used or not at build time.

Also I'm not sure how this fact is related to the issue being discussed. If you don't switch to 32k XTAL at run time, do you observe the issue (board locks up after light sleep)?

Can you please confirm that when you run *exactly* the following code, you see that the board locks up after light sleep?

```c++
#include "esp_sleep.h"
void app_main() {
  esp_sleep_enable_timer_wakeup(1000000);
  esp_light_sleep_start();
  esp_light_sleep_start();
  esp_light_sleep_start();
  esp_light_sleep_start();
  esp_light_sleep_start();
  esp_sleep_enable_timer_wakeup(5000000);
  esp_deep_sleep_start();
}
```

If this exact code does not reproduce the issue on your side, we need to converge on some code which does, and which you can share.

Electrozest, 2018-05-23:
If I don't switch to 32k XTAL at run time with just the above code, I am not observing the issue.
If I switch to 32k XTAL once at startup, the esp_light_sleep_start() is also ok. 

It is when the deep sleep triggers restart, switching to 32k XTAL works, but subsequent  esp_light_sleep_start() either runs very slowly or blocks.