espressif/esp-idf #1865: On the issue of sending broadcast packets at the AP end

Gaoqianjin, 2018-04-16:
hi,
 Problem: AP as a UDP server,sends a broadcast packet with a size of 1440 bytes, 
which is sent every 1 milliseconds.   it is found that station and AP can't create links
(The SYSTEM_EVENT_STA_GOT_IP event did not succeed )

[send_log_1ms.txt](https://github.com/espressif/esp-idf/files/1914587/send_log_1ms.txt)

 see attschments send_log_1ms.rar,recv_log_1ms.txt.
（The fail_count in send_log_1ms.rar is transmitted by Not enough space, and each pixmap_index sends 80 broadcast packets. ）
[recv_log_1ms.txt](https://github.com/espressif/esp-idf/files/1914583/recv_log_1ms.txt)
[Uploading send_log_1ms.txt…]()

By grabbing data packets, it is found that AP rarely sends Beacon,  Maybe AP does not have time to send Beacon.
In this case,Beacon seldom leads to unsuccessful connection, and failure of connection results in buffer overflow. Is it justified by what I say? 
Is there a solution to ensure high speed and massive data transmission? 
For example, in this case, AP can send beacon normally. 
I don't want to reduce sending speed and reduce the size of each packet. 
However, station sent the broadcast packet to AP to receive the solution, which is much better, but this is not what I want.

Gaoqianjin, 2018-04-16:
A pixmap is made up of 80 broadcast packets,The number of fail_count in send_log_1ms.txt is the number of transmission failures due to Not enough space.

liuzfesp, 2018-05-08:
@Gaoqianjin sorry for late response, could you help to check:
1.  If AP doesn't send UDP broadcast packets, can the sta connect to AP?
2. Does the sta use the correct password?

liuzfesp, 2018-05-28:
Hi @Gaoqianjin, does this issue still exist?

closed 2018-06-13 by FayeY