espressif/esp-idf #1866: spi_bus_add_device SPI_CECK

tcmichals, 2018-04-16:
There is an error with:

SPI_CHECK(dev_config->spics_io_num < 0 || GPIO_IS_VALID_OUTPUT_GPIO(dev_config->spics_io_num), "spics pin invalid", ESP_ERR_INVALID_ARG);

It should be:
SPI_CHECK(dev_config->spics_io_num > 0 && GPIO_IS_VALID_OUTPUT_GPIO(dev_config->spics_io_num), "spics pin invalid", ESP_ERR_INVALID_ARG);

The issue is dev_config->spics_io_num < 0 if the pin is defined and it does not check the validity of the pin.

negativekelvin, 2018-04-16:
The intent is that either the value is either -1 or a valid gpio output number

ginkgm, 2018-04-17:
agree with @negativekelvin , -1 is also a valid value, SPI will work without CS output in this case.

tcmichals, 2018-04-17:
The issue is, IO35, it is only an input, so, the dev_config->spics_io_num < 0  passes and  GPIO_IS_VALID_OUTPUT_GPIO(dev_config->spics_io_num) does not. 

If the user does not have a logic analyzer or some way to look at the CS, then how is to know something is wrong?

negativekelvin, 2018-04-18:
35 is less than zero?

FayeY, 2018-04-27:
Closing this, please feel free to reopen it if your problem is still unsolved. Thanks.

closed 2018-04-27 by FayeY