espressif/esp-idf #1872: Is it possible to disable the Wi-Fi re-transmit?

lubingxian, 2018-04-18:
I found that ESP32 WiFi driver will re-transmit the unicast packet.
Is it possible to disable the re-transmit or change the max re-transmit time?

liuzfesp, 2018-04-19:
@lubingxian currently we don't have such kind of API to set the re-transmit counter and max re-transmit time, it's hard-coded in WiFi driver.

lubingxian, 2018-04-19:
@liuzfesp Many thanks for your reply!
Will this function be easy to implement? Since we really need this function in our research project, could you make a prototype for us?

lubingxian, 2018-04-19:
@liuzfesp What we need is a WiFi driver without the re-transmit. Thank you!

liuzfesp, 2018-05-08:
@lubingxian I'm afraid that currently we don't have API to disable the WiFi retransmit.

closed 2018-06-22 by FayeY