espressif/esp-idf #1876: std::chrono::steady_clock is NOT monotonic

jemdiggity, 2018-04-20:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Core (if using chip or module): [ESP32-PICO-D4]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
   dfb2247
- Development Env:		[Make]
- Operating System:		[MacOS]
- Power Supply:         [external 3.3V]

## Problem Description
std::chrono::steady_clock is NOT monotonic

toolchain version: xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz
esp-idf branch: release/v3.0

### Expected Behavior
steady_clock should be monotonic

### Actual Behavior
steady_clock isn't monotonic, calling `settimeofday()` affects steady_clock

### Steps to repropduce
If I call this every second, it logs increasing seconds.
```cpp
ESP_LOGI(TAG, "steady_clock %lld", std::chrono::duration_cast<std::chrono::seconds>(std::chrono::steady_clock::now().time_since_epoch()).count());
```

If I call the following to set the system time to 0, steady_clock restarts from zero as well.
```cpp
struct timeval raw_time = {};
int err = settimeofday(&raw_time, NULL);
```

However, esp_timer_get_time() is monotonic.
std::timed_mutex depends on steady_clock.
nghttp2 also depends on steady_clock.

### Code to reproduce this issue

## Debug Logs
```
I (36595) Peripheral: system_clock 36
I (36595) Peripheral: steady_clock 36
I (36595) Peripheral: esp timer 36519
I (37599) Peripheral: system_clock 37
I (37599) Peripheral: steady_clock 37
I (37599) Peripheral: esp timer 37523
I (38599) Peripheral: system_clock 0
I (38599) Peripheral: steady_clock 0
I (38599) Peripheral: esp timer 38523
I (39599) Peripheral: system_clock 1
I (39599) Peripheral: steady_clock 1
I (39599) Peripheral: esp timer 39524
```
## Other items if possible

peteralm80, 2019-04-10:
Any updates on this? Almost  a year later and this still seems to not have been fixed.

igrr, 2019-04-10:
@peteralm80 Hello, this will be fixed soon in the new toolchain. We have validated the fix and are preparing the release.

igrr, 2019-04-12:
Preview release of the toolchain which fixes this issue is now available, see https://www.esp32.com/viewtopic.php?f=10&t=7400&p=31257#p41667.

closed 2019-06-10 by igrr