espressif/esp-idf #1877: AP stops responding (no Ping either)

pglen, 2018-04-20:
## Environment

- Development Kit:      [ESP32-Wrover-Kit AND ESP32-DevKitC|
- Kit version (for WroverKit/PicoKit/DevKitC): v3
- Core (if using chip or module): ESP32-Wrover AND ESP-WROOM32|ESP32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.):  da278163 (reproducible in two prev idf updates)
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:		Make
- Operating System:		Windows  (W7)
- Power Supply:                USB

## Problem Description
  Have a small http server and  rdate server running on the ESP32. Everything works great, except  I get random TCP communication stops. When the com stops, I cannot ping the board either. The board is started with softAP, connects to the internet fine, and serves its home page fine. Rdate responds as expected. The board appears on IP 192.168.4.1. Ping, works fine, however, it randomly stops responding to pings. Could not find a rhyme or reason. Then randomly 'heals' and everything is back to normal, the board responds as expected. All the while the WiFi appears connected.

## Tried:

  Disconnecting the station part from the internet. The softAP still froze randomly. Tried restating the TCP stack. Tried socket timeout. Tried forceARP (described elsewhere). Tried

//Detailed problem description goes here.

 I tried many 'workarounds'. Enabled / Disable soft AP / STA. En / Dis Power management. Different Clients. (The freeze is less frequent with W10)

### Expected Behavior

connectivity

### Actual Behavior

freeze in connectivity

### Steps to repropduce

 ping 

### Code to reproduce this issue

https://github.com/pglen/esp-code

Any freely available HTTPD code will do it.

// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

 No debug log, as the freeze does not produce errors.

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.) 

 Default SDK config, only change is the COM port

zhangyanjiaoesp, 2018-06-04:
@pglen Can I run your code directly? I have downloaded  your  code, and run it on my PC, but it failed.

Alvin1Zhang, 2019-10-12:
@pglen Thanks for reporting and sorry for the very slow turnaround. Would you please help share if any updates for this issue? Thanks.

pglen, 2019-10-12:
This has been largely addressed since ... I am referring to posts about the
power management arrangements, posts about the missing arp broadcasts
because of modem sleep. That being said, it would be nice to see an httpd
example that offers a stable long term AP connectivity.

Thank you for your reply.

Peter

liuzfesp, 2020-07-08:
Hi @pglen we have optimized modem sleep a lot since this issue have been reported, please check.
Since it's a very old issue, we would like to close it first, please free feel to reopen it if you encounter similar issue again.

closed 2020-07-08 by Alvin1Zhang