espressif/esp-idf #1879: xtensa-esp32-elf-gcc  error

zqclzz, 2018-04-23:
Hello， Espressif Engineers:

xtensa-esp32-elf-gcc: error: unrecognized command line option '-mfix-esp32-psram-cache-issue'

I have got the 'make' error, showm above, how to fix issue?

igrr, 2018-04-23:
When you run `xtensa-esp32-elf-gcc --version`, what output do you get?

FayeY, 2018-05-09:
Closing due to lack of response, please feel free to reopen if this is still an issue and you can supply more details.
Thanks.

closed 2018-05-09 by FayeY

manmohan777, 2020-05-26:
I am getting the same error while building.
my xtensa-esp32-elf-gcc version is 8.2.0

igrr, 2020-05-26:
@manmohan777 What is the full output of `xtensa-esp32-elf-gcc --version`, please? This flag was introduced in 2020-r1 toolchain release, you might be using an older version?

andrey0303, 2020-07-02:
I have the same error.

unrecognized command line option '-mfix-esp32-psram-cache-strategy=memw';

xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.

howroyd, 2020-07-24:
Same issue here on release v4.1 after doing a pull and submodule update init & recursive:

```xtensa-esp32-elf-gcc.exe: error: unrecognized command line option '-mfix-esp32-psram-cache-strategy=memw'; did you mean '-mfix-esp32-psram-cache-issue'?
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

E:\Simon\Documents\Code\voip_cpp>xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.```

howroyd, 2020-07-24:
Not sure if it's the right thing to do but setting the following in `menuconfig` seems to have fixed the compile issue and my code appears to be running normally...
```
SPIRAM_CACHE_WORKAROUND(=n)
```

9ae8sdf76, 2021-03-01:
2021 here, and I'm getting a similar error:

```
Problems were encountered while collecting compiler information:
	xtensa-esp32-elf-g++: error: unrecognized command line option '-mfix-esp32-psram-cache-strategy=memw'
	xtensa-esp32-elf-gcc: error: unrecognized command line option '-mfix-esp32-psram-cache-strategy=memw'
```

Version of `xtensa-esp32-elf-gcc`: `xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0`

Setting `-D SPIRAM_CACHE_WORKAROUND=n` did not work.

Can we re-open the ticket?

antmak, 2021-03-02:
@9ae8sdf76 Hi! Please show me the output from commands

```
xtensa-esp32-elf-gcc --version
command -v xtensa-esp32-elf-gcc
grep WORKAROUND sdkconfig
idf.py fullclean && idf.py build
```

9ae8sdf76, 2021-03-02:
**`xtensa-esp32-elf-gcc --version`**
```
xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

**`command -v xtensa-esp32-elf-gcc`**
```
/Users/kstanl27/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
```

---

I'm using the [framework-arduinoespressif32](https://github.com/espressif/arduino-esp32/) package with PlatformIO and CLion for my project. I was asked to use their [latest version](https://github.com/espressif/arduino-esp32/issues/4833#issuecomment-787961308), which when building would generate the error described above.

That said...

I'm not using the `CONFIG_ESP32_DPORT_WORKAROUND` that's by default available in an sdkconfig provided by esp-idf. Adding that to my build flags made no difference. 

Here is the full output from my build, as requested:

```
Processing esp32dev (build_flags: -g, -D LV_CONF_INCLUDE_SIMPLE, -I ., -I include, -D CONFIG_ESP32_DPORT_WORKAROUND=y, -DWIFI_SSID='"nebakanezer"', -DWIFI_PASS='"2f67478590"'; platform: espressif32; board: esp32dev; framework: arduino; board_build.mcu: esp32; board_build.f_cpu: 240000000L; board_build.f_flash: 80000000L; board_build.flash_mode: dio; lib_archive: False; lib_deps: EEPROM, SPI, Wire, adafruit/Adafruit BusIO@^1.7.1, adafruit/Adafruit GFX Library@^1.10.4, adafruit/Adafruit RA8875@^1.4.0, adafruit/Adafruit TouchScreen@^1.1.1, adafruit/RTClib @ ^1.12.4, lvgl/lvgl @ ^7.9.1, m5ez/ezTime @ ^0.8.3, nrwiersma/ConfigManager @ ^2.0.1, sourcesimian/uICAL @ ^0.0.1; platform_packages: platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/archive/idf-release/v4.2.zip; monitor_port: /dev/cu.usbserial-0001; monitor_speed: 115200; monitor_filters: time, colorize, default, printable, esp32_exception_decoder; src_filter: +<*>)
--------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.0.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 0.0.0 (https://github.com/espressif/arduino-esp32/archive/idf-release/v4.2.zip) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <EEPROM> 1.0.3 (/Users/kstanl27/.platformio/packages/framework-arduinoespressif32/libraries/EEPROM)
|-- <SPI> 1.0 (/Users/kstanl27/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|-- <Wire> 1.0.1 (/Users/kstanl27/.platformio/packages/framework-arduinoespressif32/libraries/Wire)
|-- <Adafruit BusIO> 1.7.2 (/Users/kstanl27/Development/dbuddy/.pio/libdeps/esp32dev/Adafruit BusIO)
|   |-- <Wire> 1.0.1 (/Users/kstanl27/.platformio/packages/framework-arduinoespressif32/libraries/Wire)
|   |-- <SPI> 1.0 (/Users/kstanl27/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|-- <Adafruit GFX Library> 1.10.5 (/Users/kstanl27/Development/dbuddy/.pio/libdeps/esp32dev/Adafruit GFX Library)
|   |-- <Adafruit BusIO> 1.7.2 (/Users/kstanl27/Development/dbuddy/.pio/libdeps/esp32dev/Adafruit BusIO)
|   |   |-- <Wire> 1.0.1 (/Users/kstanl27/.platformio/packages/framework-arduinoespressif32/libraries/Wire)
|   |   |-- <SPI> 1.0 (/Users/kstanl27/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|   |-- <Wire> 1.0.1 (/Users/kstanl27/.platformio/packages/framework-arduinoespressif32/libraries/Wire)
|   |-- <SPI> 1.0 (/Users/kstanl27/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|-- <Adafruit RA8875> 1.4.0 (/Users/kstanl27/Development/dbuddy/.pio/libdeps/esp32dev/Adafruit RA8875)
|   |-- <Adafruit GFX Library> 1.10.5 (/Users/kstanl27/Development/dbuddy/.pio/libdeps/esp32dev/Adafruit GFX Library)
|   |   |-- <Adafruit BusIO> 1.7.2 (/Users/kstanl27/Development/dbuddy/.pio/libdeps/esp32dev/Adafruit BusIO)
|   |   |   |-- <Wire> 1.0.1 (/Users/kstanl27/.platformio/packages/framework-arduinoespressif32/libraries/Wire)
|   |   |   |-- <SPI> 1.0 (/Users/kstanl27/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|   |   |-- <Wire> 1.0.1 (/Users/kstanl27/.platformio/packages/framework-arduinoespressif32/libraries/Wire)
|   |   |-- <SPI> 1.0 (/Users/kstanl27/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|   |-- <EEPROM> 1.0.3 (/Users/kstanl27/.platformio/packages/framework-arduinoespressif32/libraries/EEPROM)
|   |-- <SPI> 1.0 (/Users/kstanl27/.platformio/packages/framework-arduinoespressif32/libraries/SPI)
|-- <Adafruit TouchScreen> 1.1.1 (/Users/kstanl27/Development/dbuddy/.pio/libdeps/esp32dev/Adafruit TouchScreen)
|-- <RTClib> 1.12.5 (/Users/kstanl27/Development/dbuddy/.pio/libdeps/esp32dev/RTClib)
|   |-- <Wire> 1.0.1 (/Users/kstanl27/.platformio/packages/framework-arduinoespressif32/libraries/Wire)
|-- <lvgl> 7.10.1 (/Users/kstanl27/Development/dbuddy/.pio/libdeps/esp32dev/lvgl)
[... cut ...]
============================================ [FAILED] Took 2.59 seconds ============================================

Environment     Status    Duration
--------------  --------  ------------
esp32dev        FAILED    00:00:02.588
esp32dev:debug  IGNORED
====================================== 1 failed, 0 succeeded in 00:00:02.588 ======================================
```

It's late where I'm at, so I will get back in another comment after I create a bonafide IDF project with my code and see what happens there.

antmak, 2021-03-04:
@9ae8sdf76 Sorry for delayed. It's a bit weird, xtensa-esp32-elf-g++ has the option and it works with that. But I don't have arduino installation. I'd suppose, the real xtensa-esp32-elf-g++ that actually runs is not `/Users/kstanl27/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++`, a different version. Do you have other toolchains installed in some other directories? Maybe in opt/ or in others than ~/.espressif dirs.

Also I suggest to create a separate issue here, because the original one has other reasons, and was closed.