espressif/esp-idf #1881: build broken because errors in esp-idf

ataweg, 2018-04-24:
Error is introduced with commit e37ceaed3278859b1d0ddec9a7077c9d16a5d551
commit db5f927087ac0d56712f1eb5ab8fdc6f0df4a6bb compiles fine.
from file C:\Espressif\esp-idf\components\heap\multi_heap_platform.h _#include <freertos/task.h>_ is removed and this caused the build failure.

**Compile log:**
In file included from C:/Espressif/esp-idf/components/heap/multi_heap_poisoning.c:27:0:
C:/Espressif/esp-idf/components/heap/multi_heap_platform.h:66:32: error: unknown type name 'TaskHandle_t'
 #define MULTI_HEAP_BLOCK_OWNER TaskHandle_t task;
                                ^
C:/Espressif/esp-idf/components/heap/multi_heap_poisoning.c:50:5: note: in expansion of macro 'MULTI_HEAP_BLOCK_OWNER'
     MULTI_HEAP_BLOCK_OWNER
     ^
C:/Espressif/esp-idf/components/heap/multi_heap_poisoning.c: In function 'poison_allocated_region':
C:/Espressif/esp-idf/components/heap/multi_heap_platform.h:67:57: error: implicit declaration of function 'xTaskGetCurrentTaskHandle' [-Werror=implicit-function-declaration]
 #define MULTI_HEAP_SET_BLOCK_OWNER(HEAD) (HEAD)->task = xTaskGetCurrentTaskHandle()
                                                         ^
C:/Espressif/esp-idf/components/heap/multi_heap_poisoning.c:71:5: note: in expansion of macro 'MULTI_HEAP_SET_BLOCK_OWNER'
     MULTI_HEAP_SET_BLOCK_OWNER(head);
     ^
C:/Espressif/esp-idf/components/heap/multi_heap_poisoning.c: In function 'multi_heap_get_block_owner':
C:/Espressif/esp-idf/components/heap/multi_heap_platform.h:68:42: warning: return makes pointer from integer without a cast [-Wint-conversion]
 #define MULTI_HEAP_GET_BLOCK_OWNER(HEAD) ((HEAD)->task)
                                          ^
C:/Espressif/esp-idf/components/heap/multi_heap_poisoning.c:284:12: note: in expansion of macro 'MULTI_HEAP_GET_BLOCK_OWNER'
     return MULTI_HEAP_GET_BLOCK_OWNER((poison_head_t*)multi_heap_get_block_address_impl(block));
            ^
cc1.exe: some warnings being treated as errors
make[1]: *** [/c/Espressif/esp-idf/make/component_wrapper.mk:286: multi_heap_poisoning.o] Fehler 1

negativekelvin, 2018-04-24:
https://github.com/espressif/esp-idf/commit/573ea385b433c3c0e0397f15f4f5c1866edad3aa#diff-2d1a7887a6bbb6e6c852ffc8e44b6787

mahavirj, 2018-04-25:
@ataweg Thanks for reporting this, please expect fix for this in some time.

closed 2018-04-26 by igrr