espressif/esp-idf #1884: Advertising time interval

chegewara, 2018-04-25:
Im not sure what kind issue it is, maybe bluetooth or maybe clock. I have 2 devKitC boards and i uploaded the same arduino app on both. This is result, one device advertising interval is short and other one is over 1200 ms. I have this issue for very long time now, but i never tried to compare on 2 esp32 boards. One more important info: after reseting "bad" board advertising intervals are short on both. Tested on 3 android smartphones with different android and result is the same on all.

[image]

anujdeshpande, 2018-04-26:
Can you point to the exact app that you are using? The struct `esp_ble_adv_params_t` has two variables called `adv_int_min` and `adv_int_max`. Maybe it's something to do with those values?

chegewara, 2018-04-27:
Maybe, but why it sometimes good and sometimes has 1200+ ms intervals?

```
min_interval        = 0x20;
max_interval        = 0x40;
```

anujdeshpande, 2018-04-27:
Can you reliably reproduce the issue? 
Please share details (code or snippet) about your setup so that it's possible to reproduce the issue

chegewara, 2018-04-27:
Different server applications based on Neil Kolban's library, arduino and esp-idf applications. Like i said, sometimes its enough to reset esp32 to get expected advertising interval time and sometimes its 1200+ ms. I cant say what part is responsible, bluedroid, clock, or BLE stack. Its something i can reproduce, so i decided to report, when i have more info i will post here.

ArTyGill, 2018-08-29:
Faced the same problem. After trying to change the interval, I constantly get an interval of 1280 milliseconds.

chegewara, 2018-08-29:
Its not the only issue topic about it, but devs are telling its all ok and my test procedure is wrong.

chegewara, 2018-08-29:
@ArTyGill https://github.com/espressif/esp-idf/issues/2185

ArTyGill, 2018-09-09:
The problem was solved with the help of flashing the old bootloader, I do not know what the problem is. I use Arduino (VStudio).

chegewara, 2018-09-15:
Hi @Weijian-Espressif, i dont want to open new issue because it is related to advertising time interval.
I dont know if this is nRF connect issue or esp32 (most likely its nRF connect), but advertising time interval is not properly mapped. What i mean is this advertising opcode:
https://github.com/espressif/esp-idf/blob/892fc4ac3f04fb300c7ac5855b51af940c3e3e65/components/bt/bluedroid/api/include/api/esp_gap_ble_api.h#L187

When i setup esp32 for example with values:
- min interval 0x80
- max interval 0x80
i should have get 80 ms interval. The problem is nRF connect is displaying 160ms. The good new is there is no longer 1200+ advertising with master esp-idf and new nRF connect update (4th september).

chegewara, 2018-09-21:
@Weijian-Espressif Hi, i think i found issue or at least some useful log that can be related to this issue. It seems i have advertising interval 1200+ when this log occurs:
`E (1490) BT_HCI: hci write adv params error 0x30`

I dont know yet what is causing it and i dont know what error 0x30 means but maybe you guys can help me with it.

chegewara, 2018-10-01:
@Weijian-Espressif It seems that even with proper adv interval min and max values, when other advertising parameter is wrong and this error occurs `E (2654) BT_HCI: hci write adv params error 0x30` advertising time is 1200+ ms like on the screenshot from OP. This time i am trying to set adv_type to any value different than `BLE_ADDR_TYPE_PUBLIC`.
Advertising intervals:
```
D (2561) BT_HCI: p_data[13] = 5
D (2564) BT_HCI: p_data[14] = 12
D (2568) BT_HCI: p_data[15] = 20
D (2571) BT_HCI: p_data[16] = 0
D (2574) BT_HCI: p_data[17] = 40
D (2578) BT_HCI: p_data[18] = 0
```

ArTyGill, 2018-10-02:
Something else has changed in the bootloader, since the firmware of the old bootloader helps to fix this problem.

LaurentLouf, 2018-10-05:
@chegewara 

> 
> When its called in correct order then we have correct advertising packet:
> 
Could you elaborate on what is the correct order ? What did you have before and after ? My order is the following : BLEDevice.init, BLEDevice.createServer, BLEServer.setCallbacks, BLEServer.createService, BLEService.start and finally BLEServer.startAdvertising. Does something looks wrong ?

chegewara, 2018-10-05:
> Could you elaborate on what is the correct order ? What did you have before and after ? My order is the following : BLEDevice.init, BLEDevice.createServer, BLEServer.setCallbacks, BLEServer.createService, BLEService.start and finally BLEServer.startAdvertising. Does something looks wrong ?

It looks good. Try to add this one line after init:
```
		BLEDevice::init;
		BLEAdvertising* pAdvertising = BLEDevice::getAdvertising();

```

LaurentLouf, 2018-10-05:
I'm not quite sure it will change anything : 
```
BLEAdvertising* BLEServer::getAdvertising() {
	return &m_bleAdvertising;
}
``` 
And : 
```
void BLEServer::startAdvertising() {
	ESP_LOGD(LOG_TAG, ">> startAdvertising");
	m_bleAdvertising.start();
	ESP_LOGD(LOG_TAG, "<< startAdvertising");
} // startAdvertising
```
So actually, get advertising doesn't do much, so that seems a bit odd that calling it early fixes anything.

chegewara, 2018-10-05:
I understand that, but it works to me. When i call `BLEAdvertising* pAdvertising = BLEDevice::getAdvertising();` after server->start() im having intervals 1200+, when i move that line right after init im having correct intervals.

chegewara, 2018-10-15:
@LaurentLouf I found bug in BLE library and i am fixing it. All function in this sequence needs to be separate with semaphore, because sometimes are called too earlier before previous has been processed.

LaurentLouf, 2018-10-15:
Thanks for investigating ! Do you think you will be able to send a PR soon ?

chegewara, 2018-10-15:
I am working on so many additions, changes and bugfixes so i cant promise. First i am pushing all to my own repo and when will be tested and confirmed that all works i will prepare PR. I can think about preparing some PR with only bug fixes, but again i cant promise anything.

https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test

closed 2018-10-15 by chegewara

chegewara, 2018-10-15:
@Weijian-Espressif I am having weird bt stack behavior. When i dont follow procedure and try to call all advertising functions one after another without waiting for events (it was bug in library), then sometimes i am having problems with time interval 1200+ described in this issue, sometimes there is no issue. But i am having also side effect, i dont have to stop advertising to connect peer device and start advertising again to have multi connect server app.

When i fix that bug and i follow procedure with events, then i dont have problems with advertising interval anymore (which is good) but when i connect peer device without stopping advertising my esp32 is no longer discoverable. 

So i have question, do i have to follow similar procedure like with scan. Which is `start advertise->stop->connect->start advertising` or is it bug? i like option when i dont have to stop advertising but i can live with requirements that i have to stop it.