espressif/esp-idf #1887: Enterprise PEAP connection issue

ali-slim-sst, 2018-04-26:
I'm testing wpa2 enterprise example but the ESP32 is unable to connect I'm setting the EAP method to EAP_PEAP. Below is a log for the AP when the device is trying to connect. I tried different versions of the IDF with no joy.

My laptop and phone can connect to the same SSID using the username and password with no issues.

Any ideas what might be going wrong? why the authentication is failing?
 
[image]

chegewara, 2018-04-26:
```
Supplicant used different EAP type: 3
authentication failed - EAP type: 4
```

FayeY, 2018-05-09:
Hi, did the above suggestion help you resolve the issue?

ali-slim-sst, 2018-05-09:
No it did not I'm using the correct authentication method in the ESP32 firmware. I'm still unable to use wpa2 enterprise, even with the latest update to IDF v3.0. 

Can you please share a test setup where the wpa2 enterprise example was tested?

XinDeng11, 2018-05-09:
Hi, @ali-slim-sst, if you want to use the wpa_enterprise example, I think most important thing is make sure your have correct certificate, because the server and client certificates should be matched so that it can build TLS tunnel after check certificates, if you can use the server.key, server.crt and server.pem in example instead of them in the server, it would be OK, or you can try whether you have client cerificate to instead of them in the example. If you don have client certificate or can't exchange the certificate in server, may be you can mark down esp_wifi_wpa_ent_set_ca_cert() and esp_wifi_sta_wpa2_ent_set_cert_key() so that it will jump certificate check. If it still fail, maybe you can send us the air packet and let me check when it fail

ali-slim-sst, 2018-05-09:
Hi @XinDeng11 , please explain why do i need the certificates if I'm using PEAP? If i remove esp_wifi_wpa_ent_set_ca_cert() this should skip the certifcate check?

XinDeng11, 2018-05-09:
PEAP methds do security check based on certificate and username/password, it need check certificate to build TLS tunnel, and then do MSCHAPv2 check in the TLS tunnel by username and password, because server need decode the pre master secret created by client according to the server key which is in the certificate, and the pre master secret created by client was encoded by public key in the ca certificate, so that why certificate is necessary in PEAP/TTLS/TLS method. Why the certificate should be matched is due to RSA algrithm, there are two private keys and one public key, each private key is in the client certificate and server certificate, and public key is in the certificate, so it need check the ca certificate to make sure client and server are safe. Remove ca cert check and set key will jump this step and build a tunnel directly, but it can't in the TTLS method.

ali-slim-sst, 2018-05-09:
Thanks @XinDeng11 for the explanation. I removed esp_wifi_sta_wpa2_ent_set_ca_cert() and esp_wifi_sta_wpa2_ent_set_cert_key() but I still cannot connect in PEAP mode. Let me know if there's anything else I can try.

Please note that I'm able to connect other devices like my phone and laptop to the same router using PEAP just using username and password.

XinDeng11, 2018-05-09:
If you can offer sniffer log(air packets), we can do a continue check when it failed, because we 100% sure the example is OK when connect to a radius server.

ali-slim-sst, 2018-05-11:
@XinDeng11 I've tried to capture some useful wifi air packets, but I'm not getting anything useful so far, can you please share how you're sniffing the packets? Thanks.

XinDeng11, 2018-05-11:
@ali-slim-sst what's your software and tool use

XinDeng11, 2018-05-11:
If  it that, I just wonder whether you are in the correct channel, because use wireshark in Linux need set channel by command in terminal. If you still can't found useful message, you can try to open the log by menuconfig, the log output option in the components option. If you still fail to get message, you have to wait for the 3.1 version release. We will open source the wpa enterprise code at that time

ali-slim-sst, 2018-05-14:
@XinDeng11 below is debug output. What are reasons 204 and 201?

I (0) cpu_start: App cpu up.
I (332) heap_init: Initializing. RAM available for dynamic allocation:
D (339) heap_init: New heap initialised at 0x3ffae6e0
I (344) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
D (350) heap_init: New heap initialised at 0x3ffb8f28
I (355) heap_init: At 3FFB8F28 len 000270D8 (156 KiB): DRAM
I (361) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (367) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (374) heap_init: New heap initialised at 0x4009154c
I (379) heap_init: At 4009154C len 0000EAB4 (58 KiB): IRAM
I (385) cpu_start: Pro cpu start user code
D (397) clk: RTC_SLOW_CLK calibration value: 3332237
V (71) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (71) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (76) intr_alloc: Connected src 46 to int 2 (cpu 0)
V (81) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (87) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xC0E
D (95) intr_alloc: Connected src 57 to int 3 (cpu 0)
V (100) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (107) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
D (115) intr_alloc: Connected src 24 to int 9 (cpu 0)
I (120) cpu_start: Starting scheduler on PRO CPU.
V (0) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (0) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x40E
D (10) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (10) cpu_start: Starting scheduler on APP CPU.
D (170) heap_init: New heap initialised at 0x3ffe0440
D (170) heap_init: New heap initialised at 0x3ffe4350
V (180) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (180) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (180) intr_alloc: Connected src 16 to int 12 (cpu 0)
D (200) nvs: nvs_flash_init_custom partition=nvs start=9 count=6
D (270) nvs: nvs_open_from_partition misc 1
D (270) nvs: nvs_get_str_or_blob log
I (270) wifi: wifi firmware version: 6450435
I (270) wifi: config NVS flash: enabled
I (270) wifi: config nano formating: disabled
I (280) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (290) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
D (290) nvs: nvs_open_from_partition nvs.net80211 1
D (300) nvs: nvs_get opmode 1
D (300) nvs: nvs_get_str_or_blob sta.ssid
D (310) nvs: nvs_get_str_or_blob sta.mac
D (310) nvs: nvs_get sta.authmode 1
D (310) nvs: nvs_get_str_or_blob sta.pswd
D (320) nvs: nvs_get_str_or_blob sta.pmk
D (320) nvs: nvs_get sta.chan 1
D (330) nvs: nvs_get auto.conn 1
D (330) nvs: nvs_get bssid.set 1
D (330) nvs: nvs_get_str_or_blob sta.bssid
D (340) nvs: nvs_get sta.phym 1
D (340) nvs: nvs_get sta.phybw 1
D (340) nvs: nvs_get_str_or_blob sta.apsw
D (350) nvs: nvs_get_str_or_blob sta.apinfo
D (350) nvs: nvs_get sta.scan_method 1
D (360) nvs: nvs_get sta.sort_method 1
D (360) nvs: nvs_get sta.minrssi 1
D (360) nvs: nvs_get sta.minauth 1
D (370) nvs: nvs_get_str_or_blob ap.ssid
D (370) nvs: nvs_get_str_or_blob ap.mac
D (370) nvs: nvs_get_str_or_blob ap.passwd
D (380) nvs: nvs_get_str_or_blob ap.pmk
D (380) nvs: nvs_get ap.chan 1
D (390) nvs: nvs_get ap.authmode 1
D (390) nvs: nvs_get ap.hidden 1
D (390) nvs: nvs_get ap.max.conn 1
D (400) nvs: nvs_get bcn.interval 2
D (400) nvs: nvs_get ap.phym 1
[... 155 lines cut ...]
I (36530) example: IP:0.0.0.0
I (36530) example: MASK:0.0.0.0
I (36530) example: GW:0.0.0.0
I (36530) example: ~~~~~~~~~~~
I (38530) example: ~~~~~~~~~~~
I (38530) example: IP:0.0.0.0
I (38530) example: MASK:0.0.0.0
I (38530) example: GW:0.0.0.0
I (38530) example: ~~~~~~~~~~~
I (38860) wifi: n:6 0, o:6 0, ap:255 255, sta:6 0, prof:1
I (38860) wifi: state: init -> auth (b0)
I (38870) wifi: state: auth -> assoc (0)
I (38870) wifi: state: assoc -> run (10)
I (38870) wpa: wpa2_task prio:2, stack:6656

I (39960) wpa: >>>>>wpa2 FIALED

I (39970) wpa: wpa2 task delete

I (40530) example: ~~~~~~~~~~~
I (40530) example: IP:0.0.0.0
I (40530) example: MASK:0.0.0.0
I (40530) example: GW:0.0.0.0
I (40530) example: ~~~~~~~~~~~
I (41870) wifi: pm start, type:0

ali-slim-sst, 2018-05-15:
Hi @XinDeng11 , I found the issue with eap connection. ESP32 EAP messages does not specify which eap type it will be using, so the radius server is using the set default which was md5. 

I changed the freeradius default_eap_type from md5 to peap in the configuration file "..mods-enabled/eap", and the device was able to connect.

Is there a way to set the eap type in the ESP32 eap connection messages?

ali-slim-sst, 2018-05-16:
Any updates? @XinDeng11 @FayeY

XinDeng11, 2018-05-21:
@ali-slim-sst actually, we only support PEAP/TTLS/TLS in 32, and they have been start when enable, STA can not set eap type by its own, just start one of the methods base on radius server, server decides use which method. When start handshake, sever will send the type to STA by EAP frame, if STA support this method, it will go on, otherwise the connect would stop

ali-slim-sst, 2018-05-23:
Hi @FayeY  no not really.

@XinDeng11  there is a bit of misunderstanding, the device i.e the STA should be able to specify EAP type  in the EAP frame, like when I connect using another device such as my phone I have to choose a EAP method, in case of the ESP32 it's not specifying the EAP method and failing to connect.

I will post more references later.  

Thanks.

XinDeng11, 2018-05-23:
@ali-slim-sst this is just products behavior, real flow is decided by server
[image]
look at point5 and 6

XinDeng11, 2018-05-23:
@ali-slim-sst any other question?

negativekelvin, 2018-05-23:
>if STA support this method, it will go on, otherwise the connect would stop

>Where a peer receives a Request for an unacceptable authentication
      Type (4-253,255), or a peer lacking support for Expanded Types
      receives a Request for Type 254, a Nak Response (Type 3) MUST be
      sent.  The Type-Data field of the Nak Response (Type 3) MUST
      contain **one or more octets indicating the desired authentication
      Type**(s), one octet per Type, or the value zero (0) to indicate no
      proposed alternative.

ali-slim-sst, 2018-05-24:
@XinDeng11 

Below is form the Radius server configuration file. When the EAP-Identity message does not specify the EAP type, it will use the default_eap_type.

In your test setup, the ESP32 is able to connect if default_eap_type = md5 ?

https://github.com/FreeRADIUS/www.freeradius.org/blob/master/radiusd/raddb/eap.conf

> 	eap {
		#  Invoke the default supported EAP type when
		#  EAP-Identity response is received.
		#
		#  The incoming EAP messages DO NOT specify which EAP
		#  type they will be using, so it MUST be set here.
		#
		#  For now, only one default EAP type may be used at a time.
		#
		#  If the EAP-Type attribute is set by another module,
		#  then that EAP type takes precedence over the
		#  default type configured here.
		#
		default_eap_type = md5

XinDeng11, 2018-05-24:
@ali-slim-sst No, we didn't support md5 yet

ali-slim-sst, 2018-05-24:
@negativekelvin Yes your right, it should keep negotiating

https://tools.ietf.org/html/rfc3748#section-5.3.1

@XinDeng11 Freeradius debug output
Thu May 24 11:56:10 2018 : Debug: (9) eap: Peer sent packet with method EAP NAK (3)
Thu May 24 11:56:10 2018 : Debug: (9) eap: Peer NAK'd indicating it is not willing to continue 
Thu May 24 11:56:10 2018 : Debug: (9) eap: Sending EAP Failure (code 4) ID 139 length 4

ali-slim-sst, 2018-05-24:
@XinDeng11 The picture below is form the wireshark capture, the device on the left is my phone and the device on right is the ESP32. Notice that the phone is specifying the "Desired Auth Type" (EAP-PEAP) (25) while the ESP32 just sends "0". 

[image]

I can send the logs by email if needed.

[image]

XinDeng11, 2018-05-25:
@ali-slim-sst OK, please attach the sniffer log in the Github, we will discussion whether we can add this in the next version, thanks

XinDeng11, 2018-05-25:
@ali-slim-sst we have locked the feature development in our 3.1 version, now we can't add feature casual. However, we are doing plan for 4.0 version now, we can add this feature to a high priority. If it is urge, you can contract with our business affairs, we can do this ahead of schedule after it, thanks

negativekelvin, 2018-05-25:
This is not really a feature, it is a bugfix.  Add 3 bytes to the nak is minor.

murraysst, 2018-05-27:
I agree with @negativekelvin, this is definitely a bug (or at least a deviation from the standard) that stops ESP32 wifi enterprise from functioning in normal enterprise IT environments where it is simply not possible to pre-configure a radius server to work around a non-standard behaviour in the ESP32.

All that is needed is a simple additional field for desired auth type to be added to the IDF. Really not a big deal to add. 

@XinDeng11: This is very high priority for us (it stops our product from working). We are prepared to test this bugfix immediately.

mikaelkanstrup, 2018-05-29:
I investigated this error and now that parts of wpa_supplicant code was moved to esp-idf sources I was able to find a solution.

I'm not sure whether this is an error done when porting supplicant code to esp-idf or done on purpose to solve some specific interop problem but anyway. In function eap_sm_build_nak:

esp-idf loops through all registered eap_methods but set the desired method to EAP_TYPE_NONE:
```
	for (m = methods; m; m = m->next) {
		if (type == EAP_TYPE_EXPANDED) {
			wpabuf_put_u8(resp, EAP_TYPE_EXPANDED);
			wpabuf_put_be24(resp, m->vendor);
			wpabuf_put_be32(resp, m->method);
		} else
			wpabuf_put_u8(resp, EAP_TYPE_NONE);
		found++;
	}

```

This is changed compared to corresponding code in supplicant (function eap_sm_buildNak). In mainline supplicant type is retrieved from registered method(s). Like this:

`wpabuf_put_u8(resp, m->method);
`

That simple change makes wpa eap functional on esp-idf also with servers that not necessarily are configured to make the initial attempt with a supported eap type. With this change esp device will respond with suggested type and server continue trying alternative methods until one that both sides allow is found.

I suggest to first apply the change above and then expose an api for configuring accepted methods. Mainline wpa_supplicant support this through a space separated config in wpa_supplicant.conf (network "eap" option). See functions eap_allowed_method and wpa_config_parse_eap. Something similar should be easy to implement for esp-idf too.

liuzfesp, 2018-05-29:
@mikaelkanstrup, you are right! It's the root cause, we will fix it soon.

zhangyanjiaoesp, 2018-06-04:
@ali-slim-sst this bug has been fixed, the MR is what liuzfesp  pushed above

closed 2018-06-04 by FayeY

andrefilipin, 2018-06-19:
How can i set to mschapv2 method? i have a customer with wpa2 security, when i connected on my mac the authorize was GTC, on linux i can set to mschapv2 and it works, i need to configure EPS32 use mschapv2, please help