espressif/esp-idf #1889: esp32 workins as BLE Gatt client cannot write some value to server`s characteristic

KryvosheyaIvan, 2018-04-26:
Hello!
I have an issue!
There is an device called "FoodLock". And this device locks the door and it is an gatt server.
The task for me is to control the Lock using esp32.
It was verified that the "FoodLock" works correctly.

On the basis of "gatt_client" standard example i managed to connect to FoodLock and discover it services but i can`t write values to it characteristics!

Is an clear example of what i must check before write something extsts? 
or something that cold help me?

`    case ESP_GATTC_OPEN_EVT: //When GATT virtual connection is set up, the event comes
        if (param->open.status != ESP_GATT_OK){
            ESP_LOGE(GATTC_TAG, "open failed, status %d", p_data->open.status);
            break;
        }
        ESP_LOGI(GATTC_TAG, "open success");

        printf(" \n Connection ID = %d \n", p_data->connect.conn_id);

        uint8_t open_array [4] = "open";
        esp_ble_gattc_write_char(  gattc_if,
                gl_profile_tab[PROFILE_A_APP_ID].conn_id,
                gl_profile_tab[PROFILE_A_APP_ID].char_handle,
									sizeof(open_array),
									open_array,
									ESP_GATT_WRITE_TYPE_RSP,
                                    ESP_GATT_AUTH_REQ_NONE);
        printf("written");`

closed 2018-05-02 by KryvosheyaIvan

sowmimalu, 2019-11-26:
Hai ,
   i having question on GATT CLIENT 
    i am able to connect with the service uuid with nrf52840 .
then i am writing data on  characteristic uuid  , is it possible to write data on another characteristic uuid with the same service uuid ..

sowmimalu, 2019-11-26:
i need  to update nrf52840 over dfu with esp32 through bluetooth . i am using gatt client example. 
i am able pair with service uuid , in nrf having two characteritics uuid . i need to write data on both . i am able send a command to one characteristic uuid as per the dfu  protocol . with the same service is it possible to write data on another characteristic uuid  in GATT CLIENT