espressif/esp-idf #1890: Piconet Example

Kolsha, 2018-04-27:
Hi guys, saw in the source code of the makings of piconet, but the implementation examples are nowhere to be found.
I will be glad if you show me how to do this or give links to already existing options.
Thank you!
P.S. I already sought in Google and on your forum.

Kolsha, 2018-04-30:
Anybody?

Weijian-Espressif, 2018-05-14:
where is the source code of the makings of piconet ? is it about  wifi or bt  piconet ?

Kolsha, 2018-05-17:
So, I solved my problem in this way: https://github.com/Kolsha/ESP32-SPP-QT
I will be glad to criticism and suggestions.

blueMoodBHD, 2018-05-30:
@Kolsha Thanks for your demo very mush. It's great!

P.S. **bt_spp_acceptor** and **bt_spp_initiator** in ESP_IDF examples, can connect more than 1 device.

And, is there any problems more?

Kolsha, 2018-06-01:
No, thank you!

closed 2018-06-01 by Kolsha