espressif/esp-idf #1891: "Interrupt wdt timeout on CPU0" caused by a busy loop in esp_timer_impl_set_alarm

alexey-morozov, 2018-04-27:
## Environment

- Development Kit: custom, commercial system design
- Kit version: NA
- Core (if using chip or module): ESP32-Wrover
- IDF version: ESP-IDF v3.1-dev-858-gf885e8b8
- Development Env: Make
- Operating System: CentOS 7
- Power Supply: external 3.3V

## Problem Description

A multi-tasking application which continuously receives data via multiple TCP connections and forwards received data to UART sporadically crashes with an error "Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)". After device auto-reboot the same error ocurrs infinitely.  After rebooting the system with "make monitor" the device started to work again.

According to EspExceptionDecoder (see below) the error is caused by a busy loop in "esp_timer_esp32.c line 217". Changing the number of tasks in the application had an influence on the rate of the error occurrence: with 4 tasks (2 of them TCP receiving using select/recv) the error occurs after a few hours after application start, with 3 tasks (only one of them TCP receiving), the error occurs after ~24 hours.

## Debug Logs

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400820dc  PS      : 0x00060934  A0      : 0x800d187c  A1      : 0x3ffaf9d0  
A2      : 0x3ff47024  A3      : 0x00000000  A4      : 0x3ffb1088  A5      : 0x00000000  
A6      : 0x03003dda  A7      : 0x000000a0  A8      : 0x1e920a63  A9      : 0xf16df59c  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000050  A13     : 0x00000001  
A14     : 0x000000fe  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40094230  LEND    : 0x4009425e  LCOUNT  : 0xffffffff  

Backtrace: 0x400820dc:0x3ffaf9d0 0x400d1879:0x3ffafa00

Core 1 register dump:
PC      : 0x400d2498  PS      : 0x00060e34  A0      : 0x8008ef08  A1      : 0x3ffbcd20  
A2      : 0x00000020  A3      : 0x00000001  A4      : 0x00000020  A5      : 0x3ffb3b48  
A6      : 0x3ffb3e98  A7      : 0x3ffb3c3c  A8      : 0x3ffb3b84  A9      : 0x3ffbcd00  
A10     : 0x00000000  A11     : 0x7fffffff  A12     : 0x8008e16b  A13     : 0x3ffbcc20  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d2498:0x3ffbcd20 0x4008ef05:0x3ffbcd40

Rebooting...

###################################################################

EspExceptionDecoder decoding output:

CORE0
PC: 0x400820dc: esp_timer_impl_set_alarm at /home/morozova/esp/esp-idf/components/esp32/esp_timer_esp32.c line 217
EXCVADDR: 0x00000000

Decoding stack results
0x400820dc: esp_timer_impl_set_alarm at /home/morozova/esp/esp-idf/components/esp32/esp_timer_esp32.c line 217
0x400d1879: timer_task at /home/morozova/esp/esp-idf/components/esp32/esp_timer.c line 294

CORE1
PC: 0x400d2498: esp_vApplicationIdleHook at /home/morozova/esp/esp-idf/components/esp32/freertos_hooks.c line 62
EXCVADDR: 0x00000000

Decoding stack results
0x400d2498: esp_vApplicationIdleHook at /home/morozova/esp/esp-idf/components/esp32/freertos_hooks.c line 62
0x4008ef05: prvIdleTask at /home/morozova/esp/esp-idf/components/freertos/tasks.c line 3409

###################################################################

AFTER AUTO-REBOOT 1:

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40083978  PS      : 0x00050034  A0      : 0x400d7870  A1      : 0x3ffb0a20  
A2      : 0x00000000  A3      : 0x00000040  A4      : 0x00002000  A5      : 0x3ffb0a00  
A6      : 0x3ffb2124  A7      : 0x00002000  A8      : 0x00000000  A9      : 0x4008fb58  
[... 245 lines cut ...]
#define CONFIG_FREERTOS_ISR_STACKSIZE 1536
#define CONFIG_OPENSSL_ASSERT_DO_NOTHING 1
#define CONFIG_SPIRAM_MEMTEST 1
#define CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED 1
#define CONFIG_MBEDTLS_ECP_DP_SECP192R1_ENABLED 1
#define CONFIG_MBEDTLS_ECP_DP_BP512R1_ENABLED 1
#define CONFIG_MBEDTLS_KEY_EXCHANGE_ECDH_ECDSA 1
#define CONFIG_ESP32_WIFI_STATIC_TX_BUFFER_NUM 64
#define CONFIG_SYSTEM_EVENT_QUEUE_SIZE 32
#define CONFIG_FATFS_MAX_LFN 255
#define CONFIG_ESP32_WIFI_TX_BUFFER_TYPE 0
#define CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED 1
#define CONFIG_LWIP_LOOPBACK_MAX_PBUFS 8
#define CONFIG_APP_OFFSET 0x10000
#define CONFIG_SPI_FLASH_ROM_DRIVER_PATCH 1
#define CONFIG_SPIFFS_PAGE_SIZE 256
#define CONFIG_MBEDTLS_ECP_DP_SECP192K1_ENABLED 1
#define CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU0 1
#define CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT 3072
#define CONFIG_MONITOR_BAUD_OTHER_VAL 115200
#define CONFIG_NEWLIB_STDOUT_LINE_ENDING_CRLF 1
#define CONFIG_ESPTOOLPY_PORT "/dev/ttyUSB2"
#define CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS 1
#define CONFIG_OPTIMIZATION_LEVEL_RELEASE 1
#define CONFIG_FATFS_API_ENCODING_ANSI_OEM 1

igrr, 2018-05-02:
Thanks for reporting this @alexey-morozov! I think I see one case when this might happen, but that would be with CONFIG_PM_ENABLE=1, which you don't seem to have. Could you please modify the loop in `esp_timer_impl_set_alarm` function as follows, and try to reproduce the issue again? That's going to give some clues about the case in which the issue can happen.

```diff
diff --git a/components/esp32/esp_timer_esp32.c b/components/esp32/esp_timer_esp32.c
index 4c21910e8..343331089 100644
--- a/components/esp32/esp_timer_esp32.c
+++ b/components/esp32/esp_timer_esp32.c
@@ -213,9 +213,19 @@ void IRAM_ATTR esp_timer_impl_set_alarm(uint64_t timestamp)
     //And we should wait current timer count less than ALARM_OVERFLOW_VAL,
     //maybe equals to 0.
     if (cur_count + offset >= ALARM_OVERFLOW_VAL) {
+        int timeout = 100;
         do {
             overflow = timer_overflow_happened();
             cur_count = REG_READ(FRC_TIMER_COUNT_REG(1));
+            if (--timeout == 0) {
+                portEXIT_CRITICAL(&s_time_update_lock);
+                printf("count=%x\n", (uint32_t) cur_count);
+                printf("alarm=%x\n", REG_READ(FRC_TIMER_ALARM_REG(1)));
+                printf("mask=%d overflow=%d status=%d\n",
+                        s_mask_overflow, s_overflow_happened,
+                        (REG_READ(FRC_TIMER_CTRL_REG(1)) & FRC_TIMER_INT_STATUS) != 0);
+                abort();
+            }
         } while(!overflow || cur_count == ALARM_OVERFLOW_VAL);
     }
```

Alternatively, if that's easier for you, you can use menuconfig to enable GDBStub, and use GDB connection to read the same variables which are being logged here.

alexey-morozov, 2018-05-03:
Hi @igrr !

I have changed "esp_timer_impl_set_alarm" as you suggested and started a test under conditions similar to the ones reported previously. I will report the results as soon as I get a crash.

Meanwhile, I've got some additional information which, I hope, could be helpful. With more tasks (e.g. with 4 instead of 3) "Interrupt wdt timeout" crashes occur quicker (after a few hours). However, in two cases I had, the backtrace points out to a different code location - FREERTOS's "vListInsert" in "list.c" (please see below). For both cases ESP IDF version was v3.1-dev-888-gcc2d1fda.

# CASE A - Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

## Core 0 register dump:
PC      : 0x400900b2  PS      : 0x00060934  A0      : 0x8008f426  A1      : 0x3fff0110  
A2      : 0x3ffecee8  A3      : 0x3fff0338  A4      : 0x3fff031c  A5      : 0x00000001  
A6      : 0x000000fe  A7      : 0x00060023  A8      : 0x3fff0338  A9      : 0x00000014  
A10     : 0x3fff0338  A11     : 0x00000014  A12     : 0x3f4037f5  A13     : 0x3f40363d  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40094b6c  LEND    : 0x40094b77  LCOUNT  : 0xffffffff  

Backtrace: 0x400900b2:0x3fff0110 0x4008f423:0x3fff0130 0x4008de17:0x3fff0150 0x4008ceed:0x3fff0190 0x4012e1a9:0x3fff01b0 0x400d4ebb:0x3fff0230 0x400db24d:0x3fff0260 0x400d8919:0x3fff0290

### EspExceptionDecoder output
PC: 0x400900b2: vListInsert at /home/morozova/esp/esp-idf/components/freertos/list.c line 188
EXCVADDR: 0x00000000

Decoding stack results
0x400900b2: vListInsert at esp-idf/components/freertos/list.c line 188
0x4008f423: vTaskPlaceOnEventList at esp-idf/components/freertos/tasks.c line 2922
0x4008de17: xQueueGenericReceive at esp-idf/components/freertos/queue.c line 1589
0x4008ceed: sys_arch_sem_wait at esp-idf/components/lwip/port/freertos/sys_arch.c line 161
0x4012e1a9: lwip_select at esp-idf/components/lwip/api/sockets.c line 1728
0x400d4ebb: NetSockets::SocketEventMonitor::WaitForEvents() at esp-idf/components/lwip/include/lwip/lwip/sockets.h line 595
0x400db24d: WebHttp::Server::Update() at /myproject/WebHttpServer.cpp line 318
0x400d8919: WebServerTask(void*) at /myproject/WebServerTask.cpp line 80

## Core 1 register dump:
PC      : 0x4008e7c9  PS      : 0x00060034  A0      : 0x8008f238  A1      : 0x3ffbcce0  
A2      : 0x3ffb10bc  A3      : 0x00060023  A4      : 0x00000001  A5      : 0x0000cdcd  
A6      : 0xb33fffff  A7      : 0x00000001  A8      : 0x0000cdcd  A9      : 0x3ffbccd0  
A10     : 0x0000abab  A11     : 0x00000000  A12     : 0x02c4209e  A13     : 0x00000000  
A14     : 0x3ff47024  A15     : 0x00000000  SAR     : 0x00000011  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4008e7c9:0x3ffbcce0 0x4008f235:0x3ffbcd10 0x4008ff50:0x3ffbcd40 0x4008ff06:0x00000000

### EspExceptionDecoder output
PC: 0x4008e7c9: vTaskEnterCritical at esp-idf/components/freertos/include/freertos/portmacro.h line 283
EXCVADDR: 0x00000000

Decoding stack results
0x4008e7c9: vTaskEnterCritical at esp-idf/components/freertos/include/freertos/portmacro.h line 283
0x4008f235: vTaskSwitchContext at esp-idf/components/freertos/tasks.c line 2775

# CASE B - Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

## Core 0 register dump:
PC      : 0x400900ac  PS      : 0x00060134  A0      : 0x8008f426  A1      : 0x3fff9110  
A2      : 0x3ffecee8  A3      : 0x3fff9338  A4      : 0x3fff931c  A5      : 0x00000001  
A6      : 0x000000fe  A7      : 0x00060023  A8      : 0x3fff9338  A9      : 0x00000014  
A10     : 0x3fff9338  A11     : 0x00000014  A12     : 0x3f4037f5  A13     : 0x3f40363d  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40094ed1  LEND    : 0x40094ee1  LCOUNT  : 0xffffffff  

Backtrace: 0x400900ac:0x3fff9110 0x4008f423:0x3fff9130 0x4008de17:0x3fff9150 0x4008ceed:0x3fff9190 0x4012e149:0x3fff91b0 0x400d4ebb:0x3fff9230 0x400db1f1:0x3fff9260 0x400d88bd:0x3fff9290

### EspExceptionDecoder output
PC: 0x400900ac: vListInsert at esp-idf/components/freertos/list.c line 162
EXCVADDR: 0x00000000

Decoding stack results
0x400900ac: vListInsert at esp-idf/components/freertos/list.c line 162
0x4008f423: vTaskPlaceOnEventList at esp-idf/components/freertos/tasks.c line 2922
0x4008de17: xQueueGenericReceive at esp-idf/components/freertos/queue.c line 1589
0x4008ceed: sys_arch_sem_wait at esp-idf/components/lwip/port/freertos/sys_arch.c line 161
0x4012e149: lwip_select at esp-idf/components/lwip/api/sockets.c line 1728
0x400d4ebb: NetSockets::SocketEventMonitor::WaitForEvents() at esp-idf/components/lwip/include/lwip/lwip/sockets.h line 595
0x400db1f1: WebHttp::Server::Update() at /myproject/WebHttpServer.cpp line 318
0x400d88bd: WebServerTask(void*) at /myproject/WebServerTask.cpp line 80

## Core 1 register dump:
PC      : 0x4008e7cb  PS      : 0x00060034  A0      : 0x8008f238  A1      : 0x3ffbcce0  
A2      : 0x3ffb10bc  A3      : 0x00060023  A4      : 0x00000001  A5      : 0x0000abab  
A6      : 0xb33fffff  A7      : 0x00000001  A8      : 0x0000cdcd  A9      : 0x3ffbccd0  
A10     : 0x0000abab  A11     : 0x0000000d  A12     : 0x01256f85  A13     : 0x00000000  
A14     : 0x3ff47024  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4008e7cb:0x3ffbcce0 0x4008f235:0x3ffbcd10 0x4008ff50:0x3ffbcd40 0x4008ff06:0x3ffb3edc

### EspExceptionDecoder output
PC: 0x4008e7cb: vTaskEnterCritical at esp-idf/components/freertos/include/freertos/portmacro.h line 283
EXCVADDR: 0x00000000

Decoding stack results
0x4008e7cb: vTaskEnterCritical at esp-idf/components/freertos/include/freertos/portmacro.h line 283
0x4008f235: vTaskSwitchContext at esp-idf/components/freertos/tasks.c line 2775

alexey-morozov, 2018-05-17:
Dear @igrr,

So far I got 3 additional cases with "Interrupt wdt timeout on CPU0" crashes. None of them is induced by the abort in your modified esp_timer_impl_set_alarm code. In the attached "interrupt.wdt.cases.17-05-2018.txt" below you can see some details. For some reason, the traceback for case C could not be decoded by EspExceptionDecoder. For the cases B & C I have coredump output in Base64 format printed over the UART (see attached files below). If it can help, I'm willing to provide you with the corresponding ELF files (by e-mail).

In all cases, as reported previously, the system could not recover from the crash and continued to crash indefinitely with a message "Core 0 was running in ISR context". Do you see any possibility to change this behaviour so that the application can boot again and continue it's normal operation?

[interrupt wdt cases 17-05-2018.txt](https://github.com/espressif/esp-idf/files/2013971/interrupt.wdt.cases.17-05-2018.txt)
[coredump case B.txt](https://github.com/espressif/esp-idf/files/2013972/coredump.case.B.txt)
[coredump case C.txt](https://github.com/espressif/esp-idf/files/2013974/coredump.case.C.txt)

APokorny, 2018-05-18:
So this type of error happens more frequently now because the time window in esp_timer_impl_set_alarm is now bigger - thus there is a higher probability that the ISR and the busy loop block each other. We currently deal (with an ugly workaround) with errors that may occur with the 22th Oct. 2017 version of this file and we are afraid to upgrade to master. 

Is there a more detailed spec of those registers available? We would like to see these problems fixed and would like to a see an implementation of the ISR that never blocks.

igrr, 2018-05-18:
@APokorny There has been a number of bug fixes to esp_timer since Oct 2017, please try the latest version (or at least the latest release, v3.0). If you are facing issues after updating to latest master branch or latest release, please apply the patch I've given to @alexey-morozov above, and post debug output you get when the issue happens. This will help us understand the conditions and reproduce the issue.

@alexey-morozov the infinite loop in vListInsert usually indicates one of the two issues:

1. memory corruption (something has overwritten a member of ListItem_t)
2. attempt to call a blocking API when interrupts are disabled (e.g. inside a critical section).

I see that you are running with `CONFIG_HEAP_POISONING_DISABLED`, could you please enable heap poisoning at "Comprehensive" level and try reproducing the issue again? Heap poisoning checks might help if the issue is caused by 1).

andreapedica, 2018-05-22:
Hi everyone,

I am having the same problem that user @alexey-morozov stated in the original post of this issue.

I have implemented the code log suggested by @igrr and here is presented the result:

```
count=f0002cfb
alarm=efffffff
mask=0 overflow=0 status=0
abort() was called at PC 0x400820e6 on core 0
```

I found the problem letting the code run for ~1 hour and half, fully repeatable. 

I did this test using official 3.0 release of ESP-IDF but even latest releases do the same.

Hope this data could help to investigate the issue, in attach you can find sdkconfig of my project (I'm running single core but also dual core replicates the problem).

@igrr if you have additional requests to instrument code and perform debug to help you better understand the issue, I'm fully open for your suggestions.

Thanks for your support!
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/2025631/sdkconfig.txt)

andreapedica, 2018-05-22:
Hi all,

I am analyzing the code around the area of error and I have a thought: to perform evaluations and exit conditions from the loop, we are looking at this condition mainly:

`REG_READ(FRC_TIMER_ALARM_REG(1)) == ALARM_OVERFLOW_VAL`

but the read of the register is practically asynchronous and, most likely, much slower than the update frequency of the freee running counter addressed at `FRC_TIMER_ALARM_REG(1)`. 

So, looking also at the data print in the debug trap, the counter actually overlapped the value 0xefffffff.

This makes me think that maybe changin the check condition with:

`REG_READ(FRC_TIMER_ALARM_REG(1)) >= ALARM_OVERFLOW_VAL`

should permit to unlock the loop with higher probability rather than catch "exactly" the moment that the free running counter matches the value 0xefffffff right when we are doing the exit condition check.

Does this makes sense for you?

igrr, 2018-05-23:
Thank you for the additional information @andreapedica! It is quite useful. I don't think that your evaluation of exit condition is correct though. To exit the loop, it is required (but not sufficient) that `REG_READ(FRC_TIMER_ALARM_REG(1)) != ALARM_OVERFLOW_VAL`.

That being said, I think this loop is not needed anymore. It was added when `ALARM_OVERFLOW_VAL` was equal to 2^32 - 1, and was necessary to prevent wraparound condition from happening while esp_timer_impl_set_alarm was running. Now, `ALARM_OVERFLOW_VAL` is less than 2^32 - 1, so there is sufficient headroom for the counter to increase while we set the alarm. I'm writing addition tests for this case and will have a fix soon.

andreapedica, 2018-05-23:
Dear @igrr you are completely right, I did evaluation of exit condition with opposite logic, sorry for my mistake :) 

Ok then I'll wait for your fix, if you need some support on my side I'm fully available!

igrr, 2018-05-23:
@APokorny @andreapedica Could you please give `bugfix/esp_timer_set_alarm` branch a try and see if you still observe any issue?

andreapedica, 2018-05-24:
Dear @igrr, thanks for your fast support, I am going to download your fix and let it run during the day.
I'll come back with my foundings later on.

andreapedica, 2018-05-24:
Dear @igrr so far the sw is running without triggering any blocking failure. 

But unfortunately it seems like the timer is loosing precision, most likely I suppose the timer callback periodically is not called with expected timeout. I have set up an esp timer callback to rise every second as time base for clock shown on display, and I noticed that in 3 hours of running it lost around 1 minute and half.

Looking at the comments in your new code, I see the notice you put:

```
    // Use calculated alarm value if it is less than ALARM_OVERFLOW_VAL.
    // Note that if by the time we update ALARM_REG, COUNT_REG value is higher,
    // interrupt will not happen for another ALARM_OVERFLOW_VAL timer ticks,
    // so need to check if alarm value is too close in the future (e.g. <2 us away).
```

I'm afraid this is actually happening. Do you have already an idea of a fix for this problem? If you want me to instrument code to log debug data I'm fully available.

jjimmy-S, 2018-05-30:
Hi everyone, I'm getting a similar error when running encoder interrupts......

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      : 0x400852d4  PS      : 0x00060034  A0      : 0x800858cb  A1      : 0x3ffc05a0  
A2      : 0x3ffc1094  A3      : 0x00000000  A4      : 0x00000005  A5      : 0x00000020  
A6      : 0x00000020  A7      : 0x00060023  A8      : 0x0000abab  A9      : 0x0000cdcd  
A10     : 0xb33fffff  A11     : 0x0000abab  A12     : 0x00060021  A13     : 0x00000001  
A14     : 0x00060021  A15     : 0x00060023  SAR     : 0x0000001b  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400852d4:0x3ffc05a0 0x400858c8:0x3ffc05c0 0x4008674f:0x3ffc05e0 0x400864a1:0x3ffc0600 0x4008168a:0x3ffc0610 0x400d5fdf:0x00000000

Rebooting...

⸮⸮t"ǜ⸮⸮Q1,⸮01⸮Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      : 0x400852d4  PS      : 0x00060034  A0      : 0x800858cb  A1      : 0x3ffc05a0  
A2      : 0x3ffc1094  A3      : 0x00000000  A4      : 0x00000005  A5      : 0x00000020  
A6      : 0x00000020  A7      : 0x00000000  A8      : 0x0000abab  A9      : 0x0000cdcd  
A10     : 0xb33fffff  A11     : 0x0000abab  A12     : 0x00060021  A13     : 0x00000001  
A14     : 0x00060021  A15     : 0x3ffc68e0  SAR     : 0x0000001b  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400852d4:0x3ffc05a0 0x400858c8:0x3ffc05c0 0x4008674f:0x3ffc05e0 0x400864a1:0x3ffc0600 0x4008168a:0x3ffc0610 0x400d5fdf:0x00000000

Rebooting...
⸮⸮⸮⸮⸮K⸮
⸮ؔ!XـA

Spritetm, 2018-05-30:
jjimmy-S: Please start a new issue for that. Interrupt wdt timeouts can be because of a fair few reasons and I don't think yours is related to the one in this issue.

APokorny, 2018-06-04:
@igrr I integrated that change and had a system running for four days now. So far we did not observe any issues. System time seems to stay in sync. We will keep it running the next weeks. (When the test run is complete I can have a detailed look at our logs on the sd card .. i.e. whether there were any intermittent hickups)

andreapedica, 2018-06-11:
Dear @igrr,

also from my side I have not seen anymore reset events. 

Here below you can see a plot of measurements logged via serial port, to monitor the call rate of a timer task set up to trigger periodically every 1 second (using esp_timer_start_periodic() method).

The values in usec have been calculated doing delta between periodical readings of esp_timer_get_time() function:

[image]

It is visible a period jitter of +/- ~20usec, most probably due to RTOS latency to launch timer task, high resolution timer precision and also other tasks with higher priority running.

Anyway, I think that for the reset problem this issue can be considered closed. For the precise time counting, I will rely on independent clock sources (like internal RTC through the time() function).

So do you agree to pull the `bugfix/esp_timer_set_alarm` branch into `master`?

Thanks for your support!

igrr, 2018-06-11:
Hi @andreapedica, thanks for confirming the fix. You are correct to note that interrupt latency and presence of other tasks (most notably, high priority WiFi/BT tasks) will add jitter to the callback execution time.
You can still call `esp_timer_get_time` function from within the callback, to obtain the actual current time. This value should exhibit much lower jitter (less than 1us, which is on the order of the execution time of `esp_timer_get_time`). With the default configuration in ESP-IDF, `time` and `gettimeofday` functions derive their results from values returned by `esp_timer_get_time`.
The bugfix branch is undergoing testing at the moment, we plan to merge it as soon as all the tests pass. This issue will be updated when this happens.

alexey-morozov, 2018-06-11:
Hi guys, I also do not observe WDT timeout resets anymore. 

Thanks a lot!

closed 2018-06-13 by igrr

xgenesys, 2018-11-06:
Hi @igrr ,

I am using the mqtt/tcp example to publish some data to mqtt broker (use WIFI), I also try to scan BLE packages also.

The scanning duration is infinite, any beacon package found, I will forward it to MQTT broker.

I got the same issue as this thread discussion, you can see the log below:

`ASSERT_PARAM(0 4), in lld_evt.c at line 1749
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40084729  PS      : 0x00060b34  A0      : 0x800863da  A1      : 0x3ffbe260
0x40084729: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x000e0001  A5      : 0x60008054
A6      : 0x3ffbdcd0  A7      : 0x60008050  A8      : 0x80084729  A9      : 0x3ffbe240
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  Q3     : 0xffffffff
Q4     : 0x00000000  Q5     : 0xfffffffb  SAR     : 0x00000004  EXCCQSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40084661  LEND    : 0x40084668  LCOUNT  : 0x00000000
0x40084661: r_assert_param at ??:?

0x40084668: r_assert_param at ??:?

Core 0 was running in ISR context:
EPC1    : 0x40090967  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40084729
0x40090967: vTaskEnterCritical at C:/msys32/home/thanh-IOT/esp/esp-idf/components/freertos/tasks.c:3564

0x40084729: r_assert_param at ??:?

Backtrace: 0x40084729:0x3ffbe260 0x400863d7:0x3ffbe280 0x40086fa6:0x3ffbe2a0 0x40087ee2:0x3ffbe2c0 0x4008901b:0x3ffbe2e0 0x40082609:0x3ffbe300 0x4000bfed:0x00000000
0x40084729: r_assert_param at ??:?

0x400863d7: r_lld_evt_end at ??:?

0x40086fa6: r_lld_evt_end_isr at ??:?

0x40087ee2: r_rwble_isr at ??:?

0x4008901b: r_rwbtdm_isr_wrapper at intc.c:?

0x40082609: _xt_lowint1 at C:/msys32/home/thanh-IOT/esp/esp-idf/components/freertos/xtensa_vectors.S:1105

Core 1 register dump:
PC      : 0x400d3996  PS      : 0x00060334  A0      : 0x8009106c  A1      : 0x3ffbc350
0x400d3996: esp_vApplicationIdleHook at C:/msys32/home/thanh-IOT/esp/esp-idf/components/esp32/freertos_hooks.c:88

A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x00060021  A7      : 0x00000000  A8      : 0x3ffc8aa0  A9      : 0x3ffc8a64
A10     : 0x00000000  A11     : 0x80000001  A12     : 0x00000000  Q3     : 0x00000001
Q4     : 0x00000003  Q5     : 0x00060023  SAR     : 0x00000000  EXCCQSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x400d3996:0x3ffbc350 0x40091069:0x3ffbc370 0x40091a15:0x3ffbc390
0x400d3996: esp_vApplicationIdleHook at C:/msys32/home/thanh-IOT/esp/esp-idf/components/esp32/freertos_hooks.c:88

0x40091069: prvIdleTask at C:/msys32/home/thanh-IOT/esp/esp-idf/components/freertos/tasks.c:3564

0x40091a15: vPortTaskWrapper at C:/msys32/home/thanh-IOT/esp/esp-idf/components/freertos/port.c:403

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6200
load:0x40078000,len:9404
load:0x40080400,len:7128
entry 0x400807a0
`

My esp_timer_impl_set_alarm function as below, I think it is the latest version:

`void IRAM_ATTR esp_timer_impl_set_alarm(uint64_t timestamp)
{
    portENTER_CRITICAL(&s_time_update_lock);
    // Alarm time relative to the moment when counter was 0
    uint64_t time_after_timebase_us = timestamp - s_time_base_us;
    // Adjust current time if overflow has happened
    bool overflow = timer_overflow_happened();
    uint64_t cur_count = REG_READ(FRC_TIMER_COUNT_REG(1));

    if (overflow) {
        assert(time_after_timebase_us > s_timer_us_per_overflow);
        time_after_timebase_us -= s_timer_us_per_overflow;
        s_overflow_happened = true;
    }
    // Calculate desired timer compare value (may exceed 2^32-1)
    uint64_t compare_val = time_after_timebase_us * s_timer_ticks_per_us;
    uint32_t alarm_reg_val = ALARM_OVERFLOW_VAL;
    // Use calculated alarm value if it is less than ALARM_OVERFLOW_VAL.
    // Note that if by the time we update ALARM_REG, COUNT_REG value is higher,
    // interrupt will not happen for another ALARM_OVERFLOW_VAL timer ticks,
    // so need to check if alarm value is too close in the future (e.g. <2 us away).
    const uint32_t offset = s_timer_ticks_per_us * 2;
    if (compare_val < ALARM_OVERFLOW_VAL) {
        if (compare_val < cur_count + offset) {
            compare_val = cur_count + offset;
            if (compare_val > ALARM_OVERFLOW_VAL) {
                compare_val = ALARM_OVERFLOW_VAL;
            }
        }
        alarm_reg_val = (uint32_t) compare_val;
    }
    REG_WRITE(FRC_TIMER_ALARM_REG(1), alarm_reg_val);
    portEXIT_CRITICAL(&s_time_update_lock);
}`

Could you please tell me where is the problem come from? I am new to ESP32.

Thank you very much,

Alvin1Zhang, 2018-11-06:
@xgenesys Hi, would you please help create a separate issue for your questions? Thanks.

xgenesys, 2018-11-06:
> @xgenesys Hi, would you please help create a separate issue for your questions? Thanks.

already created a new issue, sorry.