espressif/esp-idf #1892: 我使用ESP32-PICO-KIT开发板调试blufi例子的时候，开机加载报错，log如下，请帮忙看下，谢谢

xiao5914234, 2018-04-27:
I (31) boot: ESP-IDF v3.1-dev-743-g6c44fc70 2nd stage bootloader
I (31) boot: compile time 12:14:44
I (31) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3659c (222620) map
I (173) esp_image: segment 1: paddr=0x000465c4 vaddr=0x3ffc0000 size=0x03b58 ( 15192) load
I (179) esp_image: segment 2: paddr=0x0004a124 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at C:/msys32/home/Administrator/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (180) esp_image: segment 3: paddr=0x0004a52c vaddr=0x40080400 size=0x05ae4 ( 23268) load
I (199) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xb259c (730524) map
0x400d0018: _stext at ??:?

I (455) esp_image: segment 5: paddr=0x001025bc vaddr=0x40085ee4 size=0x0db0c ( 56076) load
0x40085ee4: r_lld_evt_restart at ??:?

I (478) esp_image: segment 6: paddr=0x001100d0 vaddr=0x400c0000 size=0x00000 (     0) load
E (479) esp_image: Image length 1048832 doesn't fit in partition length 1048576
E (484) boot: Factory app partition is not bootable
E (490) boot: No bootable app partitions in the partition table

Weijian-Espressif, 2018-05-02:
@xiao5914234 menuconfig中将gattc 和 SMP, classic bluetooth都关掉, 会减少bin的大小

lmayainsite, 2018-07-09:
**Good day, this error happends to me, I dont know if this is a hardware problem or a programming problem. I have this log. How can I fix it?**

I (30) boot: ESP-IDF v3.0.1-rc-dirty 2nd stage bootloader
I (30) boot: compile time 08:30:03
I (30) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 26.7MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x071e4 ( 29156) map
I (106) esp_image: segment 1: paddr=0x0001720c vaddr=0x3ffb0000 size=0x022b0 (  8880) load
I (111) esp_image: segment 2: paddr=0x000194c4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/programador/esp_32/esp-idf/components/freertos/./xtensa_vectors.S:1685

I (115) esp_image: segment 3: paddr=0x000198cc vaddr=0x40080400 size=0x06744 ( 26436) load
I (136) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x165a4 ( 91556) map
0x400d0018: _stext at ??:?

I (174) esp_image: segment 5: paddr=0x000365c4 vaddr=0x40086b44 size=0x01e54 (  7764) load
0x40086b44: block_data_size at /home/programador/esp_32/esp-idf/components/heap/./multi_heap.c:142
 (inlined by) multi_heap_realloc_impl at /home/programador/esp_32/esp-idf/components/heap/./multi_heap.c:525

I (178) esp_image: segment 6: paddr=0x00038420 vaddr=0x400c0000 size=0x00000 (     0) load
E (180) esp_image: Checksum failed. Calculated 0x1 read 0x7d
E (187) boot: Factory app partition is not bootable
E (192) boot: No bootable app partitions in the partition table
user code done

TobeChan, 2018-09-12:
问题已解决！谢谢大神！@redchenjs