espressif/esp-idf #1893: esp_wifi_scan_get_ap_records crashes when non-malloc'ed pointer passed

cocoabox, 2018-04-27:
## Environment

- Development Kit:      ESP32-DevKitC 
- Kit version (for WroverKit/PicoKit/DevKitC): v3
- Core (if using chip or module): ESP-WROOM32
- IDF version (``git rev-parse --short HEAD`` to get the commit id.):  cc2d1fda964afb310c0ffa9b1bf9d2065b7121f2
- Development Env:		Make
- Operating System:		MacOS
- Power Supply:         USB

## Problem Description

Upon receiving the SYSTEM_EVENT_SCAN_DONE event, esp_wifi_scan_get_ap_records() was called to collect the list of AP names/SSID/etc 
But the call crashes. The crash occurs only when the 2nd argument points to stack a buffer. When a malloc'ed buffer is called
the function works normally.

### Expected Behavior

esp_wifi_scan_get_ap_records() should populate date into the buffer passed.

### Actual Behavior

Crashes

### Steps to repropduce

call esp_wifi_scan_start to being scanning; when scan completes the program crashes :

```
            wifi_scan_config_t config;
            memset(&config, 0, sizeof(config));
            config.ssid = NULL;
            config.bssid = NULL;
            config.channel = 0; // all channels
            config.scan_type = WIFI_SCAN_TYPE_ACTIVE;
            config.show_hidden = true;
            esp_wifi_scan_start(&config, true);
```

### Code to reproduce this issue

```cpp
    static esp_err_t __wifi_event_handler(void *ctx, system_event_t *event) {
        switch (event->event_id) {
//                  :
//		      :
            case SYSTEM_EVENT_SCAN_DONE: {
                {
                    ESP_LOGI(TAG, "[%s] SYSTEM_EVENT_SCAN_DONE", __func__);
                    uint16_t apCount = 0;
                    esp_wifi_scan_get_ap_num(&apCount);

                    wifi_ap_record_t list[40];
                    memset(list, 0, sizeof(list));
                    uint16_t size = 40;
                    esp_wifi_scan_get_ap_records(&size, list);             // <<---- CRASHES

                    for (uint16_t i = 0; i < size; i++) {
                        wifi_ap_record_t *ap = list + i;
                        char *ssid = (char *) ap->ssid;
                        char ap_str[20] = {0};
                        sprintf(ap_str, "%d,%s", ap->rssi,
                                ap->authmode == WIFI_AUTH_OPEN ? "open" :
                                ap->authmode == WIFI_AUTH_WEP ? "wep" :
                                ap->authmode == WIFI_AUTH_WPA_PSK ? "wpa-psk" :
                                ap->authmode == WIFI_AUTH_WPA2_PSK ? "wpa-psk" :
                                ap->authmode == WIFI_AUTH_WPA_WPA2_PSK ? "wpa-wpa2-psk" :
                                ap->authmode == WIFI_AUTH_WPA2_ENTERPRISE ? "wpa-eap" :
                                "unknown"
                        );
                        ESP_LOGD(TAG, "[%s | %s:%d] got AP %s : %s", __func__, __FILE__, __LINE__, ssid, ap_str);
                        MyWiFi::scanned_networks->set(ssid, ap_str);
                    }
                }

            }
            default:
                break;
        }
        return ESP_OK;
    }
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
     :
     :
V (3406) event: exit default callback
D (5766) event: SYSTEM_EVENT_SCAN_DONE, status:0, number:18
I (5766) test8: [__wifi_event_handler] SYSTEM_EVENT_SCAN_DONE
reset...
***ERROR*** A stack overflow in task eventTask has been detected.
abort() was called at PC 0x4008d768 on core 0
0x4008d768: vApplicationStackOverflowHook at /Users/u0/esp/esp-idf/components/esp32/panic.c:648

Backtrace: 0x4008d578:0x3ffbf830 0x4008d74f:0x3ffbf850 0x4008d768:0x3ffbf870 0x4008afcf:0x3ffbf890 0x4008c670:0x3ffbf8b0 0x4008c626:0x00000020
0x4008d578: invoke_abort at /Users/u0/esp/esp-idf/components/esp32/panic.c:648

0x4008d74f: abort at /Users/u0/esp/esp-idf/components/esp32/panic.c:648

0x4008d768: vApplicationStackOverflowHook at /Users/u0/esp/esp-idf/components/esp32/panic.c:648

0x4008afcf: vTaskSwitchContext at /Users/u0/esp/esp-idf/components/freertos/tasks.c:4574

0x4008c670: _frxt_dispatch at /Users/u0/esp/esp-idf/components/freertos/portasm.S:406

0x4008c626: _frxt_int_exit at /Users/u0/esp/esp-idf/components/freertos/portasm.S:206

Rebooting...
ets Jun  8 2016 00:22:5```

mahavirj, 2018-04-27:
@cocoabox 
From the logs this is related stack overflow in event loop task,
```
***ERROR*** A stack overflow in task eventTask has been detected.
```
Default stack size (2304 bytes) of event loop task (in which context, event handler `__wifi_event_handler` will get invoked) is not sufficient to hold big buffer in the form of `wifi_ap_record_t`.

So, dynamic allocation for buffer is right approach. (Having said that it is also possible to configure stack size for event loop task from config option `ESP32 specific -> Event loop task stack size`, that should help to resolve the problem).

cocoabox, 2018-05-11:
@mahavirj  @FayeY  Sorry for the late reply. Yes that was my problem! Sorry it was my bad!  Increasing the stack size helped resolve the issue. Thanks!

closed 2018-05-11 by cocoabox