espressif/esp-idf #1895: Receive UDP

rinripper, 2018-04-29:
## Environment

- Development Kit:      |ESP32-DevKitC|
- Kit version [v1]
- IDF version: master (Updated 2 days ago by jack0c)
- Development Env: Eclipse
- Operating System: Windows
- Power Supply: [USB]

## Problem Description

Hi, I´m having problems receiving Multicast (UDP). I know UDP it is a protocol without secure reception, but I only receive al least the 20% of the packets . 

The problem started after the update (release/v3.0 (Updated 5 days ago by igrr)) and still in the lastest update of the ESP-IDF (master (Updated 2 days ago by jack0c)).

* I use Packet Sender (PC program) to send UDP Multicast packets, I receive in others PCs in the wireless netwotk.
*My wireless network is used only by me, it´s not trafic conflict. (Wireshark confirmation)
*The ESP32 and the antena of the router are separate by 30 cms. 
*The data I´m sending to the esp32 is literally "50 4585", super short message. 
*The ESP32 are powered by USB with enough current.
*Change to bigger UDP_RECVMBOX_SIZE doesn´t make any different
*The speed is 1 UDP every 2 seconds, super slow
* I´m using the multicast ip 239.7.7.7, and port 50793.

I think the ESP32 it is dropping UDP packets , because TCP works fine. 

I found I´m the only one, people are having trouble with UDP related topic these days

http://bbs.esp32.com/viewtopic.php?t=5411
https://github.com/espressif/arduino-esp32/issues/1317 

Thanks in advance
Mauricio

negativekelvin, 2018-04-29:
What version were you previously using that wasn't producing the problem?

rinripper, 2018-05-03:
Hi.

Problem Fixed in branch master (Updated 14 hours ago by igrr)

Thanks igrr

liuzfesp, 2018-06-22:
@rinripper sorry for the late response, could you help to check whether this issue still exist?

carbonadam, 2018-07-03:
Seems to be a pretty major bug for being able use esp32 with UDP for anything decent

liuzfesp, 2018-07-08:
@carbonadam could you describe your issue in detailed please?

NZSmartie, 2018-07-11:
I'm having a similar issue. 
In my application, i'm joining the multicast group "224.0.1.187" using LWIP's netconn API

```c
if( netconn_bind( listen_socket, IP_ADDR_ANY, 5683 ) != ERR_OK )
{
    ESP_LOGE( TAG, "netconn_bind( ... ): Failed" );
    netconn_delete( listen_socket );
    return -1;
}

ip_addr_t multicast_addr;
IP4_ADDR( ip_2_ip4(&multicast_addr) , 224, 0, 1, 187 );
if( netconn_join_leave_group( listen_socket, &multicast_addr, NULL, NETCONN_JOIN ) != ERR_OK )
    ESP_LOGE( TAG, "netconn_join_leave_group( ... ): Failed" );
```

I can see IGMPv2 packets on Wireshark confirming that the ESP32 has joined the multicast group. However, when sending a multicast packet (in my case, using the `coap-client` from `libcoap`
```bash
coap-client -N coap://224.0.1.187/.well-known/core
```

I don't see any activity on the ESP32. Even with `UDP_DEBUG` set to 1, i don't see them coming through.

liuzfesp, 2018-07-17:
@NZSmartie does this issue only specific to UDP multicast receiving? Or it's related to UDP unicast receiving?

NZSmartie, 2018-07-17:
Multicast. Unicast works without issues

liuzfesp, 2018-07-17:
Got it. I will debug it.

liuzfesp, 2018-07-18:
@rinripper could you help to capture the UDP multicast packets between the router and ESP32? It's helpful for us to debug your issues. 

More information about UDP multicast, take IPv4 for example. Generally speaking, different protocol header will be appended to the UDP multicast packet before it's finally sent out: 
1> Encapsulate UDP  
2> Add IP header
3> Add 802.3 Ethernet header
4> Convert 802.3 Ethernet Header to 802.11 WiFi Header
5> WiFi sends out the packet

For UDP multicast packet, in step 3>, the destination MAC of the Ethernet header is converted from the multicast address, for example, if the multicast address is 232.10.11.12, then the destination MAC of Ethernet will be 01:00:5e:0a:0b:0c. In step 4>, the 802.3 Ethernet header is converted to 802.11 WiFi header, the destination MAC of WiFi packet is same as destination MAC of Ethernet MAC: 01:00:5e:0a:0b:0c. 

Because different address scheme, the multicast transmit behavior is very different between the station and soft-AP. If the packet is sent from station to soft-AP, because the station can only connects to one specific soft-AP, **so there is no actual multicast/broadcast from station to soft-AP in WiFi layer. In other words, the multicast UDP is actually a unicast WiFi packet from station to soft-AP.** To be detailed, the address 1 is the BSSID of soft-AP, the address of is the MAC of station, the address 3 is the destination mac: 01:00:5e:0a:0b:0c.  For each unicast packet, WiFi has positive acknowledge, if the soft-AP doesn't receives the packet, the station reties sometimes ... ... so the multicast from station to soft-AP is relatively reliable. **The packet drop ratio should be same as the unicast packet drop ratio.**

On the other hand, if the UDP multicast is from soft-AP to the station. Because more than one stations can connects to the soft-AP, the soft-AP has real multicast/broadcast. In WiFi layer, the address 1 of WiFi packet is the destination address: 01:00:5e:0a:0b:0c, the address 2/3 is the BSSID of soft-AP. According to WiFi Spec, **there is no positive acknowledge for multicast, in other words, there is not retransmit mechanism for multicast on soft-AP side. So the multicast UDP from soft-AP to station only has one chances to be sent.**

rinripper, 2018-07-30:
Hi liuzfesp, sorry for the late respond.

Here I upload a capture of the UDP multicast packet in Wireshark. 

The test I made was:

1. Turn on wireshark with multicast address in filter
2. Turn on ESP32 with UDP receive code:

`	// DECLARE MULTICAST
	struct ip_mreq mc_req;
	struct sockaddr_in multicastAddress;

	// CREATE A UDP SOCKET
	sockUDP = socket(AF_INET, SOCK_DGRAM, IPPROTO_UDP);
	if (sockUDP < 0) {
		close(sockUDP);
		ESP_LOGI(TAG, "CREATE UDP SOCKET STATUS: %s", strerror(errno));
	 	return 0;
	}

	memset(&multicastAddress, 0, sizeof(multicastAddress));
	multicastAddress.sin_family      = AF_INET;
	multicastAddress.sin_addr.s_addr = htonl(UDP_IP_DEFAULT);
	multicastAddress.sin_port        = htons(UDP_PORT_DEFAULT);

	// BIND SOCKET TO MULTICAST ADDRESS
	int rc  = bind(sockUDP, (struct sockaddr *)&multicastAddress, sizeof(multicastAddress));
	if (rc < 0) {
		close(sockUDP);
		ESP_LOGI(TAG, "BIND UDP SOCKET STATUS: %s", strerror(errno));
		return 0;
	}

	mc_req.imr_multiaddr.s_addr = inet_addr(UDP_IP_DEFAULT);
	mc_req.imr_interface.s_addr = htonl(INADDR_ANY);

	// JOIN MULTICAST ADDRESS PARTY
	if ((setsockopt(sockUDP, IPPROTO_IP, IP_ADD_MEMBERSHIP, (void*) &mc_req, sizeof(mc_req))) < 0){
		close(sockUDP);
		return 0;
	}

	struct sockaddr_in serverAddress;

	// SET A BUFFER TO RECEIVE MULTICAST MESSAGE
	int sizeBuffer = 256;

	unsigned int addresLentgh =sizeof(serverAddress);

	while(1){

		ESP_LOGI(TAG, "LISTENING MULTICAST...");

		memset(&serverAddress, 0, addresLentgh);
		buffer= calloc(sizeBuffer, sizeof(char));

		// RECEIVE MESSAGE
		int recvSock = recvfrom(sockUDP, buffer, sizeBuffer, 0, (struct sockaddr *)&serverAddress, &addresLentgh);

}
`
3. Send UDP multicast packet using Packet Sender

In this brief test, i receive only 5 of 15 packet sended.

I tried to understand what you write to do, but I don´t fully understand. But with the capture on Wireshark and the number of the dropped packet you can see what you are looking for.

The number on Wireshark of the dropped packets where: 976, 1340, 1737, 2306, 2830, 2911, 2951, 3015, 3068, 3248.
Thanks
Mauricio

[Capture.zip](https://github.com/espressif/esp-idf/files/2239405/Capture.zip)

liuzfesp, 2018-08-07:
Hi @rinripper what you captured is the ethernet packets, I guess the packets are from PC to AP. Could you help to capture the WiFi packets from AP to ESP32 device?

gengshenghong, 2018-12-18:
Is there any update on this issue? I am using latest Micropython and wrote a simple SSDP (multicast) application, and it behaviors very similar to this post. I ran the same code on a PC, and everything works (I use nodejs node-ssdp to send m-search every 1 secs), with PC running ssdp server, it can recv all data. But running the ssdp server on esp32 at the same time, it can recv 1 packet every 7-8 packets.

Any thoughts?

Thanks,
Shenghong

gengshenghong, 2018-12-18:
I've found this example:
https://github.com/espressif/esp-idf/tree/master/examples/protocols

Testing it and it is also very slow to recv packets. Is it tested with keep sending packets (example: 1 secs as I used, which is slow enough and should not lost any packet but the example lost packets)?

Thanks,
Shenghong

gengshenghong, 2018-12-18:
How to reproduce:
1. build https://github.com/espressif/esp-idf/tree/master/examples/protocols/sockets/udp_multicast example and upload to esp32
2. run below script
```
#!/bin/bash

ip=`/sbin/ifconfig -a|grep inet|grep -v 127.0.0.1|grep -v inet6|awk '{print $2}'|tr -d "addr:"`

echo "Your IP: " $ip
while true
do
echo "send..."
echo "Hi there, IPv4!" | socat STDIO UDP4-DATAGRAM:239.239.239.239:55555,ip-multicast-if=$ip
sleep 1
done
```
3. expected behaviour
we should no see logs like "I (235181) multicast: Sending to IPV4 multicast address", as the esp32 side code will only "send" if "not recv" in 2.5 secs.
4. actual behaviour
we see recv logs like "I (144501) multicast: Hi there, IPv4!" every 2-3 (even 7-8 secs longer sometimes), and it will print send logs.

Thanks,
Shenghong

negativekelvin, 2018-12-18:
Use of power save modes will reduce udp multicast receive ability.

gengshenghong, 2018-12-19:
@negativekelvin is power save mode default? I am new to esp32 and nothing else was done except running the example.

gengshenghong, 2018-12-20:
Thank you for your help. I have find the root cause. It is NOT esp32 issue, it is my router issue!!! I use a mini-router for development and it seems very bad, using a new router works well.

Thanks,
Shenghong

closed 2019-03-22 by Alvin1Zhang