espressif/esp-idf #1896: Ram and Flash testing

hvishu, 2018-04-29:
creating in esp-idf as per the suggestion from the link https://github.com/aws/aws-iot-device-sdk-embedded-C/issues/165#issuecomment-383639496

In our product, we required to do Ram and Flash testing. Flash testing is being done using NVS, how to test Ram?
ESP32 WROOM module is used as the replacement for the older microcontroller. In the older microcontroller, these tests have been carried out.

negativekelvin, 2018-04-29:
What problems do you want to test for?  That every byte of RAM passes a certain number of r/w cycles?  Do you plan to test at boot time or only during manufacturing?  How does an NVS test prove all flash sectors are ok?

hvishu, 2018-04-30:
During boot up (not running time), we would like to test every bit of available RAM by writing 0xAA and 0x55 to every byte.  And reading them back.

Using NVS, we are testing only the flash sectors used as non-volatile storage used by the application, not the entire flash region.

These two tests are done, as same were performed on the previous microcontroller used for the same project.  

Please suggest whether these kinds of tests will help on ESP to find faulty RAM/ Flash sectors.