espressif/esp-idf #1897: NVS partition size in application

hvishu, 2018-04-29:
Raising this issue as per the suggestion from the link https://github.com/aws/aws-iot-device-sdk-embedded-C/issues/163

Is there a way to get the NVS partition size from the application (from source code of examples/get-started/hello_world/main/hello_world_main.c

negativekelvin, 2018-04-29:
https://esp-idf.readthedocs.io/en/latest/api-reference/storage/spi_flash.html#_CPPv2N15esp_partition_t4sizeE

igrr, 2018-05-02:
@hvishu As @negativekelvin says, and as you have [already found yourself](https://github.com/aws/aws-iot-device-sdk-embedded-C/issues/163#issuecomment-383009805), you can use `esp_partition_find_first` function to obtain a pointer to `esp_partition_t` structure describing NVS partition. From there, you can get the partition size in bytes.

Please use esp32.com forum for "how do I..." kind of questions and use this issue tracker for reporting bugs in ESP-IDF or for feature requests.

closed 2018-05-02 by igrr