espressif/esp-idf #1898: ulp example compile issue

chegewara, 2018-04-29:
With latest commit (dont know how far backward) there is issue with compiling ulp example. Logs:

```
CPP main/ulp/pulse_cnt.S
ULP_AS build/main/pulse_cnt.ulp.o
make[1]: esp32ulp-elf-as: Cant be found
make[1]: *** [/d/msys/home/admin/esp/esp-idf/components/ulp/component_ulp_common.mk:41: pulse_cnt.ulp.o] error 127
make: *** [/d/msys/home/admin/esp/esp-idf/make/project.mk:467: component-main-build] error 2
```

Environment:
- windows 10, msys2

```
xtensa-esp32-elf-gcc -v
Using built-in specs.
COLLECT_GCC=D:\msys\opt\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe
COLLECT_LTO_WRAPPER=d:/msys/opt/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/lto-wrapper.exe
Target: xtensa-esp32-elf
Configured with: /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/configure --build=x86_64-build_pc-linux-gnu --host=i686-host_pc-mingw32 --ta    rget=xtensa-esp32-elf --prefix=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf --with-local-prefix=/builds/idf/crosstool-NG/builds/xtensa-    esp32-elf/xtensa-esp32-elf/sysroot --with-sysroot=/builds/idf/crosstool-NG/builds/xtensa-esp32-elf/xtensa-esp32-elf/sysroot --with-newlib -    -enable-threads=no --disable-shared --with-pkgversion='crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5' --disable-__cxa_atexit --enable-cxx-f    lags='-fno-rtti -ffunction-sections' --with-gmp=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools/complibs-host --with-mpfr=/buil    ds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools/complibs-host --with-mpc=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools    /complibs-host --with-isl=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools/complibs-host --with-cloog=/builds/idf/crosstool-NG/.    build/xtensa-esp32-elf/buildtools/complibs-host --with-libelf=/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/buildtools/complibs-host --e    nable-lto --enable-target-optspace --without-long-double-128 --disable-libgomp --disable-libmudflap --disable-libssp --disable-libquadmath     --disable-libquadmath-support --disable-nls --enable-languages=c,c++ --disable-libstdcxx-verbose --enable-threads=posix --enable-gcov-custo    m-rtio
Thread model: posix
gcc version 5.2.0 (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5)

```

redchenjs, 2018-04-29:
You can follow this guide to setup the environment:
[https://github.com/espressif/esp-iot-solution/blob/master/documents/low_power_solution/esp32_ulp_co-processor_and_assembly_environment_setup_en.md](https://github.com/espressif/esp-iot-solution/blob/master/documents/low_power_solution/esp32_ulp_co-processor_and_assembly_environment_setup_en.md)

chegewara, 2018-04-29:
Thanks, will do.

closed 2018-04-29 by chegewara