espressif/esp-idf #1904: esp_ble_gatts_set_attr_value() returns success but doesn't update attribute value right away.

sahilkhanna, 2018-05-01:
I planned on setting some values in the attribute table and then notifying the user about that set value.
I thought I might "set" the value in the attr table and then "get" it as soon as it's set and finally notify.
However calling esp_ble_gatts_get_attr_value() right after esp_ble_gatts_set_attr_value() returns "ESP_OK" but doesn't get the new set value.
From some debugging and observation I found out esp_ble_gatts_set_attr_value() raises a callback event, which then can be traced by ESP_GATTS_SET_ATTR_VAL_EVT(I couldn't find this in the API docs).  However I'm not sure how to get the handle number from the callback. I did however saved the handle number in a static variable whenever I set an attribute but that's more of a duct tape fix (IMO). 
I wonder if there is a more graceful way of achieving the above.

chegewara, 2018-05-01:
http://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_gatts.html?highlight=ESP_GATTS_SET_ATTR_VAL_EVT#_CPPv2N24esp_ble_gatts_cb_param_t12set_attr_valE

http://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_gatts.html?highlight=ESP_GATTS_SET_ATTR_VAL_EVT#_CPPv2N24esp_ble_gatts_cb_param_t28gatts_set_attr_val_evt_paramE

sahilkhanna, 2018-05-01:
Thanks! I guess It's my fault all along. I'll close this issue.

closed 2018-05-01 by sahilkhanna