espressif/esp-idf #1905: [cmake] mconf builds on Windows

dschaefer, 2018-05-01:
Great work with CMake and Ninja! Build times are blazing, especially incremental which was a big issue for Windows.

I do have an issue though. I downloaded the mconf.exe package and added it to my PATH and it runs fine from the command line. When running 'ninja menuconfig' it tries to build mconf and that fails in my environment. Am I missing a step to call the downloaded mconf instead of trying to build it's own?

projectgus, 2018-05-02:
@dschaefer The commit reference above you should fix mconf for Windows on the cmake branch. Please let us know if it doesn't.

closed 2018-05-02 by projectgus

dschaefer, 2018-05-02:
Works great. Thanks! This is going to be fun to work with.