espressif/esp-idf #1907: MCPWM: Expose simultaneous rising/falling edge capture ability

ubergeek801, 2018-05-01:
_(Removed all the template stuff since this applies to the IDF framework itself and not to any particular environment or runtime behavior.)_

The MCPWM peripheral's capture unit has the ability to capture rising and falling edges the the same time, but the IDF's `mcpwm_capture_enable()` implementation doesn't allow it. This is a request to please consider a minor modification to the IDF framework to expose this functionality.

The offending line of code is in `mcpwm.c:661`:

```cpp
    MCPWM[mcpwm_num]->cap_cfg_ch[cap_sig].mode = (1 << cap_edge);
```

which unnecessarily constrains `mode` to have a single bit set.

`cap_edge` is a parameter of type `mcpwm_capture_on_edge_t`, which has this definition:

```cpp
typedef enum {
    MCPWM_NEG_EDGE = 0,  /*!<Capture starts from negative edge*/
    MCPWM_POS_EDGE,      /*!<Capture starts from positive edge*/
} mcpwm_capture_on_edge_t;
```

Interestingly, this is inconsistent with the documentation of `mcpwm_capture_enable()`, which includes this description:

```cpp
 * @param cap_edge set capture edge, BIT(0) - negative edge, BIT(1) - positive edge
```

The most straightforward solution would be to set `mode` directly from the `cap_edge` parameter:

```cpp
    MCPWM[mcpwm_num]->cap_cfg_ch[cap_sig].mode = cap_edge;
```

This would not only allow the caller to specify capture of both edges, but would also restore consistency with the `cap_edge` parameter documentation. It would, however, then be inconsistent with the current definition of `mcpwm_capture_on_edge_t`.

Alternatively, a third enum e.g. `MCPWM_BOTH_EDGES` (= 2) could be added to `mcpwm_capture_on_edge_t`, and then in `mcpwm.c`:

```cpp
    MCPWM[mcpwm_num]->cap_cfg_ch[cap_sig].mode = cap_edge + 1;
```

This would preserve backward compatibility with existing values in `mcpwm_capture_on_edge_t` at the expense of violating the usual conventions of specifying individual bits to be set.

Ultimately, I'd be OK with any solution that allows both rising and falling edges to be captured, as I need to be able to measure the duration of high pulses. Thanks!

vgonet, 2018-11-06:
Hello,
I have exactly the same need as ubergeek801. Would it be possible to implement the changes in mcpwm.c so that rising and falling edges could be detected ?
Thanks and best regards.

koobest, 2018-11-07:
Hi, @vgonet, This feature will soon be merged into master. the change list is as follows:

 ```
typedef enum {
    MCPWM_NEG_EDGE = 0x1,   /*!<Capture starts from negative edge*/
    MCPWM_POS_EDGE = 0x2,   /*!<Capture starts from positive edge*/
    MCPWM_BOTH_EDGE = 0x3,  /*!<Capture starts from both edge*/
} mcpwm_capture_on_edge_t;

float mcpwm_get_duty(mcpwm_unit_t mcpwm_num, mcpwm_timer_t timer_num, mcpwm_operator_t op_num)
{
...
 MCPWM[mcpwm_num]->cap_cfg_ch[cap_sig].mode = cap_edge;
...
}
```
@ubergeek801， thanks for your Thank you for your advice too.

ubergeek801, 2018-11-07:
Glad to see it. Thanks for making this change!

vgonet, 2018-12-26:
Hi @koobest and @ubergeek801,

I've done the modifications in the mcpwm.h and mcpwm.c.

But now, how can we manage the interrupt ? Because the interrupt is called either after a rising edge or a falling edge. I send the event to the queue after receiving the two edges. When I read the queue, I calculate the pulse duration.

`
pulse[0] = ((evt.FallingEdge - evt.RisingEdge) / 1000) * (1000000000 / rtc_clk_apb_freq_get());
`

Most of the time I get a valid pulse duration but sometimes I get a very large duration...

Do you have a better idea how I can do that ?

~~~~
static void IRAM_ATTR isr_handler()
{
  uint32_t mcpwm_intr_status;
  Capture evt;
  static uint8_t index;
  static uint32_t tmp = 0;

  mcpwm_intr_status = MCPWM[MCPWM_UNIT_0]->int_st.val;  // Read interrupt status

  if (mcpwm_intr_status & CAP0_INT_EN)  // Check for interrupt on rising edge or falling edge on CAP0 signal
  {
    if ((index % 2) == 0)  // Rising edge
    {
      tmp = mcpwm_capture_signal_get_value(MCPWM_UNIT_0, MCPWM_SELECT_CAP0);  // Get capture signal counter value
    }
    else  // Falling edge
    {
      evt.RisingEdge = tmp;
      evt.FallingEdge = mcpwm_capture_signal_get_value(MCPWM_UNIT_0, MCPWM_SELECT_CAP0);  // Get capture signal counter value
      evt.sel_cap_signal = MCPWM_SELECT_CAP0;
      xQueueSendFromISR(cap_queue, &evt, NULL);
    }

    index++;
  }

  MCPWM[MCPWM_UNIT_0]->int_clr.val = mcpwm_intr_status;
}
~~~~

Thank you and best regards !!!
Vincent

koobest, 2018-12-27:
Hi, @vgonet 
Is it possible that the count value has overflowed?

koobest, 2018-12-28:
if first interrupt is triggered  by ` Falling edge`, the 'tmp' will be the previous time.

vgonet, 2018-12-28:
Yes, you're absolutely right, but I don't know how to know which edge is detected by the interrupt. Do you have an idea ?

koobest, 2018-12-29:
hi,
'mcpwm_capture_signal_get_edge' will tell you what the edge is, also,you can use 'gpio_get_level ' to know what the edge is.

koobest, 2018-12-29:
It is not recommended to always enable the interrupt. you can enable it when you need to measure the signal and disable it after a measurement is completed. e.g

1. Enable interrupt and wait rising edge(if not rising, wait the next trigger).
2. Store the current counter, and wait the falling edge.
3. Disable the interrupt, and caculate the period.
......

vgonet, 2019-01-07:
Hello,

Thank you for your answer. It works much better now.