espressif/esp-idf #1908: File Transfer over BTClassic (SPP profile) crash

louisthetall, 2018-05-01:
## Environment

- Development Kit: none
- Core: ESP-WROOM32
- IDF version: 1e8ab3d1
- Development Env: Make
- Operating System: Windows 10
- Power Supply: external 5V

## Problem Description

I have recently been working on a project where I have switched over from WiFi to Bluetooth and I am trying to implement and test OTA over BTClassic (SPP profile). To do this, I have made a "tester" device using the bt_spp_initiator example that my main device (based off of bt_spp_acceptor and ota example) connects to and then initiates an ota update and sends the proper bin file. 

### Expected Behavior

Once connected to the main device, the test device sends file information in a json format and then begins sending the file as a stream of data packets.

### Actual Behavior

There seems to be no problem on the main device with starting the OTA update, however after a variable number of packets have been sent, the test device crashes.

### Code to reproduce this issue

**Task that reads and sends the file in packets via esp_spp_write()**
```
void BtTransmit_Init_SendFileTask(char* fileName) {
   xTaskCreate(BtTransmit_Run_SendFileTask, "BtTransmit_SendFile", 4096,
         (void *) fileName, 1, &hSendFileTask);
}
void BtTransmit_Run_SendFileTask(char* fileName) {
   _Bool doneFile = false;
   char buf[256];
   char bufFile[1024];
   int n;
   long sz = 0;
   FILE *fp;
   fp = fopen(fileName, "rb");
   printf("\n%s buffers:\n", fileName);
   if (fp == NULL) {
      printf("Error, file %s not found.", fileName);
   }
   fseek(fp, 0L, SEEK_END);
   sz = ftell(fp);
   fseek(fp, 0L, SEEK_SET);
   ESP_LOGI(TAG, ">> Start sending file:%s", fileName);
   int i = 0;
   for (;;) {
      if (!doneFile) {
         //vTaskDelay(1000 / portTICK_PERIOD_MS);
         if ((n = fread(bufFile, 1, 1024, fp)) > 0) {
            i++;
            printf(" %d - %d\n", n, i);
            ESP_ERROR_CHECK(esp_spp_write(bt_handle, strlen(bufFile), (uint8_t *) bufFile));
         } else {
            doneFile = true;
         }
      } else {
         ESP_LOGI(TAG, "<< Done sending file:%s", fileName);
         vTaskDelay(100 / portTICK_PERIOD_MS);
         vTaskDelete(hSendFileTask);
      }
   }
}
```
## Debug Logs

```
I (21038) BtSppConnInitiator: ESP_SPP_WRITE_EVT - status:0 len=145 cong=0
 1024 - 18
I (22038) BtSppConnInitiator: ESP_SPP_WRITE_EVT - status:0 len=17 cong=0
 1024 - 19
assertion "data != NULL" failed: file "C:/msys32/home/Louis/esp32/esp-idf/components/bt/bluedroid/osi/list.c", line 14
2, function: list_append
abort() was called at PC 0x400d4657 on core 0
0x400d4657: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/st
dlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x4008c9f7:0x3ffd8fc0 0x4008ca23:0x3ffd8fe0 0x400d4657:0x3ffd9000 0x400f69db:0x3ffd9030 0x400fafde:0x3ffd90
50 0x400f7492:0x3ffd90b0
0x4008c9f7: invoke_abort at C:/msys32/home/Louis/esp32/esp-idf/components/esp32/panic.c:648

0x4008ca23: abort at C:/msys32/home/Louis/esp32/esp-idf/components/esp32/panic.c:648

0x400d4657: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/st
dlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x400f69db: list_append at C:/msys32/home/Louis/esp32/esp-idf/components/bt/bluedroid/osi/list.c:214

0x400fafde: btc_spp_write at C:/msys32/home/Louis/esp32/esp-idf/components/bt/bluedroid/btc/profile/std/spp/btc_spp.c:
441
 (inlined by) btc_spp_call_handler at C:/msys32/home/Louis/esp32/esp-idf/components/bt/bluedroid/btc/profile/std/spp/b
tc_spp.c:517

0x400f7492: btc_task at C:/msys32/home/Louis/esp32/esp-idf/components/bt/bluedroid/btc/core/btc_task.c:104

```

stickbreaker, 2018-05-02:
Does `fread()` return a cstr or a byte buffer?
You define `bufFile[]` as a char[1024] and tell `fread()` it has room for 1024 characters plus the implied null terminator. That is 1025 bytes of storage.

Chuck

louisthetall, 2018-05-02:
Ah, that seems to be what was causing the issue. Thanks for pointing it out!

closed 2018-05-02 by louisthetall

ROYMURLI, 2019-04-08:
I tried this example to transfer about 1-2 MB of data but it crashed. How can I send larger size data?

laverdanny, 2019-07-02:
hi there
louisthetall  did you succeed to make OTA via BTclassic ?
regards

ROYMURLI, 2019-07-03:
Yes, I am able to send file via BT classic.

laverdanny, 2019-07-03:
and does it work to update OTA ?

ROYMURLI, 2019-07-03:
Yes, It is working perfectly fine.

testter2, 2020-03-21:
> 
> 
> Yes, It is working perfectly fine.

Could you guide me to any example that does that OTA BLUETOOTH through mobile app ?
i want to see how do you receive the file and write it esp flash.
thanks in advance.

ROYMURLI, 2020-03-23:
Hi @testter2 
The task which I used to update the FW is:-

'''
static void taskFirmwareUpdate(void *pvParameter)
{
    esp_err_t err;
	int ret;
	
	esp_ota_handle_t update_handle = 0 ;
	const esp_partition_t *update_partition = NULL;
        printf("starting main task firmware update \n");
        vTaskDelay(DELAY_IN_MS(500));
	printf("Starting OTA...(%u)", xPortGetFreeHeapSize());
	vTaskDelay(DELAY_IN_MS(500));

	const esp_partition_t *configured = esp_ota_get_boot_partition();
	const esp_partition_t *running = esp_ota_get_running_partition();

	if (configured != running) {
		printf("Configured OTA boot partition at offset 0x%08x, but running from offset 0x%08x",
				configured->address, running->address);
		printf("(This can happen if either the OTA boot data or preferred boot image become corrupted somehow.)");
	}
	printf("Running partition type %d subtype %d (offset 0x%08x)",
			running->type, running->subtype, running->address);

	vTaskDelay(DELAY_IN_MS(1000));

	update_partition = esp_ota_get_next_update_partition(NULL);
	printf("Writing to partition subtype %d at offset 0x%x", update_partition->subtype, update_partition->address);

	assert(update_partition != NULL);

	err = esp_ota_begin(update_partition, OTA_SIZE_UNKNOWN, &update_handle);
	if (err != ESP_OK) 
    {
		printf("esp_ota_begin failed, error=%d", err);
		vTaskDelete(NULL);
	}
	printf("esp_ota_begin succeeded\n-----------------------\n");

	int buff_len = 0;
    while(!flag)
    {
        printf("hello1 \n");
        vTaskDelay(DELAY_IN_MS(10));
    }
	while (flag) 
    {
    	buff_len = strlen(firmware_data);
		printf("buff_len: %d\n", buff_len);// | bytes: %d\n",  buff_len, bytes);
        
        while(!updating_firmware)
        {
            vTaskDelay(DELAY_IN_MS(1));
            if(flag == false)
                break;
        }
        int buff_len1 = 988;
            if(flag == true)
            {
            err = esp_ota_write( update_handle, (const void *)firmware_data, buff_len1);
			if (err != ESP_OK) 
            {
	        	printf("error: esp_ota_write failed! err=0x%x\n", err);
				vTaskDelete(NULL);
			}
            }
	        
            printf("bytes written to ota image: %d\n", buff_len1);
            updating_firmware = false;
    }	

	if (esp_ota_end(update_handle) != ESP_OK) 
    {
		printf("esp_ota_end failed!");
        vTaskDelete(NULL);
	}
        vTaskDelay(DELAY_IN_MS(1000));
	err = esp_ota_set_boot_partition(update_partition);
	if (err != ESP_OK) 
    {
		printf("esp_ota_set_boot_partition failed! err=0x%x", err);
		vTaskDelete(NULL);
	}
}
'''

Let me know if you need any help to understand this.

testter2, 2020-03-24:
> Hi @testter2
> The task which I used to update the FW is:-
>
> ```
>
> static void taskFirmwareUpdate(void *pvParameter)
> {
>         esp_err_t err;
> 	int ret;
> 	esp_ota_handle_t update_handle = 0 ;
> 	const [...]

@ROYMURLI
Could you show me where you receive the bin file from the Bluetooth and save it first?
Do you divide the bin file into chunks of data and where to save that data ?
I really appreciate your help.

testter2, 2020-03-26:
> Hi @testter2
> The task which I used to update the FW is:-
>
> ```
>
> static void taskFirmwareUpdate(void *pvParameter)
> {
>         esp_err_t err;
> 	int ret;
> 	esp_ota_handle_t update_handle = 0 ;
> 	const [...]

@ROYMURLI
For the above code
for the variable flag what does it refer to? where does firmware_data variable get its data?

ROYMURLI, 2020-03-26:
Here "flag" is used to control the incoming data in ESP_SPP_DATA_IND_EVT . You can send the data in the chunks of 988 bytes and then write in a global variable (here "firmware_data[]" used) in ESP_SPP_DATA_IND_EVT this event.  In the above example I used some print statement to debug which u can ignore.

borisnimcevic, 2020-03-26:
Hi @ROYMURLI , I have the same question as @testter2 has. Do you mind showing how does "firmware_data" get downloaded in the first place?

Thank you very much for your time.

ROYMURLI, 2020-03-26:
Hi all,
Let me start from begining -
1- First after BT connection send some command to initiate the above task.
2- Then start sending the data in the chunks of 988 bytes.
3- Now, the data will come in this "ESP_SPP_DATA_IND_EVT" event where you can store in a variable here (firmware_data[]).

my code is below:-
case ESP_SPP_DATA_IND_EVT:
        printf("got a read event, writing %d values\n", write_ptr);
        printf("%d bytes | %s\n",param->data_ind.len, (char *) param->data_ind.data);
        if(strstr((char *)param->data_ind.data, "FWCOMPLETED")!= NULL)
         {
                flag = false;
                updating_firmware = false;
         }
         else
         {
                for(int i=0;i<988;i++)
                {
                    firmware_data[i] = param->data_ind.data[i];
                    printf(",%c",firmware_data[i]);
                    // vTaskDelay(DELAY_IN_MS(0.5));
                }
                flag = true;
                updating_firmware = true;
            }
            
            printf("size of FD %d \n", strlen(firmware_data));
            //strcpy(firmware_data,param->data_ind.data);

            printf("firmware data is %s \n",firmware_data);
            while(updating_firmware)
            {
                vTaskDelay(DELAY_IN_MS(1));
            }
        }

testter2, 2020-03-31:
> Hi all,
> Let me start from begining -
> 1- First after BT connection send some command to initiate the above task.
> 2- Then start sending the data in the chunks of 988 bytes.
> 3- Now, the data will come [...]

thanks for your time
how much time does it takes to transfer  1MB file?

I tried using Realterm to transfer small bin file (about 213,264 bytes of data) through Serial Cable using UART but it crashed

that's my code
`
#include <esp_ota_ops.h>

void updatefirmware()
{
  esp_err_t err;
  int ret;
  esp_ota_handle_t update_handle = 0 ;
  const esp_partition_t *update_partition = NULL;
  //    printf("starting main task firmware update \n");
  delay(500);//   vTaskDelay(DELAY_IN_MS(500));
  printf("Starting OTA...(%u)", xPortGetFreeHeapSize());
  delay(500);//vTaskDelay(DELAY_IN_MS(500));

  const esp_partition_t *configured = esp_ota_get_boot_partition();
  const esp_partition_t *running = esp_ota_get_running_partition();

  if (configured != running) {
    printf("Configured OTA boot partition at offset 0x%08x, but running from offset 0x%08x",
           configured->address, running->address);
    printf("(This can happen if either the OTA boot data or preferred boot image become corrupted somehow.)");
  }
  printf("Running partition type %d subtype %d (offset 0x%08x)",
         running->type, running->subtype, running->address);

  delay(1000);//vTaskDelay(DELAY_IN_MS(1000));

  update_partition = esp_ota_get_next_update_partition(NULL);
  printf("Writing to partition subtype %d at offset 0x%x", update_partition->subtype, update_partition->address);

  assert(update_partition != NULL);

  err = esp_ota_begin(update_partition, OTA_SIZE_UNKNOWN, &update_handle);
  if (err != ESP_OK)
  {
    printf("esp_ota_begin failed, error=%d", err);
    vTaskDelete(NULL);
  }
  printf("esp_ota_begin succeeded\n-----------------------\n");

  //  int buff_len = 0;
  //    while(!flag)
  //    {
  //        printf("hello1 \n");
  //        vTaskDelay(DELAY_IN_MS(10));
  //    }
  //unsigned char testing;
  unsigned long i = 0;
  int counter = 0;
  char firmwaredata[988];
  while (  i < 213264)
  {
    //if(i==0){ Serial.println("waiting");}
    if (Serial.available())
    {
      //Serial.write(Serial.read());
      firmwaredata[counter] = Serial.read();
      //err = esp_ota_write( update_handle, (const void *)(testing), 1);
      Serial.println(i++);
    }
    if (counter == 987)
    {
      Serial.println("Writing");
      err = esp_ota_write( update_handle, (const void *)(firmwaredata), 988);
      counter = 0;
    }
    //      buff_len = strlen(firmware_data);
    //    printf("buff_len: %d\n", buff_len);// | bytes: %d\n",  buff_len, bytes);
    //
    //        while(!updating_firmware)
    //        {
    //            vTaskDelay(DELAY_IN_MS(1));
    //            if(flag == false)
    //                break;
    //        }
    //        int buff_len1 = 988;
    //            if(flag == true)
    //            {
    //            err = esp_ota_write( update_handle, (const void *)firmware_data, buff_len1);
    //      if (err != ESP_OK)
    //            {
    //            printf("error: esp_ota_write failed! err=0x%x\n", err);
    //        vTaskDelete(NULL);
    //      }
    //            }
    //
    //            printf("bytes written to ota image: %d\n", buff_len1);
    //            updating_firmware = false;
  }

  if (esp_ota_end(update_handle) != ESP_OK)
  {
    printf("esp_ota_end failed!");
    //vTaskDelete(NULL);
  }
  delay(1000);//  vTaskDelay(DELAY_IN_MS(1000));
  err = esp_ota_set_boot_partition(update_partition);
  if (err != ESP_OK)
  {
    printf("esp_ota_set_boot_partition failed! err=0x%x", err);
    //vTaskDelete(NULL);
  }
  Serial.println("Resarting");
  delay(10000);
  ESP.restart();
}

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");

}
String datar;
void loop() {

  if (Serial.available())
  {
    datar = Serial.readString();
    Serial.println(datar);
    if (datar == "StartSending")
    {
      Serial.println("StartSending");
      updatefirmware();

    }
  }

 
  delay(1);

}`

ROYMURLI, 2020-03-31:
@testter2  send me your logs also. There is no need for above code if you are flashing your with serial cable/port. You can simply flash by using "make flash" command after building your code.

testter2, 2020-03-31:
> 
> 
> @testter2 send me your logs also. There is no need for above code if you are flashing your with serial cable/port. You can simply flash by using "make flash" command after building your code.

No,i'm just using Serial cable for testing to get the data from bin file and store it in ota partition,But at the end i'll use the SPP of bluetooth to send the bin file.

so i'm asking about an average time to send 1MB bin file through SPP of bluetooth?

here's the logs from the last code
```
48816
48817
48818
48819
                                                                         
ØÄ$ ØD$$ ÈDD$ ØDä$ ØÄG$ØD$Ç$ØDÄÇ$ØÄäÇ4ÈÄ¤ ØD$$ 
Øä´ Øä$$ ØäD¤ Øää´ Èä„Ç$Øä$G$ØäÄÇ$ÈtäÇ$Øt$ Øä$´ 
ØÇ$ ØÇ8;$ØÇ;$È„Ç;$ØGG4ØG;;4ØGÇ4ÈÇÇ$Ø„Ç$ Ø„Ç<;$
Ø´Ç$ Ø$G8;$È$G;4Ø´Ç9$Ø´ÇÇ$Ø$Ç;;$Ø$GG4Ø$GÇ4Ø´Ç$ Ø$Ç<;$
ØÄG´ ÈÄG8;4ØÄÇ;$ØÄÇ;$ØÄÇG$ÈÄG;;4ØÄÇÇ$ØÄÇÇ$ØÄG´ ÈÄ‡<;$
ØäÇ$`ÈäG8;4ØôÇ;$ØäG;$ÈäGÇ$ØäÇ;;$ØäGG4ØôÇÇ$ØäÇ$ ØäG<;$
È´ Ø$$ ØÄ$ Øä$ ØG4Ø¤Ç$ØÄG$ØäG4È¤ Ø$$ 
Ø$$`Ø$$$ Ø$Ä$ Øü#¤Üä''ÃÄ#ô'qÃã'¸ä'Ü¼ #ÃÃ##Ã‰''ß¼Û'''\?[
$Dû;Û´ßÛ  

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee                                
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
csum err:0x62!=0xfc
ets_main.c 371 
Falling back to built-in command interpreter.
OK
>1CPU 0ets Jun  8 2016 00:22:5o
                                              
                                  
                                            
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
The device started, now you can pair it with bluetooth!
    

```

the number at the first of the logs indicates to the byte number.

testter2, 2020-04-01:
i used realterm to send the bin file and i make 10ms delay between every char to just ensure that i receive the whole fie (it really took long time ) but it failed to update the firmware and that the logs i got

```
Starting OTA...(132800)Running partition type 0 subtype 16 (offset 0x00010000)Wr
iting to partition subtype 17 at offset 0x1f0000esp_ota_begin succeeded
-----------------------
Finished i= 213264
esp_ota_end failed!
E (2948956) esp_image: image at 0x1f0000 has invalid magic byte
E (2948956) boot_comm: image has invalid chip ID, expected at least 0, found 655
35
E (2948958) boot_comm: image has invalid chip revision, expected at least 1, fou
nd 255
                                    
```

tonilopezmr, 2020-06-19:
@testter2 Did you finish the Serial Bluetooth OTA? Did you find some problems doing it? 
Thanks!!