espressif/esp-idf #1911: Problem with sampling using ADC via DMA

Splediferous, 2018-05-02:
- Development Kit:     Platform IO
- Kit version (for WroverKit/PicoKit/DevKitC): v3.1-dev-661-gf586f5e6
- Core (if using chip or module): ESP32 Lolin
- IDF version                    v3.1-dev-661-gf586f5e6
- Development Env:		VS Code + PlatformIO
- Operating System:		Windows
- Power Supply:                USB

Platform Espressif 32 (Stage)
--------
Updating espressif32                                  @ 29b3a81
Updating tool-esptoolpy                            @ 1.20100.0
Updating framework-arduinoespressif32    @ 6826508
Updating tool-espotapy                             @ 1.1.0

Hi all.
This is a [cross-post from esp32.com](https://esp32.com/viewtopic.php?f=13&t=5522) since it didn't get any love over there.

I'm trying to acquire a single channel analogue input using the I2S DMA. This is somewhat successful in that I can configure the I2S and read data correctly and it gives me sensible values. However. I seem to be getting two channels of the same measurement, separated by 1/f, instead of one.

The image below shows some waveforms with their resultant FFTs. The top waveform is a plot of the buffer (with adjustment to remove the DC) assuming it is a single channel (f=120KHz sampling rate) with a signal source of about 10KHz being read by the ADC. The FFT yields 2 major peaks at about 10KHz and at about (f/2)-10KHz. These FFT signals converge until they overlap and then proceed to diverge as the test signal is increased, so they are definitely related to the single signal.

[image]

The bottom two are the result of the same data but de-interleaved. The raw measurements (buffer dump of the i2s_read_bytes) can be found [here](https://pastebin.com/Kuj8UMcu). For these two plots, I am assuming that we have two channels, each sampling at f/2 which yield the expected results.

So it seems I have two interleaved channels, of the same signal, at f/2 sample rate that are "out of phase" by 1/fs. These two channels are then interleaved into the read buffer as if it were a single signal.
By de-interleaving the buffer and then shifting the second signal by 1/fs (delete the first sample from the array of channel 2) and then interleave back, I now get the signal I expect. 
[image]

Can anyone explain this behaviour? Is there some configuration setting I'm missing?

[Code to reproduce](https://pastebin.com/QxL2msG6): [Note: a physical connection is required between pins 25 and  34]

Splediferous, 2018-05-19:
So. No one is going to suggest it is an IQ signal? 

There seems to be no documentation for this or how it basically halves the requested sample rate. How does it work with multiple channels? Does the sample rate halve for each channel acquired?

phonec, 2018-05-20:
For example, if add after i2s_adc_enable():

SYSCON.saradc_ctrl.sar1_patt_len = 3;
SYSCON.saradc_sar1_patt_tab[0] = 0x0e3e6e7e;
SYSCON.saradc_ctrl2.sar1_inv = 1;

This can scan 0,3,6,7 ADC1 with a 12-bit resolution and 6 dB attenuation. In this case, the sampling frequency remains the same f/2 for each channel with the same distortion.

phonec, 2018-05-21:
#include "soc/syscon_reg.h"
#include "soc/syscon_struct.h"

Splediferous, 2018-05-21:
@phonec 

> For example, if add after i2s_adc_enable():
>
> SYSCON.saradc_ctrl.sar1_patt_len = 3;
> SYSCON.saradc_sar1_patt_tab[0] = 0x0e3e6e7e;
> SYSCON.saradc_ctrl2.sar1_inv = 1;
>
> This can scan 0,3,6,7 ADC1 with a [...]

That doesn't seem to do anything. It is still a single channel.

koobest, 2018-12-17:
Hi, @Splediferous
Have you solved the issue?

Alvin1Zhang, 2019-03-25:
@Splediferous Hi, would you help share if any updates for this issue? Thanks.

shansted, 2019-03-27:
Are you sure this is an issue with the ADC. Haven't tested fully yet, but found that all appeared OK from the ADC using my scope's test square wave (Added a cap to get a sloping at the start of the waveform). My scope also has crazy jitter trying to view the DAC signal from the ESP32. Have you sampled other waveforms apart from the DAC cosine? Will test with another, as soon as I get the ad9833 on my bench going.

Splediferous, 2019-03-27:
Sorry for the delay in replying. We went with another product in the end.

I haven't had the opportunity to revisit this but I will in the next couple of weeks and report back.

T1bolus, 2019-04-10:
Hey,
I had exactly the same issue like @Splediferous in his first post. ADC with I2S & DMA works fine but the results are very weird. I made the same experience:

> So it seems I have two interleaved channels, of the same signal, at f/2 sample rate that are "out of phase" by 1/fs. These two channels are then interleaved into the read buffer as if it were a single signal.

This was my first raw measurement with 16kHz:
[image]

I measured a 1 kHz reference signal from an oscilloscope. It was noticeable that some values are repeated after a certain period of time. This must be the "interleaved channels" problem. 
So I added more channels to the ADC pattern table, that it can't be a pattern table reading problem. 

```
    //[...]
    i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);

    adc_gpio_init(ADC_UNIT_1, ADC_CHANNEL_1);
    adc_gpio_init(ADC_UNIT_1, ADC_CHANNEL_3);
    adc_gpio_init(ADC_UNIT_1, ADC_CHANNEL_6);
    adc_gpio_init(ADC_UNIT_1, ADC_CHANNEL_7);
    

    i2s_set_adc_mode(ADC_UNIT_1, ADC1_CHANNEL_7); //Channel doenst care, will be overridden later

    // ADC-I2S Scanning does not normal start (more badly) after the reset of ESP32. 
    // It is necessary to add a delay of 5 seconds before switching on the ADC for a reliable start, this is issue.
    vTaskDelay(5000/portTICK_RATE_MS);

    i2s_adc_enable(I2S_NUM_0);

    //Enable Multipattern scanning
    SYSCON.saradc_ctrl.sar1_patt_len = 1; //2 Items in pattern table
    SYSCON.saradc_sar1_patt_tab[0] = 0x7f3f6f1f; //Array[0] first 4 items, Array[1] next 4 items...
                                                 ///ch_sel[3:0] bit_width[1:0] atten[1:0]
    SYSCON.saradc_ctrl2.sar1_inv = 1; //Inversion
```

Now my measurement looks much cleaner.

[image]

But wait, again this repeating values issue. And I've added another Channel, so one period of this 1 kHz signal must have 16/2 = 8 values. But it has 16. I found an ?error? in my config.

I've changed 
`.channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,`
to
`.channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,`

and voillà
[image]

Increasing the sample rate to 32kHz and adding another 2 channels also works fine now.

My full I2S Init is now:

```
void I2Sinit(uint32_t sample_rate)
{
    i2s_config_t i2s_config = 
    {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN),
    .sample_rate =  (int)(sample_rate),
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
    .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
    .communication_format = I2S_COMM_FORMAT_I2S_MSB,
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
    .dma_buf_count = 2,
    .dma_buf_len = 1024,
    .use_apll = false,
    };

    i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);

    adc_gpio_init(ADC_UNIT_1, ADC_CHANNEL_1);
    adc_gpio_init(ADC_UNIT_1, ADC_CHANNEL_3);
    adc_gpio_init(ADC_UNIT_1, ADC_CHANNEL_6);
    adc_gpio_init(ADC_UNIT_1, ADC_CHANNEL_7);
    

    i2s_set_adc_mode(ADC_UNIT_1, ADC1_CHANNEL_7); //Channel doenst care, will be overridden later

    // ADC-I2S Scanning does not normal start (more badly) after the reset of ESP32. 
    // It is necessary to add a delay of 5 seconds before switching on the ADC for a reliable start, this is issue.
    vTaskDelay(5000/portTICK_RATE_MS);

    i2s_adc_enable(I2S_NUM_0);

    //Enable Multipattern scanning
    SYSCON.saradc_ctrl.sar1_patt_len = 3; //4 Items in pattern table
    SYSCON.saradc_sar1_patt_tab[0] = 0x7f3f6f1f; //Array[0] first 4 items, Array[1] next 4 items...
                                                 ///ch_sel[3:0] bit_width[1:0] atten[1:0]
    SYSCON.saradc_ctrl2.sar1_inv = 1; //Inversion
}
```

hopefully i could help someone :)

zekageri, 2019-09-16:
That does not works for me. I still get strange singals. I'am measuring a 4khz signal from a signal generator with my esp32 with 80.000hz. My signals looks just like yours before your corrections. 
My i2s init:

```
void configure_i2s() {
  i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_TX | I2S_MODE_ADC_BUILT_IN),                                        // I2S receive mode with ADC
    .sample_rate = samplingFrequency,                      // sample rate (80000)
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,          // 16 bit I2S
    .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,           // only the right channel
    .communication_format = I2S_COMM_FORMAT_I2S,           // I2S format
    .intr_alloc_flags = 0,                                 // none
    .dma_buf_count = 8, //8                               // number of DMA buffers
    .dma_buf_len = NUM_SAMPLES,                            // number of samples 100
    .use_apll = 0,                                         // no Audio PLL
  };
  // ADC channel 0 with 11db (divide by input 3.6)
  adc1_config_channel_atten(ADC_CHANNEL, ADC_ATTEN_11db);
  // 12 bit ADC
  adc1_config_width(ADC_WIDTH_12Bit);
  // install I2S 0 driver, using event queue
  i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
  // ADC should use ADC_CHANNEL
  i2s_set_adc_mode(ADC_UNIT_1, ADC_CHANNEL);
  // The raw ADC data is written in DMA in inverted form. This add aninversion:
  SET_PERI_REG_MASK(SYSCON_SARADC_CTRL2_REG, SYSCON_SARADC_SAR1_INV);
  // enable I2S ADC
  vTaskDelay(5000/portTICK_RATE_MS);
  i2s_adc_enable(I2S_NUM_0);
  olvasunk_e = true;
  Serial.println("I2S started");
}
```

**And the ADC Sampling function where i split the data to the web:**

```
static const inline void ADC_Sampling(){
  //portENTER_CRITICAL_ISR(&timerMux);
  uint16_t i2s_read_buff[NUM_SAMPLES];
  int num = i2s_read_bytes(I2S_NUM_0, (char*)i2s_read_buff,  NUM_SAMPLES * sizeof(uint16_t), portMAX_DELAY);
  //collected_samples += num / sizeof(uint16_t);
  if(millis() - getfastdatamillis >= Frissit && Chart_is_ok_to_Send_Data && olvasunk_e){
    getfastdatamillis = millis();
    for (int i=0;i<NUM_SAMPLES;i++) {
      String json = "";
      json += String((i2s_read_buff[i] & 0xFFF)/ (float)40.95);
      json += "}";
      webSocket.broadcastTXT(json.c_str(), json.length());
    }
  }
  //portEXIT_CRITICAL_ISR(&timerMux);
}
```

**The signals that i'am getting:**

[image]
[image]
[image]

T1bolus, 2019-09-24:
> That does not works for me. I still get strange singals. I'am measuring a 4khz signal from a signal generator with my esp32 with 80.000hz. My signals looks just like yours before your corrections.

Have you added multipattern scanning? I had the problem when only one channel is in this pattern table. So I needed at least two channels.

zekageri, 2019-09-26:
I just had to modify the channel format to : 

`.channel_format = I2S_CHANNEL_FMT_ALL_LEFT,`

The signals are okay but now i have a buffer overflow issue. :'D

zekageri, 2019-09-26:
> Have you added multipattern scanning? I had the problem when only one channel is in this pattern table. So I needed at least two channels.

I just had to modify the channel format to : 

`.channel_format = I2S_CHANNEL_FMT_ALL_LEFT,`

The signals are okay but now i have a buffer overflow issue. :'D

zekageri, 2019-10-09:
_The buffer overflow issue has been solved by separating the i2s read and the plotting functions to the two cores._

**My i2s init now:**

```
void configure_i2s(){
  i2s_config_t i2s_config = 
    {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN),  // I2S receive mode with ADC
    .sample_rate = samplingFrequency,                                             // sample rate 153260hz idk why but works with 4khz signals O_o
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,                                 // 16 bit I2S
    .channel_format = I2S_CHANNEL_FMT_ALL_LEFT,                                   // all the left channel
    .communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),   // I2S format
    .intr_alloc_flags = 0,                                                        // none
    .dma_buf_count = 2,                                                           // number of DMA buffers
    .dma_buf_len = 1024,                                                   // number of samples
    .use_apll = 0,                                                                // no Audio PLL
  };
  adc1_config_channel_atten(ADC_CHANNEL, ADC_ATTEN_11db);
  adc1_config_width(ADC_WIDTH_12Bit);
  i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
 
  i2s_set_adc_mode(ADC_UNIT_1, ADC_CHANNEL);
  SET_PERI_REG_MASK(SYSCON_SARADC_CTRL2_REG, SYSCON_SARADC_SAR1_INV);
  i2s_adc_enable(I2S_NUM_0);
  vTaskDelay(1000);
}
```

**ADC i2s read function on core 1:** 
just like that: O_o
```
uint16_t i2s_read_buff[NUM_SAMPLES];

static const inline void Sampling(){
    i2s_read(I2S_NUM_0, (char*)i2s_read_buff,NUM_SAMPLES * sizeof(uint16_t), &bytes_read, portMAX_DELAY);
}
```

**Reading and plotting function to web on core 0:**

```
static const inline void Plotting(uint16_t* Buffer){
   String data;
  if (Chart_is_ok_to_Send_Data) // if user opened the chart page
  {
    for(int i = 0;i<NUM_SAMPLES;i++)
    {
      data = String((Buffer[i])/(float)40.95); // divide for % representation
      if(olvasunk_e){  // turn on/off flag for visualization
      data += "}";
      webSocket.broadcastTXT(data.c_str(), data.length());
      }
    }
  }
}
```

rkit1, 2019-12-31:
SYSCON.saradc_ctrl2.meas_num_limit = 0; 
This will help with issues with random skipping and reordering of samples.

aguaviva, 2021-04-14:
I am having similar issues. Too sad this issue got so little attention.

Alvin1Zhang, 2023-09-05:
Thanks for reporting, feel free to reopen.

closed 2023-09-05 by Alvin1Zhang

raulamd, 2023-09-05:
Is this fixed?

L-KAYA, 2023-09-08:
The issue looks like mainly related to the I2S configuration. Due to the limitation of the legacy driver, `channel_format` and `communication_format` have to be set to a particular value to avoid the adjacent data swapped. Glad to see that the comments above have give out the solution ([solution 1](https://github.com/espressif/esp-idf/issues/1911#issuecomment-481658770), [solution 2](https://github.com/espressif/esp-idf/issues/1911#issuecomment-539951841)). 

It should have been fixed in the new `esp_adc` since v5.0 driver, the new driver uses fixed I2S slot format instead of configuring by users. Suggest to migrate to the version that >= v5.0 to have a try.