espressif/esp-idf #1912: aws_iot thing shadow x509 cert verify failed

nvl1109, 2018-05-02:
## Environment

- Development Kit:      [ESP-WROOM32]
- Core (if using chip or module): [ESP-WROOM32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.):  0d65f3b (0d65f3b7f9a1dace8853719871d920d80aa3604b)
- Development Env:		[esp-idf]
- Operating System:		[Windows WSL]
- Power Supply:         [USB]

## Problem Description

The example aws_iot/thing_shadow does not work. I follow the readme instructions of thing_shadow example, but it is failed to verify the certificate.

### Steps to repropduce

1. create new IoT thing in AWS IoT, also create new certificate for that thing.
2. Download the certificate.pem.crt and private.pem.key files, place them into main/certs folder.
3. In menuconfig, setup wifi, THING_NAME, embedded certs, aws hostname, enable debug

Please see following pics:
https://photos.app.goo.gl/EeCBF1KVd7NucI2i2
https://photos.app.goo.gl/872qOa8fq8SWIXyW2
https://photos.app.goo.gl/Cqxm2G5PlBlMFR0I2
https://photos.app.goo.gl/LQKIqxW3M8FMVcAf1

## Debug Logs

```
D (4376) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: 0067

D (4386) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: c009

D (4396) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: c013

D (4396) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: 0033

D (4406) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: c0ae

D (4416) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: c0a2

D (4416) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: 009d

D (4426) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: c09d

D (4436) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: 003d

D (4436) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: 0035

D (4446) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: c032

D (4456) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: c02a

D (4456) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: c00f

D (4466) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: c02e

D (4476) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: c026

D (4476) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: c005

D (4486) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: c0a1

D (4486) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: 009c

D (4496) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: c09c

D (4506) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: 003c

D (4506) mbedtls: ssl_cli.c:887 client hello, add ciphersuite: 002f

[... 247 lines cut ...]
I (6296) mbedtls: ssl_tls.c:2701 => write record

D (6306) mbedtls: ssl_tls.c:2838 output record: msgtype = 21, version = [3:3], msglen = 2

I (6316) mbedtls: ssl_tls.c:2416 => flush output

I (6316) mbedtls: ssl_tls.c:2435 message length: 7, out_left: 7

I (6326) mbedtls: ssl_tls.c:2441 ssl->f_send() returned 7 (-0xfffffff9)

I (6336) mbedtls: ssl_tls.c:2460 <= flush output

I (6336) mbedtls: ssl_tls.c:2850 <= write record

I (6346) mbedtls: ssl_tls.c:4137 <= send alert message

D (6346) mbedtls: ssl_tls.c:4675 ! Certificate verification flags 200

I (6356) mbedtls: ssl_tls.c:4684 <= parse certificate

I (6366) mbedtls: ssl_tls.c:6567 <= handshake

E (6366) aws_iot: failed! mbedtls_ssl_handshake returned -0x2700
E (6376) aws_iot:     Unable to verify the server's certificate.
```

dhrishi, 2018-05-02:
@nvl1109 Can you please check if the certificate date/validity is correct. In general, certificate verification flag 0x200 indicates error MBEDTLS_X509_BADCERT_FUTURE. This means that the certificate validity starts in the future.

dhrishi, 2018-05-03:
@nvl1109 Ok. Is the device (ESP32) time correct too?

anujdeshpande, 2018-05-03:
Also, please make sure that you have not just created the certificate but also activated it. It's easy to miss in the AWS IoT UI

nvl1109, 2018-05-03:
Hi @anujdeshpande ,
The certificate is already activated & attached to the device.

Hi @dhrishi ,
Thank you for the suggestion, I will check the time then feedback later. But about the code, I don't modify anything from aws_iot/thing_shadow example from esp-idf.

anujdeshpande, 2018-05-03:
@nvl1109 Can you please attach the screenshots of your menuconfig again? Seems like something is wrong with them - can't seem to access them

nvl1109, 2018-05-03:
@anujdeshpande I have update the image link, it is ok now. Thank you.
@dhrishi I have check the time, the ESP32 got current time from server ``1525278825`` you can see in log above, it is Epoch time --> time at that moment is
```GMT: Wednesday, May 2, 2018 4:33:45 PM
Your time zone: Wednesday, May 2, 2018 6:33:45 PM GMT+02:00 DST```
it is correct. Then I think no problem with time.
Do you have any other idea?

mahavirj, 2019-04-18:
@nvl1109 Sorry for getting back so late, but did you find any resolution for this issue? (Apparently this was not reproducible at our end)

nvl1109, 2019-04-18:
Hi @mahavirj 
I'm totally forgot about this :). I even didn't remember the issue was solved or not :( 
So, maybe marks this ticket as closed?

closed 2019-04-18 by mahavirj