espressif/esp-idf #1913: Image contains multiple IROM when booting up

eminarcissus, 2018-05-03:
I'm building a customized micro python build from ESP32 port, when it gets booting up, I'm seeing the error like this

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4436
load:0x40078000,len:0
load:0x40078000,len:11816
entry 0x4007a9fc
E (514) boot: Image contains multiple IROM segments. Only the last one will be mapped.

But the partition table is quite simple like this

# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 0x180000,

And I've searched around and never see someone have the same error yet. What can be the cause of the problem here?

This is the building message I've got after compiling the firmware.

LINK build/application.elf
   text	   data	    bss	    dec	    hex	filename
 957334	 295205	  99152	1351691	 14a00b	build/application.elf
Create build/application.bin
esptool.py v2.2.1
Create build/firmware.bin
bootloader     16352
partitions      3072
application  1363664
total        1429200

Any hints about the reason for this error?

projectgus, 2018-05-03:
Hi @eminarcissus,

```
load:0x40078000,len:0
load:0x40078000,len:11816
entry 0x4007a9fc
E (514) boot: Image contains multiple IROM segments. Only the last one will be mapped.
```

This error message has something to do with the linker script being set up in a way which generates an unusual .elf file. It this case it looks like there's a phantom 0 length section at 0x40078000, which esptool has copied into the .bin file and made the bootloader confused...

That said, although this logged as an Error it should probably be a Warning as it won't stop the boot by itself. However, the IROM segment is only 11KB which is suspciously small for any ESP-IDF-based app. So I think there are probably some other problems in the linker script causing an invalid .elf file to be produced.

Micropython have their own build system and (I believe) linker script for ESP-IDF, so you may need to start there (and/or with any customizations) to look for a root cause.

eminarcissus, 2018-05-03:
Gotcha, will check it my makefile, thx very much for the help :)

projectgus, 2018-05-04:
No problems. I am going to close this as it's probably not a bug in ESP-IDF, but I wish you luck sorting it out!

If you're still stuck then https://esp32.com forums or the MicroPython docs/forum/github can be good places to continue the discussion.

closed 2018-05-04 by projectgus