espressif/esp-idf #1914: `ledc_set_fade_with_time` returns error in sub-call to `ledc_set_fade` using specific values.

sigwinch28, 2018-05-04:
## Environment

- Development Kit: Wemos LOLIN32 v1.0.0
- IDF version: 051d8d69
- Development Env:	 Make
- Operating System: Debian Testing
- Power Supply: USB. Reproducible on external 12v supply regulated down to 5V using an LM7805C

## Problem Description

PWM fading does not work with certain values, which seems to depend on the PWM frequency, fade time, and target duty. An error occurs when performing some fades between small values over a long period of time.

### Expected Behavior

No error is returned, and the hardware fade occurs as expected.

### Actual Behavior

An error occurs in the call to `ledc_set_fade_with_time`:
```
E (320) ledc: step_num=40 duty_cyle_num=2458 duty_scale=1
```

### Steps to repropduce

1. copy `examples/peripherals/ledc` to a new directory
2. replace `ledc_example_main.c` with the code below
3. `make flash monitor`

There is no external wiring. Has also been tested in-circuit with mosfets driving LED strips.

### Code to reproduce this issue

```c
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/ledc.h"
#include "esp_err.h"
#include "esp_log.h"

#define LEDC_HS_TIMER          LEDC_TIMER_0
#define LEDC_HS_MODE           LEDC_HIGH_SPEED_MODE
#define LEDC_HS_CH0_GPIO       (18)
#define LEDC_HS_CH0_CHANNEL    LEDC_CHANNEL_0

#define LEDC_TEST_DUTY         (40)
#define LEDC_TEST_FADE_TIME    (3000)

#define LEDC_TEST_FREQUENCY    (32768)
#define LEDC_TEST_RESOLUTION   LEDC_TIMER_10_BIT

void app_main()
{
    ledc_timer_config_t ledc_timer = {
        .duty_resolution = LEDC_TEST_RESOLUTION, // resolution of PWM duty
        .freq_hz = LEDC_TEST_FREQUENCY,          // frequency of PWM signal
        .speed_mode = LEDC_HS_MODE,              // timer mode
        .timer_num = LEDC_HS_TIMER               // timer index
    };
    // Set configuration of timer0 for high speed channels
    ledc_timer_config(&ledc_timer);

    ledc_channel_config_t ledc_channel = {
      .channel    = LEDC_HS_CH0_CHANNEL,
      .duty       = 0,
      .gpio_num   = LEDC_HS_CH0_GPIO,
      .speed_mode = LEDC_HS_MODE,
      .timer_sel  = LEDC_HS_TIMER
    };

    // Set LED Controller with previously prepared configuration
    ledc_channel_config(&ledc_channel);

    ESP_LOGI("example", "setting fade");
    ESP_ERROR_CHECK(ledc_set_fade_with_time(ledc_channel.speed_mode,
					    ledc_channel.channel,
					    LEDC_TEST_DUTY,
					    LEDC_TEST_FADE_TIME));
    ESP_LOGI("example", "starting fade");
    ESP_ERROR_CHECK(ledc_fade_start(ledc_channel.speed_mode,
				    ledc_channel.channel,
				    LEDC_FADE_NO_WAIT));
}
```

## Log
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5640
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:14324
entry 0x40078620
I (30) boot: ESP-IDF v3.1-dev-890-g051d8d69 2nd stage bootloader
I (30) boot: compile time 02:21:40
I (30) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x05d1c ( 23836) map
I (103) esp_image: segment 1: paddr=0x00015d44 vaddr=0x3ffb0000 size=0x021c4 (  8644) load
I (107) esp_image: segment 2: paddr=0x00017f10 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/joe/src/esp-idf/components/freertos/xtensa_vectors.S:1685

I (113) esp_image: segment 3: paddr=0x00018318 vaddr=0x40080400 size=0x07cf8 ( 31992) load
I (134) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x12640 ( 75328) map
0x400d0018: _flash_cache_start at ??:?

I (161) esp_image: segment 5: paddr=0x00032660 vaddr=0x400880f8 size=0x00bd0 (  3024) load
0x400880f8: DPORT_REG_READ at /home/joe/src/esp-idf/components/soc/esp32/rtc_clk.c:441
 (inlined by) rtc_clk_cpu_freq_set at /home/joe/src/esp-idf/components/soc/esp32/rtc_clk.c:497

I (163) esp_image: segment 6: paddr=0x00033238 vaddr=0x400c0000 size=0x00000 (     0) load
I (173) boot: Loaded app from partition at offset 0x10000
I (174) boot: Disabling RNG early entropy source...
I (180) cpu_start: Pro cpu up.
I (183) cpu_start: Starting app cpu, entry point is 0x40081084
0x40081084: call_start_cpu1 at /home/joe/src/esp-idf/components/esp32/cpu_start.c:224

I (0) cpu_start: App cpu up.
I (194) heap_init: Initializing. RAM available for dynamic allocation:
D (200) heap_init: New heap initialised at 0x3ffae6e0
I (206) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
D (212) heap_init: New heap initialised at 0x3ffb29f0
I (217) heap_init: At 3FFB29F0 len 0002D610 (181 KiB): DRAM
I (223) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (229) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (236) heap_init: New heap initialised at 0x40088cc8
I (241) heap_init: At 40088CC8 len 00017338 (92 KiB): IRAM
I (247) cpu_start: Pro cpu start user code
D (259) clk: RTC_SLOW_CLK calibration value: 3045363
D (269) intr_alloc: Connected src 46 to int 2 (cpu 0)
D (269) intr_alloc: Connected src 57 to int 3 (cpu 0)
D (270) intr_alloc: Connected src 24 to int 9 (cpu 0)
I (275) cpu_start: Starting scheduler on PRO CPU.
D (0) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (0) cpu_start: Starting scheduler on APP CPU.
D (290) heap_init: New heap initialised at 0x3ffe0440
D (300) heap_init: New heap initialised at 0x3ffe4350
D (310) intr_alloc: Connected src 16 to int 12 (cpu 0)
D (310) ledc: LEDC_PWM CHANNEL 0|GPIO 18|Duty 0000|Time 0
I (320) example: setting fade
E (320) ledc: step_num=40 duty_cyle_num=2458 duty_scale=1
D (330) ledc: cur duty: 0; target: 40, step: 40, cycle: 2458; scale: 1

I (330) example: starting fade
```

## Suspected Origin

It appears to be down to this calculation inside `ledc_set_fade_with_time` (`ledc.c`, L538):

```C
if (total_cycles > duty_delta) {
    scale = 1;
    cycle_num = total_cycles / duty_delta;
}
```
which produces a call to `ledc_set_fade_with_step`. All `LEDC_CHECK` calls pass, then `ledc_set_fade` is called where the check (on L323):

```C
step_num > LEDC_DUTY_NUM_HSCH0_V || duty_cyle_num > LEDC_DUTY_CYCLE_HSCH0_V || duty_scale > LEDC_DUTY_SCALE_HSCH0_V
```

returns true for `duty_cyle_num > LEDC_DUTY_CYCLE_HSCH0_V`.

costaud, 2018-05-08:
Thanks for the so detailed report, we are fixing this.

costaud, 2018-05-14:
The fix is under merge process, will be merged into master branch soon.

closed 2018-07-23 by FayeY