espressif/esp-idf #1915: libpp.a - undefined reference to `wifi_create_queue'

hutualive, 2018-05-04:
----------------------------------------------------------------------------------------
## Environment
- IDF version:                           [c1c49b635a6c59b4ca0c4a285e490545241e7a73]
- Development Env:		[make]
- Operating System:		[windows/msys2]
-----------------------------------------------------------------------------------
## Problem Description
during make process of the project, during the link stage, the error reported : undefined reference to 'wifi_create_queue' in libpp.a.

I checked the repo, inside components/esp32/wifi_internal.c, there is the implementation of this function:
wifi_static_queue_t* wifi_create_queue( int queue_len, int item_size)
{
    wifi_static_queue_t *queue = NULL;

    queue = (wifi_static_queue_t*)heap_caps_malloc(sizeof(wifi_static_queue_t), MALLOC_CAP_INTERNAL|MALLOC_CAP_8BIT);
    if (!queue) {
        return NULL;
    }

but do not find the declaration of this function, is this the issue for this error ? if so, which header file should I put this function declaration hence the libpp.a know the reference ?

or it's due to other issues that need solve by the libpp.a maintainer ?

thanks.

LINK build/application.elf
C:/msys32/home/dp/esp-idf/components/esp32/lib\libpp.a(pp.o):(.text.pp_attach+0x24): undefined reference to `wifi_create_queue'
C:/msys32/home/dp/esp-idf/components/esp32/lib\libpp.a(pp.o): In function `pp_attach':
(.text.pp_attach+0xfe): undefined reference to `wifi_create_queue'
make: *** [GNUMakefile:669：build/application.elf] 错误 1

jp-96, 2018-05-05:
git clone --recursive https://github.com/espressif/esp-idf.git
cd esp-idf
git submodule update --init --recursive

hutualive, 2018-05-07:
@jp-96 thanks.
issue solved. indeed after checkout to the specified esp-idf commit(c1c49b6), I need do the submodule update too, as the specified commit depend on specified submodule commit(here is libpp.a library)too.

closed 2018-05-07 by hutualive