espressif/esp-idf #1917: esp_pm_dump_locks command causing interrupt watchdog timeout crash

mikaelkanstrup, 2018-05-04:
## Environment

- Development Kit:      [NodeMCU-32S]
- Core (if using chip or module): [ESP32D0WDQ6 (revision 1)]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 0d65f3b
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

## Problem Description

Function esp_pm_dump_locks writes to a file descriptor under critical section. This results in an interrupt watchdog timeout if the underlying FS driver is interrupt based. An example where this problem is seen is when calling the function with stdout file descriptor and stdout being registered to a UART with esp_vfs_dev_uart_use_driver.

### Expected Behavior

No crash expected

### Actual Behavior

Watchdog crash seen

### Steps to reproduce

1. Apply attached patch to esp-idf
2. Enable the following config flags and build examples/system/console
CONFIG_PM_ENABLE=y
CONFIG_PM_PROFILING=y
3. Run the added command 'pm_lock'
4. Notice interrupt watchdog crash

### Code to reproduce this issue

Patch file attached.

## Debug Logs

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4592
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:13580
entry 0x40078574
I (260) cpu_start: Pro cpu up.
I (260) cpu_start: Starting app cpu, entry point is 0x40081174
0x40081174: call_start_cpu1 at /esp/esp-idf/components/esp32/cpu_start.c:224

I (0) cpu_start: App cpu up.
I (263) heap_init: Initializing. RAM available for dynamic allocation:
I (270) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (276) heap_init: At 3FFB9468 len 00026B98 (154 KiB): DRAM
I (282) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (289) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (295) heap_init: At 40091600 len 0000EA00 (58 KiB): IRAM
I (301) cpu_start: Pro cpu start user code
I (319) pm_esp32: Frequency switching config: CPU_MAX: 160, APB_MAX: 80, APB_MIN: XTAL, Light sleep: DISABLED
I (320) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

This is an example of ESP-IDF console component.
Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
esp32> pm_locks
Time: 4934208
Lock stats:
rtos1           CPU_FREQ_MAX    0    0        578     362930    8%
rtos0           CPU_FREQ_MAX    0    1        505     405122    Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4008dfa2  PS      : 0x00060d34  A0      : 0x8008c79a  A1      : 0x3ffbba20  
0x4008dfa2: vListInsert at /esp/esp-idf/components/freertos/list.c:188

A2      : 0x3ffba368  A3      : 0x3ffbc1e8  A4      : 0x3ffb1030  A5      : 0x00000001  
A6      : 0x00060d23  A7      : 0x3ffbc0b0  A8      : 0x3ffbc1e8  A9      : 0x3ffbc1e8  
A10     : 0x00000018  A11     : 0x00000018  A12     : 0x3ffbbc00  A13     : 0x00000001  
A14     : 0x00060d23  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

Backtrace: 0x4008dfa2:0x3ffbba20 0x4008c797:0x3ffbba40 0x4008b270:0x3ffbba60 0x400e952e:0x3ffbbaa0 0x400e97ed:0x3ffbbae0 0x400e4ceb:0x3ffbbb10 0x400e4c9c:0x3ffbbb40 0x400e5081:0x3ffbbb60 0x4000bd83:0x3ffbbb80 0x4000117d:0x3ffbbba0 0x400592fe:0x3ffbbbc0 0x4005937a:0x3ffbbbe0 0x400dd875:0x3ffbbc00 0x400da72e:0x3ffbbd20 0x400d3bc5:0x3ffbc030 0x400d2478:0x3ffbc080 0x400d3315:0x3ffbc0d0 0x400e861f:0x3ffbc0f0 0x400d3261:0x3ffbc110 0x400d1116:0x3ffbc140
0x4008dfa2: vListInsert at /esp/esp-idf/components/freertos/list.c:188

0x4008c797: vTaskPlaceOnEventList at /esp/esp-idf/components/freertos/tasks.c:3529

0x4008b270: xQueueGenericReceive at /esp/esp-idf/components/freertos/queue.c:2037

0x400e952e: uart_tx_all at /esp/esp-idf/components/driver/uart.c:1205

0x400e97ed: uart_write_bytes at /esp/esp-idf/components/driver/uart.c:1205

0x400e4ceb: uart_tx_char_via_driver at /esp/esp-idf/components/vfs/vfs_uart.c:301

0x400e4c9c: uart_write at /esp/esp-idf/components/vfs/vfs_uart.c:301

0x400e5081: esp_vfs_write at /esp/esp-idf/components/vfs/vfs.c:506 (discriminator 4)

0x400dd875: __sbprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:486 (discriminator 1)

0x400da72e: _vfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:875

0x400d3bc5: fprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fprintf.c:56

0x400d2478: esp_pm_dump_locks at /esp/esp-idf/components/esp32/pm_locks.c:189 (discriminator 2)

0x400d3315: pm_locks at /esp/esp-idf/examples/system/console/main/cmd_crashes.c:23

0x400e861f: esp_console_run at /esp/esp-idf/components/console/commands.c:199

0x400d3261: app_main at /esp/esp-idf/examples/system/console/main/console_example_main.c:168

0x400d1116: main_task at /esp/esp-idf/components/esp32/cpu_start.c:466

Core 1 register dump:
PC      : 0x4008b56f  PS      : 0x00060034  A0      : 0x8008bad9  A1      : 0x3ffbcb00  
0x4008b56f: uxPortCompareSet at /esp/esp-idf/components/freertos/port.c:406
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /esp/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /esp/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vPortCPUAcquireMutex at /esp/esp-idf/components/freertos/port.c:365

A2      : 0x3ffb1094  A3      : 0x3f402bbc  A4      : 0x00000014  A5      : 0x3ffb0fe0  
A6      : 0x3ffb1028  A7      : 0x00000001  A8      : 0x0000cdcd  A9      : 0x0000abab  
A10     : 0x00060023  A11     : 0xb33fffff  A12     : 0x0000cdcd  A13     : 0x3ffb0fb0  
A14     : 0x00000008  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4008b56f:0x3ffbcb00 0x4008bad6:0x3ffbcb20 0x4008d48c:0x3ffbcb40 0x4008d442:0x00000000
0x4008b56f: uxPortCompareSet at /esp/esp-idf/components/freertos/port.c:406
 (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /esp/esp-idf/components/freertos/portmux_impl.inc.h:86
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /esp/esp-idf/components/freertos/portmux_impl.h:98
 (inlined by) vPortCPUAcquireMutex at /esp/esp-idf/components/freertos/port.c:365

0x4008bad6: vTaskSwitchContext at /esp/esp-idf/components/freertos/tasks.c:3529

0x4008d48c: _frxt_dispatch at /esp/esp-idf/components/freertos/portasm.S:406

0x4008d442: _frxt_int_exit at /esp/esp-idf/components/freertos/portasm.S:206

Rebooting...

## Other items if possible

NOTE: In fact any logging (printf, ESP_LOGx, etc calls) under a critical section trigger the same problem. The attached patch also adds another simple command 'critical_log' that print logs under a critical section. This too triggers the crash. I can of course avoid such situations in my application code but this made me worried thinking ESP-IDF implementation might internally do this.

Searching through the code base I find several places where ESP_LOGx is called under critical sections. Mostly error condition cases and 'impossible' error cases but also some real cases. Look for example at function esp_intr_free which internally calls ESP_LOGV under a critical section. Enabling verbose logging and calling that function also render a watchdog crash. Example code in attached patch under command 'internal_log'

mikaelkanstrup, 2018-05-04:
[example-patch.tar.gz](https://github.com/espressif/esp-idf/files/1974117/example-patch.tar.gz)

BrianPugh, 2019-02-24:
Can we revisit this? I am unable to dump PM profiling data while using SPIFFS.

lsgunth, 2020-04-14:
Seems this is still an issue in v4.1-beta1, but it seems it now causes a panic instead of a watchdog failure. Seems esp_pm_dump_locks() is not usable anymore.

```
0x40091624: invoke_abort at libs/esp-idf/components/esp32/panic.c:159
0x40091a31: abort at libs/esp-idf/components/esp32/panic.c:174
0x40083474: lock_acquire_generic at libs/esp-idf/components/newlib/locks.c:143
0x40083589: _lock_acquire_recursive at libs/esp-idf/components/newlib/locks.c:171
0x400db30e: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)
0x400d4bc5: fprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fprintf.c:54
0x400d4070: esp_pm_dump_locks at libs/esp-idf/components/esp_common/src/pm_locks.c:181
```

igrr, 2020-04-15:
Thanks for the reminder @lsgunth. It looks like the same issue has been fixed on master in https://github.com/espressif/esp-idf/commit/d37a419dfcbf439da6f59b8827b3db531036d2fc. Will backport the fix to release/v4.1.

igrr, 2020-07-09:
Fixed on release/v4.1 branch in https://github.com/espressif/esp-idf/commit/ffe0cca423b9c14b36d87664899a1d6a37de910a.

closed 2020-07-09 by igrr

mill1000, 2021-02-05:
This issue is still present in `release/v4.0` branch. The above commit can be cherry-picked and fixes the abort, but the line buffer is too small and results in truncated output. A buffer size of 128 should be used.

Alvin1Zhang, 2021-02-05:
@mill1000 Thanks for reporting and updates, we have back ported the fix to release/4.0, and the fix is now under sync to GitHub, will update once the fix on release/4.0 is available, thanks.