espressif/esp-idf #1921: Bluetooth classic SPP cannot connect other device

yokkohsan, 2018-05-04:
On ***bt_spp_initiator*** sample, I try to connect the other device ([HC-06](http://a.co/eGxMQ8w)), but unfortunately I received error logs as:
```
I (5673) SPP_INITIATOR_DEMO: ESP_SPP_DISCOVERY_COMP_EVT status=0 scn_num=1
I (5683) SPP_INITIATOR_DEMO: ESP_SPP_CL_INIT_EVT
E (35903) BT: port_rfc_closed RFCOMM connection in state 1 closed: Peer connection failed (res: 16)
E (35913) BT: bta_gattc_mark_bg_conn unable to find the bg connection mask for: 98:d3:31:fd:8f:28
I (35913) SPP_INITIATOR_DEMO: ESP_SPP_CLOSE_EVT
```

blueMoodBHD, 2018-05-14:
Hi @yokkohsan Sorry for late reply because we just  purchased an HC-06. After testing, I find that `bt_spp_initiator` can connect to HC-06 and send data. Please test again, Thanks!

blueMoodBHD, 2018-05-16:
Hi @yokkohsan What is the IDF version you are using? Which I use is the latest master.

In addition, we set the IO capabilities to NoInputNoOutput now, so we can connect to HC-06 without pin code . Of course, we plan to add this feature in the future

yokkohsan, 2018-05-18:
I also use the latest master;
> commit a2556229aa6f55b16b171e3325ee9ab1943e8552

but the error log remains.
> I (5673) SPP_INITIATOR_DEMO: ESP_SPP_DISCOVERY_COMP_EVT status=0 scn_num=1
> I (5683) SPP_INITIATOR_DEMO: ESP_SPP_CL_INIT_EVT
> E (35903) BT: port_rfc_closed RFCOMM connection in state 1 closed: Peer [...]

yokkohsan, 2018-05-24:
sorry for late.
this is log.

```
I (996) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_RES_EVT
I (996) SPP_INITIATOR_DEMO: 98 d3 31 fd 8f 28 
E (996) BT: BTA_JvStartDiscovery
E (996) BT: BTA got event 0x1a04

E (1006) BT: bta_jv_start_discovery in, sdp_active:0
E (1006) BT: call SDP_InitDiscoveryDb, p_data->start_discovery.num_uuid:1
E (1016) BT: BTA got event 0x201

E (1016) BT: bta_dm_search_sm_execute state:1, event:0x201
E (1026) BT: BTA got event 0x206

E (1026) BT: bta_dm_search_sm_execute state:2, event:0x206
E (1036) BT: BTA got event 0x104

E (1036) BT: bta_dm_sm_execute event:0x4
I (1046) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
E (2556) BT: BTA got event 0x104

E (2566) BT: bta_dm_sm_execute event:0x4
E (2716) BT: BTA got event 0x104

E (2716) BT: bta_dm_sm_execute event:0x4
E (2716) BT: bta_dm_acl_change info: 0x0
E (2966) BT: bta_jv_start_discovery_cback res: 0x0
E (2966) BT: bta_jv_cb.uuid: 00001101-0000-1000-8000-00805f9b34fb
E (2966) BT: in, uuid:: 00001101-0000-1000-8000-00805f9b34fb
E (2976) BT: uuid len:16
E (2976) BT: shorten to 16 bits uuid: 1101
E (2986) BT: shorten uuid:: 01110000-0000-0000-0000-000000000000
E (2986) BT: p_sdp_rec:0x3ffc4d80
E (2996) BT: p_sdp_rec:0x0
I (2996) SPP_INITIATOR_DEMO: ESP_SPP_DISCOVERY_COMP_EVT status=0 scn_num=1
E (3006) BT: BTA_JvRfcommConnect
E (3006) BT: BTA got event 0x1a0d

E (3016) BT: port_find_mcb: not found, bd_addr:98:d3:31:fd:8f:28
E (3016) BT: RFCOMM_CreateConnection()  BDA: 98-d3-31-fd-8f-28
E (3026) BT: RFCOMM_CreateConnection(): scn:1, dlci:2, is_server:0 mtu:990, p_mcb:0x0
E (3036) BT: rfc_cb.port.port[1]:0x3ffcf308 allocated, last_port:1
E (3036) BT: port_allocate_port:bd_addr:98:d3:31:fd:8f:28
E (3046) BT: RFCOMM_CreateConnection(): scn:1, dlci:2, is_server:0 mtu:990, p_mcb:0x0, p_port:0x3ffcf308
E (3056) BT: RFCOMM_CreateConnection dlci:2 signal state:0xb
E (3066) BT: port_open_continue, p_port:0x3ffcf308
E (3066) BT: rfc_alloc_multiplexer_channel: bd_addr:98:d3:31:fd:8f:28
E (3076) BT: rfc_alloc_multiplexer_channel:is_initiator:1
E (3086) BT: rfc_alloc_multiplexer_channel rfc_cb.port.rfc_mcb[0].state:0
E (3086) BT: (rfc_cb.port.rfc_mcb[i].bd_addr:00:00:00:00:00:00
E (3096) BT: rfc_alloc_multiplexer_channel rfc_cb.port.rfc_mcb[1].state:0
E (3106) BT: (rfc_cb.port.rfc_mcb[i].bd_addr:00:00:00:00:00:00
E (3106) BT: rfc_alloc_multiplexer_channel rfc_cb.port.rfc_mcb[2].state:0
E (3116) BT: (rfc_cb.port.rfc_mcb[i].bd_addr:00:00:00:00:00:00
E (3126) BT: rfc_alloc_multiplexer_channel rfc_cb.port.rfc_mcb[3].state:0
E (3136) BT: (rfc_cb.port.rfc_mcb[i].bd_addr:00:00:00:00:00:00
E (3136) BT: rfc_timer_free
E (3146) BT: rfc_alloc_multiplexer_channel:is_initiator:1, create new p_mcb:0x3ffcfca4, index:0
E (3146) BT: rfc_timer_start - timeout:60
E (3156) BT: port_select_mtu application selected 990
E (3166) BT: port_select_mtu credit_rx_max 10, credit_rx_low 4, rx_buf_critical 15
E (3166) BT: rfc_mx_sm_state_idle - evt:6
E (3176) BT: bta_jv_alloc_rfc_cb port_handle:2 handle:0x81
E (3176) BT: PORT_SetEventCallback() handle:2
E (3186) BT: PORT_SetEventMask() handle:2 mask:0x30005
E (3196) BT: PORT_GetState() handle:2
E (3196) BT: PORT_SetState() handle:2
E (3196) BT: PORT_SetState() handle:2 FC_TYPE:0xc
I (3206) SPP_INITIATOR_DEMO: ESP_SPP_CL_INIT_EVT
E (33336) BT: rfc_mx_sm_state_wait_conn_cnf - evt:9
E (33336) BT: PORT_StartCnf result:3
E (33336) BT: PORT_StartCnf failed result:3
E (33336) BT: rfc_timer_free
E (33346) BT: bta_jv_port_mgmt_cl_cback:code:12, port_handle2
E (33346) BT: bta_jv_port_mgmt_cl_cback code=12 port_handle:2 handle:129
E (33356) BT: PORT_CheckConnection() handle:2
E (33356) BT: port_release_port, p_port:0x3ffcf308
E (33366) BT: rfc_port_closed DONE
E (33366) BT: rfc_port_timer_stop
E (33376) BT: port_release_port:p_port->keep_port_handle:0
E (33376) BT: port_release_port:Clean-up handle:2
E (33386) BT: rfc_port_timer_stop
I (33386) SPP_INITIATOR_DEMO: ESP_SPP_CLOSE_EVT
E (33436) BT: BTA got event 0x104

E (33436) BT: bta_dm_sm_execute event:0x4
E (33436) BT: BTA got event 0x104

E (33436) BT: bta_dm_sm_execute event:0x4
E (33436) BT: BTA got event 0x1f02

E (33446) BT: bta_gattc_mark_bg_conn unable to find the bg connection mask for: 98:d3:31:fd:8f:28
E (33456) BT: BTA got event 0x1f0c
```

for connection, i change the sample code a little bit.
because, HC-06 dose not provide a device name.
```c
static void esp_bt_gap_cb(esp_bt_gap_cb_event_t event, esp_bt_gap_cb_param_t *param)
{
    static uint8_t REMOTE_ADDR[ESP_BD_ADDR_LEN] = {0x98, 0xD3, 0x31, 0xFD, 0x8F, 0x28};

    switch(event){
        case ESP_BT_GAP_DISC_RES_EVT:
            ESP_LOGI(SPP_TAG, "ESP_BT_GAP_DISC_RES_EVT");
            esp_log_buffer_hex(SPP_TAG, param->disc_res.bda, ESP_BD_ADDR_LEN);
            if (memcmp(param->disc_res.bda, REMOTE_ADDR, ESP_BD_ADDR_LEN) == 0)
            {
                memcpy(peer_bd_addr, param->disc_res.bda, ESP_BD_ADDR_LEN);
                esp_spp_start_discovery(peer_bd_addr);
                esp_bt_gap_cancel_discovery();
            }

            // for (int i = 0; i < param->disc_res.num_prop; i++){
            //     if (param->disc_res.prop[i].type == ESP_BT_GAP_DEV_PROP_EIR
            //         && get_name_from_eir(param->disc_res.prop[i].val, peer_bdname, &peer_bdname_len)){
            //         esp_log_buffer_char(SPP_TAG, peer_bdname, peer_bdname_len);
            //         if (strlen(remote_device_name) == peer_bdname_len
            //             && strncmp(peer_bdname, remote_device_name, peer_bdname_len) == 0) {
            //             memcpy(peer_bd_addr, param->disc_res.bda, ESP_BD_ADDR_LEN);
            //             esp_spp_start_discovery(peer_bd_addr);
            //             esp_bt_gap_cancel_discovery();
            //         }
            //     }
            // }

            break;
```

blueMoodBHD, 2018-05-24:
First, the change is OK.

According to the logs, I still don't know the reason of the error.

So, could you please test with others. Such as, using **bt_spp_initiator** connect to other devices, or using other device connect to **HC-06**, Thanks.

PS: The version of my **HC-06** is **VERSION:3.0-20170609**.

yokkohsan, 2018-06-04:
`AT+VERSION` returns `linvorV1.8`.
it may not the latest version.

I tried to connect to HC-06 by using other device. That was success.

I found several [similar issues](https://github.com/espressif/esp-idf/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aopens+spp+classic)

denyskim, 2018-09-06:
I tried to connect to HC-06 ( FW 3.0) but unfortunately I received error logs
[0;32mI (865) SPP_INITIATOR_DEMO: ESP_SPP_INIT_EVT[0m
D (865) BT_APPL: BTA got event 0x102
[0m
D (875) BT_APPL: bta_dm_sm_execute event:0x2[0m
D (875) BT_APPL: BTA is generating EIR[0m
D (885) BT_APPL: BTA got event 0x103
[0m
D (885) BT_APPL: bta_dm_sm_execute event:0x3[0m
D (895) BT_APPL: BTA got event 0x200
[0m
D (895) BT_APPL: bta_dm_search_sm_execute state:0, event:0x200[0m
D (895) BT_APPL: bta_dm_search_start avoid_scatter=1[0m
D (905) BT_APPL: bta_dm_check_av:0[0m
D (905) BT_APPL: bta_dm_search_start status=1[0m
D (915) BT_APPL: BTA got event 0x104
[0m
D (915) BT_APPL: bta_dm_sm_execute event:0x4[0m
[0;32mI (915) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_STATE_CHANGED_EVT[0m
[0;32mI (3475) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_RES_EVT[0m
[0;32mI (3485) SPP_INITIATOR_DEMO: 98 d3 71 f9 6f d5 [0m
[0;32mI (3485) BT_APPL: BTA_JvStartDiscovery[0m
D (3485) BT_APPL: BTA got event 0x1a04
[0m
D (3485) BT_APPL: bta_jv_start_discovery in, sdp_active:0[0m
D (3495) BT_APPL: call SDP_InitDiscoveryDb, p_data->start_discovery.num_uuid:1[0m
D (3505) BT_APPL: BTA got event 0x201
[0m
D (3505) BT_APPL: bta_dm_search_sm_execute state:1, event:0x201[0m
D (3515) BT_APPL: BTA got event 0x206
[0m
D (3515) BT_APPL: bta_dm_search_sm_execute state:2, event:0x206[0m
D (3525) BT_APPL: BTA got event 0x104
[0m
D (3525) BT_APPL: bta_dm_sm_execute event:0x4[0m
[0;32mI (3525) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_STATE_CHANGED_EVT[0m
D (4595) BT_APPL: BTA got event 0x104
[0m
D (4595) BT_APPL: bta_dm_sm_execute event:0x4[0m
D (4675) BT_APPL: BTA got event 0x104
[0m
D (4675) BT_APPL: bta_dm_sm_execute event:0x4[0m
D (4675) BT_APPL: bta_dm_acl_change info: 0x10[0m
D (4925) BT_APPL: bta_jv_start_discovery_cback res: 0x0[0m
D (4935) BT_APPL: bta_jv_cb.uuid: 00001101-0000-1000-8000-00805f9b34fb[0m
D (4935) BT_APPL: in, uuid:: 00001101-0000-1000-8000-00805f9b34fb[0m
D (4935) BT_APPL: uuid len:16[0m
D (4945) BT_APPL: shorten to 16 bits uuid: 1101[0m
D (4945) BT_APPL: shorten uuid:: 01110000-0000-0000-0000-000000000000[0m
D (4955) BT_APPL: p_sdp_rec:0x3ffc9bd4[0m
D (4955) BT_APPL: p_sdp_rec:0x0[0m
[0;32mI (4955) SPP_INITIATOR_DEMO: ESP_SPP_DISCOVERY_COMP_EVT status=0 scn_num=1[0m
[0;32mI (4965) BT_APPL: BTA_JvRfcommConnect[0m
D (4975) BT_APPL: BTA got event 0x1a0d
[0m
D (4975) BT_RFCOMM: port_find_mcb: not found, bd_addr:98:d3:71:f9:6f:d5[0m
[0;32mI (4985) BT_RFCOMM: RFCOMM_CreateConnection()  BDA: 98-d3-71-f9-6f-d5[0m
[0;32mI (4985) BT_RFCOMM: RFCOMM_CreateConnection(): scn:1, dlci:2, is_server:0 mtu:990, p_mcb:0x0[0m
D (4995) BT_RFCOMM: rfc_cb.port.port[1]:0x3ffc7818 allocated, last_port:1[0m
D (5005) BT_RFCOMM: port_allocate_port:bd_addr:98:d3:71:f9:6f:d5[0m
[0;32mI (5015) BT_RFCOMM: RFCOMM_CreateConnection(): scn:1, dlci:2, is_server:0 mtu:990, p_mcb:0x0, p_port:0x3ffc7818[0m
D (5025) BT_RFCOMM: RFCOMM_CreateConnection dlci:2 signal state:0xb[0m
D (5025) BT_RFCOMM: port_open_continue, p_port:0x3ffc7818[0m
D (5035) BT_RFCOMM: rfc_alloc_multiplexer_channel: bd_addr:98:d3:71:f9:6f:d5[0m
D (5035) BT_RFCOMM: rfc_alloc_multiplexer_channel:is_initiator:1[0m
D (5045) BT_RFCOMM: rfc_alloc_multiplexer_channel rfc_cb.port.rfc_mcb[0].state:0[0m
D (5055) BT_RFCOMM: (rfc_cb.port.rfc_mcb[i].bd_addr:00:00:00:00:00:00[0m
D (5055) BT_RFCOMM: rfc_alloc_multiplexer_channel rfc_cb.port.rfc_mcb[1].state:0[0m
D (5065) BT_RFCOMM: (rfc_cb.port.rfc_mcb[i].bd_addr:00:00:00:00:00:00[0m
D (5075) BT_RFCOMM: rfc_alloc_multiplexer_channel rfc_cb.port.rfc_mcb[2].state:0[0m
[... 57 lines cut ...]
[0;31mE (35515) BT_BTM: Device not found
[0m
[0;31mE (35515) BT_HCI: Ctlr H/w error event - code:0x0
[0m
D (35525) BT_APPL:  bta_sys_hw_btm_cback was called with parameter: 1[0m
D (35545) BT_APPL:  bta_sys_hw_btm_cback was called with parameter: 0[0m
D (35545) BT_APPL: BTA got event 0x104
[0m
D (35545) BT_APPL: bta_dm_sm_execute event:0x4[0m
D (35545) BT_APPL: BTA got event 0x104
[0m
D (35555) BT_APPL: bta_dm_sm_execute event:0x4[0m
D (35555) BT_APPL: BTA got event 0x5
[0m
D (35565) BT_APPL: bta_sys_sm_execute state:2, event:0x5
[0m
D (35565) BT_APPL: bta_sys_hw_error
[0m
D (35565) BT_APPL: bta_dm_sys_hw_cback with event: 4[0m
D (35575) BT_APPL: BTA got event 0x2
[0m
D (35575) BT_APPL: bta_sys_sm_execute state:2, event:0x2
[0m
[0;32mI (35585) SPP_INITIATOR_DEMO: ESP_SPP_CLOSE_EVT[0m
[0;31mE (35585) SPP_INITIATOR_DEMO: authentication failed, status:9[0m

blueMoodBHD, 2018-09-07:
@yokkohsan @denyskim Hi, in my opinion, maybe your HC-06 does not support security simple pair, using legacy pair instead. But we don't support legacy pair, so, it paired fail and connect failed.

To fix it , I have added legacy pair into ESP-IDF(master) and merge it just now. It will  be synced to github in few days. The commit id is `8484b9cf` . Please wait patient and try again when it be synced. 

Thanks.

blueMoodBHD, 2018-09-07:
When you try, after getting `ESP_BT_GAP_PIN_REQ_EVT` event,  call `esp_bt_gap_pin_reply()` with the pin code of HC-06. Just like the example.

blueMoodBHD, 2018-09-07:
It have been synced to github, now. Please test again, thanks.

chegewara, 2018-09-07:
Hi @blueMoodBHD , sorry for offtopic, will this works with ble or just with classic BT?

blueMoodBHD, 2018-09-07:
Hi @chegewara , it works only with classic BT

blueMoodBHD, 2018-09-17:
Hi @yokkohsan @denyskim, Does it work well or not?

yokkohsan, 2018-09-20:
@blueMoodBHD Sorry for late response. Finally, it works!
(On the latest master branch, `bt_spp_initiator` was able to successfully connect to HC-06 by BT classic and to successfully transfer packets by each other)

```
...
I (843) SPP_INITIATOR_DEMO: ESP_SPP_INIT_EVT
I (863) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
I (873) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_RES_EVT
I (873) SPP_INITIATOR_DEMO: 98 d3 31 fd 8f 28 
I (883) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
I (2373) SPP_INITIATOR_DEMO: ESP_SPP_DISCOVERY_COMP_EVT status=0 scn_num=1
I (2383) SPP_INITIATOR_DEMO: ESP_SPP_CL_INIT_EVT
I (2563) SPP_INITIATOR_DEMO: ESP_BT_GAP_PIN_REQ_EVT min_16_digit:0
I (2563) SPP_INITIATOR_DEMO: Input pin code: 1234
I (2773) SPP_INITIATOR_DEMO: authentication success: HC-06
I (2773) SPP_INITIATOR_DEMO: 98 d3 31 fd 8f 28 
I (3413) SPP_INITIATOR_DEMO: ESP_SPP_OPEN_EVT
I (6853) SPP_INITIATOR_DEMO: speed(3.240854s ~ 6.677992s): 13.825457 kbit/s
...
I (310033) SPP_INITIATOR_DEMO: ESP_SPP_DATA_IND_EVT
...
``` 

Thank you.

yokkohsan, 2018-09-20:
I tried to connect my laptop (supports BT classic). And then, I received **status = 1**. 
```
I (6554) SPP_INITIATOR_DEMO: ESP_SPP_DISCOVERY_COMP_EVT status=1 scn_num=0
```
What is this?

### Debug LOG
```
...
I (5754) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_RES_EVT
I (5754) SPP_INITIATOR_DEMO: a4 34 d9 92 8d 2a 
I (5754) SPP_INITIATOR_DEMO: JPC20160928
I (5754) BT_APPL: BTA_JvStartDiscovery
D (5754) BT_APPL: BTA got ePPL: call SDP_InitDiscoveryDb, p_data->start_discovery.num_uuid:1
D (5774) BT_APPL: BTA got event 0x201
D (5784) BT_APPL: bta_dm_search_sm_execute state:1, event:0x201
D (5784) BT_APPL: BTA got event 0x206
D (5794) BT_APPL: bta_dm_search_sm_execute state:2, event:0x206
D (5794) BT_APPL: BTA got event 0x104
D (5804) BT_APPL: bta_dm_sm_execute event:0x4
I (5804) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
D (6484) BT_APPL: BTA got event 0x104
D (6484) BT_APPL: bta_dm_sm_execute event:0x4
D (6514) BT_APPL: BTA got event 0x104
D (6514) BT_APPL: bta_dm_sm_execute event:0x4
D (6514) BT_APPL: bta_dm_acl_change info: 0x10
D (6524) BT_APPL: bta_jv_start_discovery_cback res: 0x0
D (6524) BT_APPL: bta_jv_cb.uuid: 00001101-0000-1000-8000-00805f9b34fb
D (6524) BT_APPL: in, uuid:: 00001101-0000-1000-8000-00805f9b34fb
D (6534) BT_APPL: uuid len:16
D (6534) BT_APPL: shorten to 16 bits uuid: 1101
D (6544) BT_APPL: shorten uuid:: 01110000-0000-0000-0000-000000000000
D (6544) BT_APPL: p_sdp_rec:0x0
I (6554) SPP_INITIATOR_DEMO: ESP_SPP_DISCOVERY_COMP_EVT status=1 scn_num=0
D (9754) BT_APPL: BTA got event 0x104
D (9754) BT_APPL: bta_dm_sm_execute event:0x4
D (9754) BT_APPL: BTA got event 0x104
D (9754) BT_APPL: bta_dm_sm_execute event:0x4
// process is stopped :(
// why??
```

yokkohsan, 2018-09-21:
I see...
Then, I tried to use [MM-BTUD43](http://a.co/d/1mjj0xb) as BT dongle. (sorry for japanese device)
This may support BT classic and SPP profile according to [specifications](https://www.sanwa.co.jp/support/setsumeisyo/pdf/M/MM-BTUD43_44.pdf).

However, I received almost the same log as bellow:
### Debug LOG
```
I (8610) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_RES_EVT
I (8610) SPP_INITIATOR_DEMO: 00 09 dd 40 48 72 
I (8610) SPP_INITIATOR_DEMO: JPC20160928
I (8610) BT_APPL: BTA_JvStartDiscovery
D (8620) BT_APPL: BTA got event 0x1a04
D (8620) BT_APPL: bta_jv_start_discovery in, sdp_active:0
D (8630) BT_APPL: call SDP_InitDiscoveryDb, p_data->start_discovery.num_uuid:1
D (8640) BT_APPL: BTA got event 0x201
D (8640) BT_APPL: bta_dm_search_sm_execute state:1, event:0x201
D (8650) BT_APPL: BTA got event 0x206
D (8650) BT_APPL: bta_dm_search_sm_execute state:2, event:0x206
D (8650) BT_APPL: BTA got event 0x104
D (8660) BT_APPL: bta_dm_sm_execute event:0x4
I (8660) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
D (9130) BT_APPL: BTA got event 0x104
D (9130) BT_APPL: bta_dm_sm_execute event:0x4
D (9200) BT_APPL: BTA got event 0x104
D (9200) BT_APPL: bta_dm_sm_execute event:0x4
D (9200) BT_APPL: bta_dm_acl_change info: 0x10
D (9230) BT_APPL: bta_jv_start_discovery_cback res: 0x0
D (9230) BT_APPL: bta_jv_cb.uuid: 00001101-0000-1000-8000-00805f9b34fb
D (9230) BT_APPL: in, uuid:: 00001101-0000-1000-8000-00805f9b34fb
D (9240) BT_APPL: uuid len:16
D (9240) BT_APPL: shorten to 16 bits uuid: 1101
D (9250) BT_APPL: shorten uuid:: 01110000-0000-0000-0000-000000000000
D (9250) BT_APPL: p_sdp_rec:0x0
I (9260) SPP_INITIATOR_DEMO: ESP_SPP_DISCOVERY_COMP_EVT status=1 scn_num=0
D (12490) BT_APPL: BTA got event 0x104
D (12490) BT_APPL: bta_dm_sm_execute event:0x4
D (12490) BT_APPL: BTA got event 0x104
D (12490) BT_APPL: bta_dm_sm_execute event:0x4
```

But I could pairing.

### Debug LOG
```
D (45200) BT_APPL: BTA got event 0x104
D (45200) BT_APPL: bta_dm_sm_execute event:0x4
D (45200) BT_APPL: bta_dm_acl_change info: 0x10
D (45200) BT_APPL: BTA got event 0x104
D (45210) BT_APPL: bta_dm_sm_execute event:0x4
D (45270) BT_APPL: bta_dm_sp_cback: 1
D (45270) BT_APPL: dm status: 1
D (45270) BT_APPL: bta_dm_sp_cback: 0
D (45270) BT_APPL: io mitm: 0 oob_data:0
D (45280) BT_APPL: dm status: 1
D (45440) BT_APPL: bta_dm_sp_cback: 2
D (45440) BT_APPL: dm status: 1
I (45440) SPP_INITIATOR_DEMO: ESP_BT_GAP_CFM_REQ_EVT Please compare the numeric value: 86362
D (45440) BT_APPL: BTA got event 0x10d
D (45450) BT_APPL: bta_dm_sm_execute event:0xd
D (47280) BT_APPL: bta_dm_sp_cback: 8
D (47280) BT_APPL: dm status: 1
I (47310) SPP_INITIATOR_DEMO: authentication success: MyComputerName
I (47310) SPP_INITIATOR_DEMO: 00 09 dd ** ** **
D (50500) BT_APPL: BTA got event 0x104
D (50500) BT_APPL: bta_dm_sm_execute event:0x4
D (50500) BT_APPL: BTA got event 0x104
D (50500) BT_APPL: bta_dm_sm_execute event:0x4
```

blueMoodBHD, 2018-09-21:
If you want to connect a classic BT device with SPP,  the remote device should create a SPP server first. What I mean is use this dongle to check your laptop have created a SPP server or not. 

**dongle log**

```
baohongde@201707-10:~$ hcitool inq
Inquiring ...
    7C:7A:91:C4:4F:5E   clock offset: 0x4bad    class: 0x7e010c
    50:8F:4C:FB:96:4B   clock offset: 0x303f    class: 0x5a020c
    00:19:86:00:07:29   clock offset: 0x2242    class: 0x0c0104
    38:A4:ED:91:CD:A0   clock offset: 0x60dc    class: 0x5a020c
    4C:49:E3:A4:17:1D   clock offset: 0x08be    class: 0x5a020c
    30:AE:A4:04:80:06   clock offset: 0x7684    class: 0x2c0414
    80:7D:3A:E6:2F:76   clock offset: 0x02d0    class: 0x2c0414
    B8:98:F7:0F:1F:5F   clock offset: 0x1fca    class: 0x5a020c
baohongde@201707-10:~$ sdptool browse 80:7D:3A:E6:2F:76
Browsing 80:7D:3A:E6:2F:76 ...
Service RecHandle: 0x10000
Service Class ID List:
  "Generic Attribute" (0x1801)
Protocol Descriptor List:
  "L2CAP" (0x0100)
    PSM: 31
  "ATT" (0x0007)
    uint16: 0x0001
    uint16: 0x0005

Service RecHandle: 0x10001
Service Class ID List:
  "Generic Access" (0x1800)
Protocol Descriptor List:
  "L2CAP" (0x0100)
    PSM: 31
  "ATT" (0x0007)
    uint16: 0x0014
    uint16: 0x001c

Service Name: SPP_SERVER
Service RecHandle: 0x10002
Service Class ID List:
  "Serial Port" (0x1101)
Protocol Descriptor List:
  "L2CAP" (0x0100)
  "RFCOMM" (0x0003)
    Channel: 1
```
Only when there is SPP server ( "Serial Port" (0x1101)), we can connect it successful.
Thanks.

yokkohsan, 2018-09-29:
> If you want to connect a classic BT device with SPP, the remote device should create a SPP server first.

I see.
I prepared a COM port for classic BT as bellow:
[image]
then opened the COM port (created a SPP server).

Finally, my laptop could receive many binary data from ESP32 via BT, however the LOG is something different from the HC-06 case. Please check it.

### Debug LOG
```
...
I (2371) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_RES_EVT
I (2371) SPP_INITIATOR_DEMO: a4 34 d9 ** ** **
I (2371) BT_APPL: BTA_JvStartDiscovery
D (2381) BT_APPL: BTA got event 0x1a04
D (2381) BT_APPL: bta_jv_start_discovery in, sdp_active:0
D (2391) BT_APPL: call SDP_InitDiscoveryDb, p_data->start_discovery.num_uuid:1
D (2401) BT_APPL: BTA got event 0x201
D (2401) BT_APPL: bta_dm_search_sm_execute state:1, event:0x201
D (2411) BT_APPL: BTA got event 0x206
D (2411) BT_APPL: bta_dm_search_sm_execute state:2, event:0x206
D (2411) BT_APPL: BTA got event 0x104
D (2421) BT_APPL: bta_dm_sm_execute event:0x4
I (2421) SPP_INITIATOR_DEMO: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
D (3221) BT_APPL: BTA got event 0x104
D (3221) BT_APPL: bta_dm_sm_execute event:0x4
D (3241) BT_APPL: BTA got event 0x104
D (3241) BT_APPL: bta_dm_sm_execute event:0x4
D (3241) BT_APPL: bta_dm_acl_change info: 0x10
D (3311) BT_APPL: bta_jv_start_discovery_cback res: 0x0
D (3311) BT_APPL: bta_jv_cb.uuid: 00001101-0000-1000-8000-00805f9b34fb
D (3311) BT_APPL: in, uuid:: 00001101-0000-1000-8000-00805f9b34fb
D (3311) BT_APPL: uuid len:16
D (3321) BT_APPL: shorten to 16 bits uuid: 1101
D (3321) BT_APPL: shorten uuid:: 01110000-0000-0000-0000-000000000000
D (3331) BT_APPL: p_sdp_rec:0x3ffc9c00
D (3331) BT_APPL: p_sdp_rec:0x0
I (3341) SPP_INITIATOR_DEMO: ESP_SPP_DISCOVERY_COMP_EVT status=0 scn_num=1
I (3341) BT_APPL: BTA_JvRfcommConnect
D (3351) BT_APPL: BTA got event 0x1a0d
D (3351) BT_RFCOMM: port_find_mcb: not found, bd_addr:a4:34:d9:**:**:**
I (3361) BT_RFCOMM: RFCOMM_CreateConnection()  BDA: a4-34-d9-**-**-**
I (3361) BT_RFCOMM: RFCOMM_CreateConnection(): scn:2, dlci:4, is_server:0 mtu:990, p_mcb:0x0
D (3371) BT_RFCOMM: rfc_cb.port.port[1]:0x3ffc8fe8 allocated, last_port:1
D (3381) BT_RFCOMM: port_allocate_port:bd_addr:a4:34:d9:**:**:**
I (3391) BT_RFCOMM: RFCOMM_CreateConnection(): scn:2, dlci:4, is_server:0 mtu:990, p_mcb:0x0, p_port:0x3ffc8fe8
D (3401) BT_RFCOMM: RFCOMM_CreateConnection dlci:4 signal state:0xb
D (3401) BT_RFCOMM: port_open_continue, p_port:0x3ffc8fe8
D (3411) BT_RFCOMM: rfc_alloc_multiplexer_channel: bd_addr:a4:34:d9:**:**:**
D (3421) BT_RFCOMM: rfc_alloc_multiplexer_channel:is_initiator:1
D (3421) BT_RFCOMM: rfc_alloc_multiplexer_channel rfc_cb.port.rfc_mcb[0].state:0
D (3431) BT_RFCOMM: (rfc_cb.port.rfc_mcb[i].bd_addr:00:00:00:00:00:00
D (3441) BT_RFCOMM: rfc_alloc_multiplexer_channel rfc_cb.port.rfc_mcb[1].state:0
D (3441) BT_RFCOMM: (rfc_cb.port.rfc_mcb[i].bd_addr:00:00:00:00:00:00
D (3451) BT_RFCOMM: rfc_alloc_multiplexer_channel rfc_cb.port.rfc_mcb[2].state:0
D (3461) BT_RFCOMM: (rfc_cb.port.rfc_mcb[i].bd_addr:00:00:00:00:00:00
D (3461) BT_RFCOMM: rfc_alloc_multiplexer_channel rfc_cb.port.rfc_mcb[3].state:0
D (3471) BT_RFCOMM: (rfc_cb.port.rfc_mcb[i].bd_addr:00:00:00:00:00:00
D (3481) BT_RFCOMM: rfc_timer_free
D (3481) BT_RFCOMM: rfc_alloc_multiplexer_channel:is_initiator:1, create new p_mcb:0x3ffc9984, index:0
D (3491) BT_RFCOMM: rfc_timer_start - timeout:60
D (3501) BT_RFCOMM: port_select_mtu application selected 990
D (3501) BT_RFCOMM: port_select_mtu credit_rx_max 10, credit_rx_low 4, rx_buf_critical 15
D (3511) BT_RFCOMM: rfc_mx_sm_state_idle - evt:6
D (3521) BT_APPL: bta_jv_alloc_rfc_cb port_handle:2 handle:0x81
I (3521) BT_RFCOMM: PORT_SetEventCallback() handle:2
I (3531) BT_RFCOMM: PORT_SetEventMask() handle:2 mask:0x30005
I (3531) BT_RFCOMM: PORT_GetState() handle:2
I (3541) BT_RFCOMM: PORT_SetState() handle:2
[... 55 lines cut ...]
I (8711) BT_APPL: BTA_JvRfcommWrite
D (8711) BT_APPL: handE (8721) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (8721) task_wdt:  - IDLE0 (CPU 0)
E (8721) task_wdt: Tasks currently running:
E (8721) task_wdt: CPU 0: Btc_task
E (8721) task_wdt: CPU 1: IDLE1
le:0x81, hi:0, si:0
I (8741) BT_APPL: write ok
D (8751) BT_APPL: BTA got event 0x1a12
..
I (9921) BT_RFCOMM: PORT_WriteDataCO() handle:2
D (9931) BT_RFCOMM: PORT_WriteData 990 bytes
D (9931) BT_RFCOMM: PORT_Write : Data is being sent
D (9941) BT_APPL: bta_jv_port_event_cl_cback:2
D (9941) BT_APPL: bta_jv_port_event_cl_cback code=x4 port_handle:2 handle:129
D (9951) BT_RFCOMM: rfc_inc_credit:31
D (9951) BT_RFCOMM: PORT_DataInd with data length 1, p_mcb:0x3ffc9984,p_port:0x3ffc8fe8,dlci:4
D (9961) BT_APPL: bta_jv_port_data_co_cback, p_cb:0x3ffc8098, p_pcb:0x3ffc81ec, len:65535, type:1
I (9971) SPP_INITIATOR_DEMO: ESP_SPP_DATA_IND_EVT
I (9981) BT_APPL: BTA_JvRfcommWrite
D (9981) BT_APPL: handle:0x81, hi:0, si:0
I (9981) BT_APPL: write ok
D (9991) BT_APPL: BTA got event 0x1a12
...
```

blueMoodBHD, 2018-10-08:
I am glad to hear that it works OK now. 
The different of the log is caused by flow control, there is no problem.

yokkohsan, 2018-10-08:
Thank you very much!

closed 2018-10-08 by FayeY

sueng66, 2022-09-02:
I try to connect to BT printer got this log : 
`I (129877) BT_L2CAP: L2CAP - Calling Config_Rsp_Cb(), CID: 0x0041
I (129887) BT_L2CAP: L2CA_ConfigRsp()  CID: 0x0041  Result: 0 MTU present:0 Flush TO:0 FCR:0 FCS:0
I (129887) BT_L2CAP: L2CA_DataWrite()  CID: 0x0041  Len: 4
I (129897) BT_L2CAP: L2CA_DataWrite()  CID: 0x0041  Len: 14
I (129927) BT_L2CAP: L2CA_DataWrite()  CID: 0x0041  Len: 4
I (130107) BT_RFCOMM: PORT_CheckConnection() handle:6
W (130117) BT_RFCOMM: port_rfc_closed RFCOMM connection in server:0 state 1 closed: Start failed (res: 12)
I (130117) BT_L2CAP: L2CA_DataWrite()  CID: 0x0041  Len: 4
W (130127) BT_RFCOMM: port_rfc_closed RFCOMM connection in server:0 state 0 closed: Closed (res: 19)
I (130137) BT_L2CAP: L2CA_DisconnectReq()  CID: 0x0041
I (130137) BT_BTM: BTM_SetPowerMode: pm_id 128 BDA: a638053e mode:0x0
I (130147) BT_STOREFORWARD: ESP_SPP_CLOSE_EVT
I (130407) BT_BTM: BTM_IsAclConnectionUp: RemBdAddr: 022fa638053e

I (130407) BT_BTM: BTM_IsAclConnectionUp: RemBdAddr: 022fa638053e

I (130407) BT_BTM: BTM_IsAclConnectionUp: RemBdAddr: 022fa638053e

I (130417) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
E (130427) BT_BTC:  btc_dm_auth_cmpl_evt() Authentication fail reason 5
E (130427) BT_STOREFORWARD: authentication failed, status:9
`
Is that the same problem ?

I am using IDF 4.4
thanks