espressif/esp-idf #1926: RMT crashes with too large clk_div and idle_threshold

imxieyi, 2018-05-05:
## Environment

- Development Kit:      [none]
- Core (if using chip or module): [ESP-WROOM32 & Widora-AIR]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //f586f5e6
- Development Env:		[Make]
- Operating System:		[Windows & Ubuntu]
- Power Supply:         [USB]

## Problem Description
Unable to use `xRingbufferReceive` to get RMT item with some value combinations of `rx_config.idle_threshold` and `clk_div`. For example, when `clk_div` is 80 with `rx_config.idle_threshold` larger than 10000, RMT shows error when calling `xRingbufferReceive`.

### Expected Behavior
Calling `xRingbufferReceive` returns a series of `rmt_item32_t`, which represent the pulse pattern from chosen gpio pin.

### Actual Behavior
RMT shows error with code `0x13000040`.

### Steps to repropduce
The following code is a minimal example. It only produces error with the selected pin attached to a signal source like the following photo:

[image]

### Code to reproduce this issue
```c
#include "driver/gpio.h"
#include "driver/rmt.h"
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/ringbuf.h"

void app_main() {
    // Setup RMT
    rmt_config_t rmt_rx;
    rmt_rx.channel = RMT_CHANNEL_0;
    rmt_rx.rmt_mode = RMT_MODE_RX;
    rmt_rx.gpio_num = GPIO_NUM_5;
    rmt_rx.mem_block_num = 1;
    rmt_rx.clk_div = 80;
    rmt_rx.rx_config.filter_en = 0;
    rmt_rx.rx_config.idle_threshold = 10000;
    rmt_config(&rmt_rx);
    rmt_driver_install(rmt_rx.channel, 1000, 0);
    ESP_LOGI("IR", "RMT driver installed");

    // Get RMT buffer handle
    RingbufHandle_t rb = NULL;
    rmt_get_ringbuf_handle(rmt_rx.channel, &rb);

    rmt_rx_start(rmt_rx.channel, 1);
    ESP_LOGI("IR", "RMT started");

    size_t rx_size = 0;
    uint32_t duration, level;
    while (1) {
        rmt_item32_t *item = (rmt_item32_t *)xRingbufferReceive(rb, &rx_size, 1000);
        ESP_LOGI("IR", "%d", (uintptr_t)item);
        if (item) {
            vRingbufferReturnItem(rb, (void *)item);
        }
    }
}
```

## Debug Logs
```
I (18) cpu_start: Starting scheduler on PRO CPU.
D (0) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (0) cpu_start: Starting scheduler on APP CPU.
D (19) heap_init: New heap initialised at 0x3ffe0440
D (19) heap_init: New heap initialised at 0x3ffe4350
D (19) intr_alloc: Connected src 16 to int 12 (cpu 0)
D (20) rmt: Rmt Rx Channel 0|Gpio 5|Sclk_Hz 80000000|Div 80|Thresold 10000|Filter 97
D (21) intr_alloc: Connected src 47 to int 13 (cpu 0)
I (21) IR: RMT driver installed
I (21) IR: RMT started
E (130) rmt: RMT[0] ERR
E (130) rmt: status: 0x13000040
I (1021) IR: 0
I (2021) IR: 0
I (3021) IR: 0
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

imxieyi, 2018-05-13:
Upon further testing I found that the problem might come from `idle_threshold`.

In my experiment, I tried different `clk_div` values (8, 10, 20, 40, 80). All of the `clk_div` values has a special `idle_threshold` value, which I named "crash point". If `idle_threshold` is below "crash point", RMT works fine. If `idle_threshold` is larger than "crash point", RMT crashes on first signal arriving. Although it sounds strange, "crash point" is always around ticks equivalent to 4960~4970us no matter what `clk_div` is, even if the number of ticks does not exceed maximum of 15-bit integer which stores the measurements.

In my project I need to measure pulses around 7000us with precision of 1us. However it seems that RMT does not support measuring any pulse longer than 5000us. I cannot find any datasheet related to RMT. The register holding the status code is not documented anywhere on the Internet.

Is this a hardware issue? Or is there any other ways to measure pulses? Thanks ahead for any help!

koobest, 2018-05-15:
Hi, @imxieyi 
Did you mean that you need to measure the 148Hz pulses ?

imxieyi, 2018-05-15:
@koobest Yes. Actually in my specific project I need to measure the 7000us gap between 100us pulses.

koobest, 2018-05-23:
Hi, @imxieyi,
If the maximum low level of your measured pulse lasts longer than idle_threshold, it will be OK, otherwise it will go wrong. in you configuration above, if the low level is less than 10000us，an error will occurs.

bschwind, 2018-10-04:
I don't think this is related, but for those searching through github issues, I wanted to leave a note.

```
E (130) rmt: status: 0x13000040
```

I still don't know what exactly this error means, but I encountered it when trying to decode 38KHz IR packets from an air conditioner remote that sends rather large packets. I ended up having to use more memory blocks for the IR channel I was using.

```c
rmt_rx.mem_block_num = 1;
```

This means the configured channel (of which there are 8), will receive 1 memory block (64 x 32 bits) from the shared RMT memory block (512 x 32 bits). You can specify more blocks if you need more memory. There are rules for how you specify the extra blocks, you can read about it [here](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/rmt.html#_CPPv221rmt_set_mem_block_num13rmt_channel_t7uint8_t)

> The 8 channels share a 512x32-bit RAM block which can be read and written by the processor cores over the APB bus, as well as read by the transmitters and written by the receivers.
> Note: If memory block number of one channel is set to a value greater than 1, this channel will occupy the memory block of the next channel. Channel 0 can use at most 8 blocks of memory, accordingly channel 7 can only use one memory block.

Additionally, I had to increase the ringbuffer size to accommodate the larger packets I was receiving:

```c
// rmt_driver_install(rmt_rx.channel, 1000, 0);
rmt_driver_install(rmt_rx.channel, 3000, 0); // Set the ringbuffer size to 3000 bytes
```

Again, likely not related to the original issue posted here, but this was the only place I found that mentioned the error status of `0x13000040`

koobest, 2018-10-29:
Hi, @bschwind, Please see the issue, https://github.com/espressif/esp-idf/issues/1175

bschwind, 2018-10-29:
Hi @koobest 

Sorry, it probably wasn't clear but I solved my issue and was just providing information for others searching for this error. Indeed it was a full memory error and increasing the number of blocks resolved it.

Thanks!

closed 2018-10-30 by FayeY