espressif/esp-idf #1928: JTAG debug of blink due to:  Error: Too large number of threads 2044975697!

jjfretired, 2018-05-05:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](http://esp-idf.readthedocs.io/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](http://esp-idf.readthedocs.io/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](http://esp-idf.readthedocs.io/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Core (if using chip or module): [ESP32-Wrover-I|ESP32-Wrover|ESP32-PICO-D4|ESP32-SOLO-I|ESP32-WROOM-32U|ESP-WROOM-32D|ESP-WROOM32|ESP32]
- IDF version (``git rev-parse --short HEAD`` to get the commit id.): 
    //bd6ea4393c7d2f059fc4decc70f1ec3eb3597268
- Development Env:		[Arduino IDE|Make|Eclipse|other]
- Operating System:		[Windows|Ubuntu|MacOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

jjfretired, 2018-05-05:
I'm using the wrover kit V3 with the ILI9341 LCD running under ubuntu.

1 - Connecting the pins as seen in the documentation gave execution errors.  Looking at the schematic I noticed that the pin above the gnd, TD was incorrect.  I removed it and it executed correctly but with further problems.   A

I copied blink to another library for testing (jjf) and  issued the make menuconfig.  I changed the gpio pin to pin 15 as pin 5 caused the LCD to blink and not the diode.
-----------------------------------------------------------------------------------------------------------------------------------
See **************** below for error message.
__________________________________________________________________________________
Execution of openocd in terminal 1

jjjf@jjjf-Z97X-SLI:~/esp/openocd-esp32$ bin/openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32-wrover.cfg -c "program_esp32 $HOME/jjf/blink/build/blink.bin 0x10000 verify"
Open On-Chip Debugger 0.10.0-dev (2018-04-18-12:03)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
none separate
adapter speed: 20000 kHz
esp32 interrupt mask on
force hard breakpoints
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: none
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: none
Info : Target halted. PRO_CPU: PC=0x40009081 (active)    APP_CPU: PC=0x00000000 
esp32: target state: halted
Info : esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000 
esp32: target state: halted
Info : esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400 
esp32: target state: halted
** Programming Started **
auto erase enabled
Info : Use core0 of target 'esp32'
Info : Target halted. PRO_CPU: PC=0x40091D07 (active)    APP_CPU: PC=0x400076DD 
Info : Auto-detected flash size 4096 KB
Info : Using flash size 4096 KB
Info : Use core0 of target 'esp32'
Info : Target halted. PRO_CPU: PC=0x40091D07 (active)    APP_CPU: PC=0x400076E2 
Info : Use core0 of target 'esp32'
Info : Target halted. PRO_CPU: PC=0x40091D07 (active)    APP_CPU: PC=0x400076E0 
wrote 139264 bytes from file /home/jjjf/jjf/blink/build/blink.bin in 3.099145s (43.883 KiB/s)
** Programming Finished **
** Verify Started **
Info : Use core0 of target 'esp32'
Info : Target halted. PRO_CPU: PC=0x40091D07 (active)    APP_CPU: PC=0x400076DD 
read 138640 bytes from file /home/jjjf/jjf/blink/build/blink.bin and flash bank 0 at offset 0x00010000 in 0.694212s (195.028 KiB/s)
contents match
** Verified OK **

-------------------------------------------------------------------------------

executiion of blink in terminal 2

jjjf@jjjf-Z97X-SLI:~/jjf/blink$ xtensa-esp32-elf-gdb -x gdbinit build/blink.elf
GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
[... 18 lines cut ...]
esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Target halted. PRO_CPU: PC=0x40000400 (active)    APP_CPU: PC=0x40000400 
esp32: target state: halted
Hardware assisted breakpoint 1 at 0x400d2043: file /home/jjjf/jjf/blink/main/./blink.c, line 45.
0x0:	0x00000000

------------------------------------------------------------------------------

output on terminal 1 after starting blink in terminal 2.

** Verified OK **   (from terminal 1)
Info : accepting 'gdb' connection on tcp/3333
***************************************************************
Error: Too large number of threads 2044975697!
***************************************************************
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
cpu0: Current bits set: BreakIn BreakOut RunStallIn
cpu1: Current bits set: BreakIn BreakOut RunStallIn
Info : esp32: Debug controller was reset (pwrstat=0x5F, after clear 0x0F).
Info : esp32: Core was reset (pwrstat=0x5F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x5000004B (active)    APP_CPU: PC=0x00000000 
esp32: target state: halted
Info : esp32: Core was reset (pwrstat=0x1F, after clear 0x0F).
Info : Target halted. PRO_CPU: PC=0x40000400 (act

igrr, 2018-05-06:
GPIO15 is one of the JTAG pins (others are 12-14). If you reassign it as GPIO, JTAG will no longer work. On wrover-kit, you can use IOs 0, 2, 4 which are connected to RGB LED.

jjfretired, 2018-05-06:
Changed the GPIO to io2 and I did get the gdb prompt and entered n.  But, Target halted and got the output belo the ----------- line..
Other times I change the GPIO to IO19 and it hangs after 
Hardware assisted breakpoint 1 at 0x400d1f8f: file /home/jjjf/jjf/blink/main/blink.c, line 45.
0x0:	0x00000000.  (no gdb prompt).
Change it back to IO2, power off, reboot and again it hangs after Hardware assisted ......... 0000.
-----------------------------------------------------------------------------------------------------------------------------------
PRO_CPU: PC=0x400D1F8F (active)    APP_CPU: PC=0x400D1AB2 
[New Thread 1073435392]
[New Thread 1073434004]
[New Thread 1073438048]
[New Thread 1073426892]
[New Thread 1073412716]
[New Thread 1073413484]
[New Thread 1073428024]
[Switching to Thread 1073432484]

Temporary breakpoint 1, app_main () at /home/jjjf/jjf/blink/main/blink.c:45
45	    xTaskCreate(&blink_task, "blink_task", configMINIMAL_STACK_SIZE, NULL, 5, NULL);
(gdb) n
Target halted. PRO_CPU: PC=0x400D1F92 (active)    APP_CPU: PC=0x400D1AB2 
Target halted. PRO_CPU: PC=0x400D1F95 (active)    APP_CPU: PC=0x400D1AB2 
Target halted. PRO_CPU: PC=0x400D1F98 (active)    APP_CPU: PC=0x400D1AB2 
Target halted. PRO_CPU: PC=0x400D1F9B (active)    APP_CPU: PC=0x400D1AB2 
Target halted. PRO_CPU: PC=0x400D1F9D (active)    APP_CPU: PC=0x400D1AB2 
Target halted. PRO_CPU: PC=0x400D1FA0 (active)    APP_CPU: PC=0x400D1AB2 
Target halted. PRO_CPU: PC=0x400D1FA3 (active)    APP_CPU: PC=0x400D1AB2 
Target halted. PRO_CPU: PC=0x400D1FA6 (active)    APP_CPU: PC=0x400D1AB2 
Target halted. PRO_CPU: PC=0x400851DC (active)    APP_CPU: PC=0x400D1AB2 
Target halted. PRO_CPU: PC=0x400D1FA9 (active)    APP_CPU: PC=0x40080340 
Target halted. PRO_CPU: PC=0x400800C3 (active)    APP_CPU: PC=0x40080340 
/builds/idf/crosstool-NG/.build/src/gdb-7.10/gdb/regcache.c:697: internal-error: regcache_raw_read_unsigned: Assertion `regnum >= 0 && regnum < regcache->descr->nr_raw_registers' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n)

igrr, 2018-05-09:
@jjfretired I'm going to close this one because the same issue is already tracked in https://github.com/espressif/openocd-esp32/issues/37. We have some patches to GDB on the way to fix this. Please go to the linked issue and click "Subscribe" to get notified when this happens.

closed 2018-05-09 by igrr