espressif/esp-idf #1929: wrover kit V3 SD_Card example fails: E (297) sdmmc_cmd: sdmmc_card_init

jjfretired, 2018-05-06:
eap32-wrover kit V3 with ILI9341 LCD.

Moved the sd_card example to another directory , make all, make flash monitor and received the error message at end of load.  Also uncommented #define USE_SPI_MODE in ...main.c issued make clean,
make flash monitor and received the following message.

E (297) sdmmc_cmd: sdmmc_card_init: go_idle_state (1) returned 0x107
E (297) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place

make flash with #define USE_SPI_MODE commented. (default mode)

.jjjf@jjjf-Z97X-SLI:~/jjf/sd_card$ make flash monitor
Flashing binaries to serial port /dev/ttyUSB1 (app at offset 0x10000)...
esptool.py v2.3.1
Connecting......
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 20080 bytes to 11819...
Wrote 20080 bytes (11819 compressed) at 0x00001000 in 1.1 seconds (effective 152.1 kbit/s)...
Hash of data verified.
Compressed 201488 bytes to 101106...
Wrote 201488 bytes (101106 compressed) at 0x00010000 in 9.0 seconds (effective 179.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 1555.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
MONITOR
--- idf_monitor on /dev/ttyUSB1 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5648
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:14324
entry 0x40078620
I (30) boot: ESP-IDF v3.1-dev-921-g0d65f3b7-dirty 2nd stage bootloader
I (30) boot: compile time 17:06:19
I (36) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0af90 ( 44944) map
I (111) esp_image: segment 1: paddr=0x0001afb8 vaddr=0x3ffb0000 size=0x021e8 (  8680) load
I (115) esp_image: segment 2: paddr=0x0001d1a8 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/jjjf/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (118) esp_image: segment 3: paddr=0x0001d5b0 vaddr=0x40080400 size=0x02a60 ( 10848) load
I (131) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x1ab9c (109468) map
0x400d0018: _flash_cache_start at ??:?

I (174) esp_image: segment 5: paddr=0x0003abbc vaddr=0x40082e60 size=0x06720 ( 26400) load
0x40082e60: esp_cpu_unstall at /home/jjjf/esp/esp-idf/components/soc/esp32/cpu_util.c:37

I (185) esp_image: segment 6: paddr=0x000412e4 vaddr=0x400c0000 size=0x00000 (     0) load
I (191) boot: Loaded app from partition at offset 0x10000
I (191) boot: Disabling RNG early entropy source...
I (195) cpu_start: Pro cpu up.
I (198) cpu_start: Starting app cpu, entry point is 0x40080e88
0x40080e88: call_start_cpu1 at /home/jjjf/esp/esp-idf/components/esp32/cpu_start.c:224

I (0) cpu_start: App cpu up.
I (209) heap_init: Initializing. RAM available for dynamic allocation:
I (216) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (222) heap_init: At 3FFB2AB8 len 0002D548 (181 KiB): DRAM
I (228) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (234) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (241) heap_init: At 40089580 len 00016A80 (90 KiB): IRAM
I (247) cpu_start: Pro cpu start user code
I (265) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (267) example: Initializing SD card
I (267) example: Using SDMMC peripheral
I (267) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (297) sdmmc_cmd: sdmmc_card_init: send_op_cond (1) returned 0x107
E (297) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines h

igrr, 2018-05-06:
Please check that you don't have JTAG jumpers installed (JTAG uses the same pins as SD card, might interfere somehow).

Then please set the log level to "Verbose" in menuconfig, run the example again, and attach the output.

Next thing to try is to reduce the clock frequency of SDMMC. To do that, add this line after `    sdmmc_host_t host = SDMMC_HOST_DEFAULT()`:

```c++
    host.max_freq_khz = SDMMC_FREQ_PROBING;
```

jjfretired, 2018-05-07:
I had the JTAG jumpers installed and when I removed them it read the card.  Thank you for the help.

closed 2018-05-07 by jjfretired