espressif/esp-idf #1930: Support for pSRAM with PICO-D4

iot-bits, 2018-05-06:
I see that the latest ESP-IDF does not yet support pSRAM on PICO-D4. Any update on when it could be supported?
Also, is there a way to get pSRAM working with PICO-D4 right now by modifying the pin definitions according to PICO-D4 flash connection in spiram_psram.c?

qwert1213131, 2018-09-03:
anyone know how to add psram to pico?

costaud, 2018-09-03:
If you want to try modifying the code in spiram_psram.c, I think you can do this:
__(Note that it is not recommended to modify the idf code, just for test purpose. Do not use it for production for now.)__
We will think it over and add support for psram with PICO in idf.

1. Change the pin index according to the PICO pin definition in data sheet.
```
#define FLASH_CLK_IO      6  //Psram clock is a delayed version of this in 40MHz mode
#define FLASH_CS_IO       11
#define PSRAM_CLK_IO      17
#define PSRAM_CS_IO       16
#define PSRAM_SPIQ_IO     7
#define PSRAM_SPID_IO     8
#define PSRAM_SPIWP_IO    10
#define PSRAM_SPIHD_IO    9
```

2. Remove the chip version check:
```
 else if (pkg_ver == EFUSE_RD_CHIP_VER_PKG_ESP32PICOD4) {
        ESP_EARLY_LOGE(TAG, "ESP32PICOD4  do not support psram yet");
        return ESP_FAIL;
    }
```

egimbernat, 2018-10-11:
Take a look at [https://www.esp32.com/viewtopic.php?t=6866](url) ;)

ESP32DE, 2018-10-31:
> 
> 
> If you want to try modifying the code in spiram_psram.c, I think you can do this:
> **(Note that it is not recommended to modify the idf code, just for test purpose. Do not use it for production for now.)**
> We will think it over and add support for psram with PICO in idf.
> 
>     1. Change the pin index according to the PICO pin definition in data sheet.
> 
> 
> ```
> #define FLASH_CLK_IO      6  //Psram clock is a delayed version of this in 40MHz mode
> #define FLASH_CS_IO       11
> #define PSRAM_CLK_IO      17
> #define PSRAM_CS_IO       16
> #define PSRAM_SPIQ_IO     7
> #define PSRAM_SPID_IO     8
> #define PSRAM_SPIWP_IO    10
> #define PSRAM_SPIHD_IO    9
> ```
> 
>     1. Remove the chip version check:
> 
> 
> ```
>  else if (pkg_ver == EFUSE_RD_CHIP_VER_PKG_ESP32PICOD4) {
>         ESP_EARLY_LOGE(TAG, "ESP32PICOD4  do not support psram yet");
>         return ESP_FAIL;
>     }
> ```

ups @costaud seems there are some things go wrong :)
```

#define FLASH_CLK_IO      6  
#define FLASH_CS_IO       11

#define PSRAM_CLK_IO      10 // 17
#define PSRAM_CS_IO       09 // 16
#define PSRAM_SPIQ_IO     7
#define PSRAM_SPID_IO     8
#define PSRAM_SPIWP_IO    10
#define PSRAM_SPIHD_IO    9
```

```

#ifdef PICO
   /* TEST Config for the PICO-D4
       * - ESP32_PICO_Core_Board_V2
       * - ESP32_PICO_Core_Board_V3
       * Note: V2 & V3 Boards have ESP32 Rev=0
       * psram_enable works correctly only with v1 (post Feb 2017) and later ESP32 revisions.
       * 29 August 2017 rudi ;-)
       */
   /* SPICS0 CS */
   gpio_matrix_out(16, SPICS0_OUT_IDX, 0, 0); //11
    /* SPICLK CLK */
   gpio_matrix_out(6, SPICLK_OUT_IDX, 0, 0);
    /* SPIQ alias SD0 */
    gpio_matrix_out(17, SPIQ_OUT_IDX, 0, 0);   // 7
    gpio_matrix_in(17,SPIQ_IN_IDX, 0);         // 7
    /* SPID alias SD1 */
    gpio_matrix_out(8, SPID_OUT_IDX, 0, 0);
    gpio_matrix_in(8, SPID_IN_IDX, 0);
    /* SPIHD HOLD alias SD2 */
    gpio_matrix_out(11, SPIHD_OUT_IDX, 0, 0); // 9
    gpio_matrix_in(11, SPIHD_IN_IDX, 0);      // 9
    /* WP alias SD3 */
    gpio_matrix_out(7, SPIWP_OUT_IDX, 0, 0); // 10 a try with 7 like D2WD
    gpio_matrix_in(7, SPIWP_IN_IDX, 0);      // 10 a try with 7 like D2WD

#else

    gpio_matrix_out(6, SPICLK_OUT_IDX, 0, 0);
    gpio_matrix_out(11, SPICS0_OUT_IDX, 0, 0);
    gpio_matrix_out(7, SPIQ_OUT_IDX, 0, 0);
    gpio_matrix_in(7,SPIQ_IN_IDX, 0);
    gpio_matrix_out(8, SPID_OUT_IDX, 0, 0);
    gpio_matrix_in(8, SPID_IN_IDX, 0);
    gpio_matrix_out(10, SPIWP_OUT_IDX, 0, 0);
    gpio_matrix_in(10, SPIWP_IN_IDX, 0);
    gpio_matrix_out(9, SPIHD_OUT_IDX, 0, 0);
    gpio_matrix_in(9, SPIHD_IN_IDX, 0);
#endif
```

best wishes
rudi :-)

qwert1213131, 2018-11-09:
@ESP32DE
Thank you for your reply.
I am a bit confused.
If you have time, can you draw a sketch of the connection between pico and sram?

ESP32DE, 2019-01-03:
> 
> 
> @ESP32DE
> Thank you for your reply.
> I am a bit confused.
> If you have time, can you draw a sketch of the connection between pico and sram?

https://www.esp32.com/viewtopic.php?f=2&t=2713&start=100#p33336

allanchen24, 2019-03-14:
The feature that support  for psram with PICO-D4 had released to github.
And the commit ID is bd8733f74f98d95aec04d610426df92cb39c92e6.

closed 2019-03-14 by Alvin1Zhang

app-z, 2021-03-31:
https://github.com/espressif/esp-idf/issues/6350

Do u have any issue?